espressif/arduino-esp32 #9280: Enable PSRAM ECC function in build flags

s54b32julian, 2024-02-22:
### Related area

PSRAM

### Hardware specification

ESP32-S3 Wroom N8 & N16

### Is your feature request related to a problem?

On Page 3 of the ESP32 s3 wroom documentation is mentioned that the max ambient temp can be raised to 85°C, if the "PSRAM ECC function" is enabled. How can it be in the build accomplished, that the ECC function is enabled?
https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf

I found very few information about this function on esp32, but 65°C is very quickly reached if the board is exposed to the sun. 
Maybe the are the Fuses: EFUSE_FLASH_ECC_EN and EFUSE_FLASH_ECC_MODE

Many thanks in advance!

### Describe the solution you'd like

set a build flag to enable ECC function on N8 and N16 variants

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-02-22:
This has to with PSRAM, not your regular flash, so the R number, not the N (or H).
It also consumes 1/16 of your PSRAM for the ECC, and slows down every transaction.  I don't think this should be a default setting.
A more useful effort is to make the [custom compilation process easier](https://github.com/espressif/arduino-esp32/tree/master/idf_component_examples/Hello_world), as the Espressif team is doing

lucasssvaz, 2024-02-22:
I agree with @lbernstone. This is also going to be made easier by https://github.com/espressif/esp32-arduino-lib-builder/pull/158.

s54b32julian, 2024-02-23:
Thanks for fast response! I think the price you have to pay. Losing 1/16th of space and (also roughly 1/16th of the speed?) is a low price to pay, to gain such a huge bump in temperature stability imo)
Especially for outdoor use or in car use. Even more in hot areas of the world.

Is there a way how I can activate this ECC function manually at the moment? I'm using platformIO.

lucasssvaz, 2024-02-23:
@s54b32julian You'll need to re-generate the static libs for your target. Check https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html

lbernstone, 2024-02-23:
Have you tried setting your CPU speed to 80MHz?  I rarely have temperature issues unless the cpu is overclocked.

s54b32julian, 2024-12-25:
Any update on this? I really need to improve the temperature stability in hot environment. The ambient temp ist only rated to 65°C on most Esp32-S3, which is really to low for some applications. The 85°C would help much.

lucasssvaz, 2024-12-26:
@s54b32julian There is nothing to be done here. You can recompile the libraries as you want with the lib builder or use arduino as an IDF component. For the fuses, they can be burnt using esptool

s54b32julian, 2024-12-26:
> @s54b32julian There is nothing to be done here. You can recompile the libraries as you want with the lib builder or use arduino as an IDF component. For the fuses, they can be burnt using esptool

But how to enable it, what is the workflow to use this feature? I don't find a build flag or config option to enable this on Arduino IDE or VS Code with platformIO. Any hints on this?

closed 2025-01-15 by Parsaabasi

lucasssvaz, 2025-01-15:
@s54b32julian Easiest way is to use Arduino as an IDF component. Follow the documentation [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html) to setup your project and then enable the ECC option found [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig-reference.html#config-spiram-ecc-enable)