espressif/arduino-esp32 #9285: FS memory: No such File or Directory

Kaistee93, 2024-02-24:
### Board

ESP32S3 Dev Module

### Device Description

 On Elecrow 7' HMI Display.
 With SD Reader, I2S DAC, Battery Support and some IO
 The board:
https://www.elecrow.com/esp32-display-7-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html

The Libraries:
https://drive.google.com/drive/folders/1StdAWhXHArhyw5HFpCY-xulWdZYrFAGz
Configuration Wiki
https://www.elecrow.com/wiki/index.php?title=Wizee-ESP32_7.0%27%27_Intelligent_Touch_Screen_Wi-Fi%26BLE_800*480_HMI_Display

### Hardware Configuration

IO 10 SD-Card CS
IO 11 SD-Card SPI MOSI / TP DIN
IO 12 SD-Card CLK / TP CLK
IO 13 SD-Card SPI MISO / TP OUT

IO 17 I2S SDIN
IO 18 I2S LRCLK
IO 42 I2S BCLK

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hello there.
I recently got myself an Elecrow 7 inch Touchscreen called Wizee-ESP32 WZ8048C070.
It's been hard to get anything working on that thing since it requires some older libraries and the Documentation isn't that great. After some days trying, i've finally been able to run a custom UI on the display. For the UI to work properly, i'd need to run some more code, which includes reading Username and Password from a csv file on a SD-Card  to authenticate a user and show settings for example....
Long story short:
After adding "#include <SD.h> the compiler's answer is that it can't find the <memory> file.
The <memory> should be an included header file Thingy according to my google research... but i didn't find anything on how to include it properly or how and where to find a replacement.

I already re-installed the IDE like five times, even tried different versions of the Arduino IDE (2.3.1 or 2.3.2 or 1.8.18), Deleted and re installed the libraries in different Versions. Changing the ESP32- Core version to 2.0.14 also doesnt help.

In the Sketch shown below (that's only the "ui_events.c" - File) the actual function code is commented out because i can't even get the thing to compile as soon as the SD and thereby FS libraries are included. 

Funny thing is:
When loading the A-7.0-TF Example (From the bottom of the Wiki Page) the sketch compiles without a problem while using precompiled libraries. 
`Library Wire has been declared precompiled:
Using precompiled library in c:\Users\kaist\Documents\Arduino\libraries\Wire\src\esp32s3
Library SPI has been declared precompiled:
Using precompiled library in c:\Users\kaist\Documents\Arduino\libraries\SPI\src\esp32s3
Library SD has been declared precompiled:
Using precompiled library in c:\Users\kaist\Documents\Arduino\libraries\SD\src\esp32s3
Sketch uses 282577 bytes (8%) of program storage space. Maximum is 3145728 bytes.
Global variables use 13368 bytes (4%) of dynamic memory, leaving 314312 bytes for local variables. Maximum is 327680 bytes.`

Your help on this would be aprreciated.

### Sketch

```cpp
// This file was generated by SquareLine Studio
// SquareLine Studio version: SquareLine Studio 1.3.4
// LVGL version: 8.3.3
// Project name: SquareLine_Project

#include "ui.h"
#include <SD.h>
#include <FS.h>

void handle_login(lv_event_t * e) {}
/*
{
  //if(event == LV_EVENT_CLICKED) {
    // Lesen der eingegebenen Benutzerdaten
    const char * username = lv_textarea_get_text(ui_UsernameBox);
    const char * password = lv_textarea_get_text(ui_PasswordBox);

    // Öffnen der CSV-Datei für das Lesen
    File file = SD.open("users.csv");
    if (file) {
      while (file.available()) {
        // Lesen einer Zeile aus der CSV-Datei
        String line = file.readStringUntil('\n');
        // Aufteilen der Zeile anhand des Trennzeichens (Komma)
        String id = getValue(line, ',', 0);
        String name = getValue(line, ',', 1);
        String pass = getValue(line, ',', 2);
        String rights = getValue(line, ',', 3);

        // Vergleich der eingegebenen Benutzerdaten mit den Daten aus der CSV-Datei
        if (strcmp(username, name.c_str()) == 0 && strcmp(password, pass.c_str()) == 0) {
          // Anmeldung erfolgreich
          Serial.println("Login erfolgreich!");
          Serial.println("Benutzername: " + name);
          login_active = true;

          lv_textarea_set_text(ui_Meldungen, "Anmeldung Erfolgreich. Willkommen ");
          lv_textarea_add_text(ui_Meldungen, name);

          // Überprüfen, ob es sich um einen Administrator handelt
          if (rights == "Admin") {
            admin_active = true;
          } else {
            admin_active = false;
          }
          
          break;
        }
        else {
          lv_textarea_set_text(ui_Meldungen, "Anmeldung fehlgeschlagen. Bitte versuchen Sie es erneut.");
          Serial.println("Fehler beim Anmelden");
          }
      }
      file.close(); // Schließen der CSV-Datei
    } else {
      Serial.println("Fehler beim Öffnen der Datei");
    }
  }
*/

void settings_handler(lv_event_t * e)
{
	// Your code here
}

void handle_dosage(lv_event_t * e)
{
	// Your code here
}

void handle_empty_hose(lv_event_t * e)
{
	// Your code here
}
```

### Debug Message

```plain
`In file included from c:\Users\kaist\Documents\Arduino\libraries\SD\src/SD.h:17,
                 from C:\Users\kaist\Documents\Arduino\sketches\HMI-7-OWN\ui_events.c:7:
C:\Users\kaist\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\FS\src/FS.h:24:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.`
```

### Other Steps to Reproduce

Windows 10 same Issue

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-02-24:
Using Arduino IDE, "ESP32S3 Dev Module" with Arduino Core 2.0.3 and 3.0.0-Alpha2 there is no compilation error with this code.

```cpp

#include <SD.h>
#include <FS.h>
#include <memory>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Output:
```
Successfully created esp32s3 image.
"C:\\Users\\espuser\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/gen_esp32part.exe" -q "C:\\Users\\espuser\\AppData\\Local\\Temp\\arduino_build_310143/partitions.csv" "C:\\Users\\espuser\\AppData\\Local\\Temp\\arduino_build_310143/BareMinimum.ino.partitions.bin"
Multiple libraries were found for "SD.h"
 Used: C:\Users\espuser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Using library SD at version 2.0.0 in folder: C:\Users\espuser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\SD 
Using library FS at version 2.0.0 in folder: C:\Users\espuser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\FS 
Using library SPI at version 2.0.0 in folder: C:\Users\espuser\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\SPI 
"C:\\Users\\espuser\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32s3-elf-size" -A "C:\\Users\\espuser\\AppData\\Local\\Temp\\arduino_build_310143/BareMinimum.ino.elf"
Sketch uses 210857 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 12816 bytes (3%) of dynamic memory, leaving 314864 bytes for local variables. Maximum is 327680 bytes.
```

SuGlider, 2024-02-24:
@Kaistee93 - From this issue, it seems to work with Arduino Core 2.0.11.
https://github.com/HASwitchPlate/openHASP/discussions/559

Kaistee93, 2024-02-25:
@SuGlider 
Thanks for trying this.
That's what i meant by funny behavoir when compiling the TF-Example Sketch. Because that Sketch compiles but the other one doesn't. Upgrading the Core from 2.0.3 to 2.0.11 didn't help either.
I Also removed the manufacturers provided SD, SPI and Wire Lib so the IDE can use the libs from the core.

[A-7.0-TF.zip](https://github.com/espressif/arduino-esp32/files/14396508/A-7.0-TF.zip)
This is the Demo TF Code provided by the manufacturer of the Board.

### Output:
```plain
Sketch uses 330385 bytes (10%) of program storage space. Maximum is 3145728 bytes.
Global variables use 19644 bytes (5%) of dynamic memory, leaving 308036 bytes for local variables. Maximum is 327680 bytes.
```

[HMI-7-OWN.zip](https://github.com/espressif/arduino-esp32/files/14396509/HMI-7-OWN.zip)
While this is basically the manufacturer's example code to run the UI on the Display but modified to run my own UI.
For handling the press of a login button i wanted to read a .csv file from included Micro Sd Card. Mentioned part of the code can be found in "ui_events.c"- Tab.

With  `#include<SD.h>`  and  `#include <FS.h>`

### Debug Message 1
```plain
In file included from C:\Users\kaist\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SD\src/SD.h:17,
                 from C:\Users\kaist\Documents\Arduino\sketches\HMI-7-OWN\ui_events.c:8:
C:\Users\kaist\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\FS\src/FS.h:24:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.
Multiple libraries were found for "SD.h"
  Used: C:\Users\kaist\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SD
  Not used: C:\Users\kaist\AppData\Local\Arduino15\libraries\SD
exit status 1
Compilation error: exit status 1
```

Putting `#Include <memory>`  beneath the `#include<SD.h>`  and  `#include <FS.h>` brings up this:
### Debug Message 2
```plain
C:\Users\kaist\Documents\Arduino\sketches\HMI-7-OWN\ui_events.c:10:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.
exit status 1
Compilation error: memory: No such file or directory
```

With core V 2.0.11 and the core's SPI and Wire libraries, excluding the SD and FS libraries (commented out), the Sketch won't compile whith this output:
### Debug Message 3
```plain
`Library LovyanGFX has been declared precompiled:
Using precompiled library in c:\Users\kaist\Documents\Arduino\libraries\LovyanGFX\src\esp32s3
c:/users/kaist/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kaist\AppData\Local\Temp\arduino\sketches\1AC098DA7C51C00269CC05F373DE3E9E\sketch\objs.a(HMI-7-OWN.ino.cpp.o):(.literal._ZN4LGFXC5Ev[_ZN4LGFXC5Ev]+0x8): undefined reference to `vtable for lgfx::v1::Bus_RGB'
c:/users/kaist/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kaist\AppData\Local\Temp\arduino\sketches\1AC098DA7C51C00269CC05F373DE3E9E\sketch\objs.a(HMI-7-OWN.ino.cpp.o):(.literal._ZN4LGFXC5Ev[_ZN4LGFXC5Ev]+0x30): undefined reference to `lgfx::v1::Panel_RGB::Panel_RGB()'
c:/users/kaist/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kaist\AppData\Local\Temp\arduino\sketches\1AC098DA7C51C00269CC05F373DE3E9E\sketch\objs.a(HMI-7-OWN.ino.cpp.o):(.literal._ZN4LGFXC5Ev[_ZN4LGFXC5Ev]+0x38): undefined reference to `lgfx::v1::Bus_RGB::config(lgfx::v1::Bus_RGB::config_t const&)'
c:/users/kaist/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kaist\AppData\Local\Temp\arduino\sketches\1AC098DA7C51C00269CC05F373DE3E9E\sketch\objs.a(HMI-7-OWN.ino.cpp.o):(.literal._ZN4LGFXC5Ev[_ZN4LGFXC5Ev]+0x3c): undefined reference to `lgfx::v1::Panel_RGB::~Panel_RGB()'
c:/users/kaist/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kaist\AppData\Local\Temp\arduino\sketches\1AC098DA7C51C00269CC05F373DE3E9E\sketch\objs.a(HMI-7-OWN.ino.cpp.o): in function `LGFX::LGFX()':
C:\Users\kaist\Documents\Arduino\sketches\HMI-7-OWN/HMI-7-OWN.ino:34: undefined reference to `lgfx::v1::Panel_RGB::Panel_RGB()'
c:/users/kaist/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kaist\Documents\Arduino\sketches\HMI-7-OWN/HMI-7-OWN.ino:84: undefined reference to `lgfx::v1::Bus_RGB::config(lgfx::v1::Bus_RGB::config_t const&)'
c:/users/kaist/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\kaist\Documents\Arduino\sketches\HMI-7-OWN/HMI-7-OWN.ino:34: undefined reference to `lgfx::v1::Panel_RGB::~Panel_RGB()'
[... cut ...]
GFX Library for Arduino           V 1.2.8
LovyanGFX                               V 1.1.9
TFT_eSPI                                   V 2.5.0
U8g2                                         V 2.33.15
lvgl                                            V 8.3.3

I may just be a simple hobbyist who doesn't have much clue of programming... but it seems like there's something wrong with my sketch, not the libraries or core so i hope to find help here even if it may be the wrong place to ask.
Have a great day y'all

lbernstone, 2024-02-26:
This question is more appropriate for esp32.com, and I suggest you move it over there.  This forum is for issues with the code in this repo.
You need to break your problem down into two pieces to make it more tractable. 1- Does the SD card work if you don't use any of the other devices on the board. 2- Can the lvgl driver be separated from the rest of the pre-compiled demo junk here so that it can actually be supported long-term.

SuGlider, 2024-02-26:
>C:\Users\kaist\Documents\Arduino\sketches\HMI-7-OWN\ui_events.c:10:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
compilation terminated.
exit status 1
Compilation error: memory: No such file or directory

`<memory>` is `C++` STL - it can only be used in `.cpp` os `.ino` files. **It can be declared a `.c` file.**

SuGlider, 2024-02-26:
@Kaistee93 - my 2 cents: 
Debug1 and 2: Change the files that include Arduino Libraries to `.cpp` instead of `.c`
Debug 3: LovyanGFX needs to be correctly configured for using 16 bits parallel display interface.
Debug 4: LVGL messages are just warnings.... it shall not prevent building the final firmware.

Kaistee93, 2024-02-29:
@SuGlider 
Well that's awkward but finally i've got some new bugs which is great :D
Thanks a lot for helping me out! From this point I'll move forward and seek help in ai assist.

SuGlider, 2024-03-01:
Good luck! I'll close this issue. Please feel free to bring new questions or issues whenever you need.

closed 2024-03-01 by SuGlider