espressif/arduino-esp32 #9286: RainMaker lib error when buid as ESP IDF component

dzungpv, 2024-02-25:
### Board

ESP32 Dev

### Device Description

Default kit

### Hardware Configuration

Default kit

### Version

v2.0.14

### IDE Name

ESP IDF 4.4.6

### Operating System

Macos 13, Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I try to build RainMaker example RMakerCustomAirCooler as IDF component project but it failed.

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/RainMaker/examples/RMakerCustomAirCooler/RMakerCustomAirCooler.ino
```

### Debug Message

```plain
components/arduino-esp32/libraries/RainMaker/src/RMakerQR.h:19:10: fatal error: qrcode.h: No such file or directory
 #include <qrcode.h>
          ^~~~~~~~~~
compilation terminated
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dzungpv, 2024-02-25:
I find out that the qrcode.h could not find in the path https://github.com/espressif/arduino-esp32/blob/2.0.14/tools/sdk/esp32/include/qrcode/include/qrcode.h and not include the .c file , I comment out code relate to QR and it work.

dzungpv, 2024-02-26:
But the provision not working it never pass.

VojtechBartoska, 2024-02-26:
cc @lucasssvaz

closed 2025-01-15 by Parsaabasi