espressif/arduino-esp32 #9288: ESP32-S3 USB with IN/OUT cannot get data

GitHobi, 2024-02-26:
### Board

ESP32-S3-DevKitC-1

### Device Description

ESP32-S3-DevKitC-1, just connected to the PC.

### Hardware Configuration

n/a

### Version

v2.0.14

### IDE Name

PlatformIO 

### Operating System

Linux

### Flash frequency

40Mhz (?)

### PSRAM enabled

yes

### Upload speed

n/a

### Description

I'm trying to implement a FIDO2 Authenticator device with a ESP32-S3.
I started by adapting the sample from libraries/USB/examples/CustomHIDDevice/CustomHIDDevice.ino.
I adjusted the report_descriptor to the FIDO2 specification - which basically works ... 

A web page supporting FIDO2 / WebAuthn starts commuicating with my implementation, but I'm not able to receive the data sent by the web page. As far as I could track it down, the USB stack tries to find my device (in tud_hid_set_report_cb) - but fails, as the incoming data uses report_id 0 - while the device is defined with report_id 6. 
I cannot use report_id 0, as the valid range for report_id is from 1-255.

Is this an issue? Or am I missing anything? How would one retrieve the incoming data?

### Sketch

```cpp
#include <Arduino.h>

#ifndef ARDUINO_USB_MODE
#error This ESP32 SoC has no Native USB interface
#elif ARDUINO_USB_MODE == 1
#warning This sketch should be used when USB is in OTG mode
void setup() {}
void loop() {}
#else
#include "USB.h"
#include "USBHID.h"
#include "USBHIDVendor.h"

USBHID HID;

#define HID_VENDOR_REPORT_SIZE 64

static const uint8_t report_descriptor[] = {HID_USAGE_PAGE_N(HID_USAGE_PAGE_FIDO, 2),
                                            HID_USAGE(HID_USAGE_FIDO_U2FHID),
                                            HID_COLLECTION(HID_COLLECTION_APPLICATION),
                                            HID_REPORT_ID(6)
                                                HID_USAGE(HID_USAGE_FIDO_DATA_IN), /* Input */
                                            HID_LOGICAL_MIN(0x00),
                                            HID_LOGICAL_MAX(0xff),
                                            HID_REPORT_SIZE(8),
                                            HID_REPORT_COUNT(HID_VENDOR_REPORT_SIZE),
                                            HID_INPUT(HID_DATA | HID_VARIABLE | HID_ABSOLUTE),
                                            HID_USAGE(HID_USAGE_FIDO_DATA_OUT), /* Output */
                                            HID_LOGICAL_MIN(0x00),
                                            HID_LOGICAL_MAX(0xff),
                                            HID_REPORT_SIZE(8),
                                            HID_REPORT_COUNT(HID_VENDOR_REPORT_SIZE),
                                            HID_OUTPUT(HID_DATA | HID_VARIABLE | HID_ABSOLUTE),
                                            HID_COLLECTION_END

};

class CustomHIDDevice : public USBHIDDevice
{
public:
  CustomHIDDevice(void)
  {
    ESP_LOGI(__FILENAME__, "", "");
    static bool initialized = false;
    if (!initialized)
    {
      initialized = true;
      HID.addDevice(this, sizeof(report_descriptor));
    }
  }

  void begin(void)
  {
    ESP_LOGI(__FILENAME__, "", "");
    HID.begin();
  }

  void _onOutput(uint8_t report_id, const uint8_t *buffer, uint16_t len)
  {
    ESP_LOGI(__FILENAME__, "", "");
  }

  uint16_t _onGetDescriptor(uint8_t *buffer)
  {
    ESP_LOGI(__FILENAME__, "", "");
    memcpy(buffer, report_descriptor, sizeof(report_descriptor));
    return sizeof(report_descriptor);
  }

  bool send(uint8_t *value)
  {
    ESP_LOGI(__FILENAME__, "", "");
    return HID.SendReport(0, value, 8);
  }
};

CustomHIDDevice myDevice;

static void vendorEventCallback(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
  ESP_LOGI(__FILENAME__, "event_id=%u", event_id);

}

void setup()
{
  Serial.begin(115200);
  delay(500);
  Serial.println();
  Serial.setDebugOutput(true);

  USB.onEvent(vendorEventCallback);
  USB.VID(0x0483);
  USB.PID(0xa2ca);
  USB.begin();

  HID.onEvent(vendorEventCallback);
  myDevice.begin();

  

  ESP_LOGI(__FILENAME__, "ready to go!", "");
}

void loop()
{
}
#endif /* ARDUINO_USB_MODE */
```

### Debug Message

```plain
6:58:42.872 > [   129][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface HID enabled
06:58:42.873 > [   129][I][main.cpp:44] CustomHIDDevice(): [main.cpp] 
06:58:42.883 > [   130][D][USBHID.cpp:65] tinyusb_enable_hid_device(): Device[0] len: 34
06:58:43.114 > [   373][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
06:58:43.623 > 
06:58:43.623 > [   879][D][esp32-hal-tinyusb.c:569] tinyusb_load_enabled_interfaces(): Load Done: if_num: 1, descr_len: 41, if_mask: 0x4
06:58:43.623 > [   880][I][main.cpp:55] begin(): [main.cpp] 
06:58:43.634 > [   883][I][main.cpp:104] setup(): [main.cpp] ready to go!
06:58:43.634 > [   886][I][main.cpp:83] vendorEventCallback(): [main.cpp] event_id=1
06:58:44.056 > [  1312][I][main.cpp:83] vendorEventCallback(): [main.cpp] event_id=0
06:58:44.056 > [  1313][V][USBHID.cpp:302] tud_hid_set_idle_cb(): instance: 0, idle_rate:0
06:58:44.067 > [  1314][I][main.cpp:83] vendorEventCallback(): [main.cpp] event_id=1
06:58:44.067 > [  1314][V][USBHID.cpp:278] tud_hid_descriptor_report_cb(): instance: 0
06:58:44.078 > [  1326][I][main.cpp:66] _onGetDescriptor(): [main.cpp] 
06:58:44.078 > [  1332][D][USBHID.cpp:229] tinyusb_load_enabled_hid_devices(): Loaded HID Desriptor with the following reports:
06:58:44.089 > [  1341][D][USBHID.cpp:238] tinyusb_load_enabled_hid_devices():   ID:   6, Type:   INPUT, Size: 64, Usage:   VENDOR
06:58:44.100 > [  1352][D][USBHID.cpp:238] tinyusb_load_enabled_hid_devices():   ID:   6, Type:  OUTPUT, Size: 64, Usage:   VENDOR
06:58:50.689 > [  7945][D][USBHID.cpp:133] tinyusb_on_set_output(): device not found 0
06:58:50.689 > [  7945][D][USBHID.cpp:133] tinyusb_on_set_output(): device not found 255
06:58:50.700 > [  7947][D][USBHID.cpp:331] tud_hid_set_report_cb(): instance: 0, report_id: 255, report_type: OUTPUT, bufsize: 63
```

### Other Steps to Reproduce

In the Debug Message you see that the USB Device was successfully created, using report_id 6.
If you then go to e.g. https://webauthn.io/ the page starts to communicate with the scetch - but it does not find the defined device, as the incoming data uses report_id 0 - while the device is defined with 1.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-02-26:
>As far as I can tell, it seems that the USB stack is not capable of supporting HID device having an IN AND OUT interface.

You can check the TinyUSB implementation. It is the USB stack for the S3.
https://github.com/hathach/tinyusb/tree/master/examples/device/hid_generic_inout

SuGlider, 2024-02-26:
@GitHobi - I've found the USB HID IN OUT declaration in the `HIDVendor` class.
Check the code in https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/src/USBHIDVendor.cpp#L70
`hid.addDevice(this, TUD_HID_REPORT_DESC_GENERIC_INOUT_FEATURE_LEN);`

Threfore, I think that it would be necessary to create a derivated class from `USBHIDVendor` that specializes the Descriptor that you need. Then you can try the example from `https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/HIDVendor/HIDVendor.ino` 

But, it sounds like ESP32-S3 USB can deal with HID IN/OUT.

GitHobi, 2024-02-27:
@SuGlider - That's actually what I'm doing in the sample code given to this bug issue. 
In the sample you refer to, a USBHIDVendor class is used, which simply derives from USBHIDDevice class. In the constructor of USBHidVendor the report_descriptor is attached to HID (addDevice). 
This 1:1 what I'm doing in my sample. 

If the USB Stack of the S3 is actually TinyUSB, I assume the problem is this code:
https://github.com/hathach/tinyusb/blob/52a9098b31d993d743ee31efe155b90c7ba5b21e/src/class/hid/hid_device.c#L412
`// Received report
  else if (ep_addr == p_hid->ep_out)
  {
    tud_hid_set_report_cb(instance, 0, HID_REPORT_TYPE_INVALID, p_hid->epout_buf, (uint16_t) xferred_bytes);
    TU_ASSERT(usbd_edpt_xfer(rhport, p_hid->ep_out, p_hid->epout_buf, sizeof(p_hid->epout_buf)));
  }`

As I understand the code, incoming data is propagated with report_id "HID_REPORT_TYPE_INVALID" - which is actually 0 ... that's exactly what I see in my debug outout, too.

> 06:58:50.689 > [  7945][D][USBHID.cpp:133] tinyusb_on_set_output(): device not found 0

This seems to be also the big difference to the original TinyUSB INOUT sample / implementation:
In their implementation they pass the data on - so "your" code can receive it. This can be seen in their sample, they implement their own "tud_hid_set_report_cb" function - so they receive "invalid" packages, too.
https://github.com/hathach/tinyusb/blob/52a9098b31d993d743ee31efe155b90c7ba5b21e/examples/device/hid_generic_inout/src/main.c#L147

In the S3 implementation you try to find the target device, using the report_id
https://github.com/espressif/arduino-esp32/blob/ccacb7e3d1dd0e58b309c83e5ebc2302ce97c7b2/libraries/USB/src/USBHID.cpp#L269C6-L269C27
but if you don't find the device, you bail out ...
https://github.com/espressif/arduino-esp32/blob/ccacb7e3d1dd0e58b309c83e5ebc2302ce97c7b2/libraries/USB/src/USBHID.cpp#L98
... and as 0 is not allowed, you can of course not find the device ...

So in theory the S3 might be able to handle HID IN/OUT  - but in practice I do not understand how you would implement it, without changing the logic in the library (USBHID.cpp)

me-no-dev, 2024-02-27:
it seems to me that this issue is not at all connected to our core or tinyusb. It seems that the host is expecting to communicate with report ID 0. If you comment out `HID_REPORT_ID(6)`, you will get the `_onOutput` callback with size 64 on report ID 0
```
[   245][D][esp32-hal-tinyusb.c:711] tinyusb_enable_interface2(): Interface HID enabled
[   253][D][USBHID.cpp:63] tinyusb_enable_hid_device(): Device[0] len: 32
[   649][D][esp32-hal-tinyusb.c:589] tinyusb_load_enabled_interfaces(): Load Done: if_num: 3, descr_len: 107, if_mask: 0x14
USB PLUGGED
[  1162][V][USBHID.cpp:227] tud_hid_descriptor_report_cb(): instance: 0
[  1169][D][USBHID.cpp:178] tinyusb_load_enabled_hid_devices(): Loaded HID Descriptor with the following reports:
[  1179][D][USBHID.cpp:181] tinyusb_load_enabled_hid_devices():   ID:   0, Type:   INPUT, Size: 64, Usage:   VENDOR
[  1189][D][USBHID.cpp:181] tinyusb_load_enabled_hid_devices():   ID:   0, Type:  OUTPUT, Size: 64, Usage:   VENDOR
onOutput[64]: report_id, 0
```

me-no-dev, 2024-02-27:
oh and in order to respond back, in `send()` you should `HID.SendReport(0, value, 64);`

GitHobi, 2024-02-27:
@me-no-dev Thanks for this information. Actually I tried this, but my Linux (Ubuntu)/Windows is not willing to accept the USB device with report_id=0 ... it complains about invalid report_id.

How did you tests this? Are there any special settings needed?

me-no-dev, 2024-02-27:
you must not define the report ID as 0, you just do not define it at all. Just comment out `HID_REPORT_ID(6)` as I said, not set it to `HID_REPORT_ID(0)`

GitHobi, 2024-02-27:
@me-no-dev  ... you made by day! Removing the HID_REPORT_ID(0) actually did the trick.

closed 2024-02-27 by GitHobi