espressif/arduino-esp32 #9292: Add missing LEDC light sleep requirements to documentation

zackees, 2024-02-27:
### Board

ESP32*

### Device Description

This is a generic bug report and applies to all devices with the ESP32 that can do light sleep for the LEDC

### Hardware Configuration

I'm personally using the XIAO ESP32C3, but it applies to other devices as well.

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

115200

### Description

This page in the documentation for LEDC does not contain all the required information to allow the LEDC device to continue functioning in light sleep: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html

It's currently missing the following:

```C++
ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC8M, ESP_PD_OPTION_ON));
ESP_ERROR_CHECK(gpio_sleep_sel_dis(PIN_LEDC)); // Needed for light sleep.
```

### Sketch

```cpp
I've posted the minimal code sample [here](https://github.com/espressif/arduino-esp32/issues/3534#issuecomment-1965779153)
```

### Debug Message

```plain
There is no debug message. This is a documentation request.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-02-27:
Hi @zackees, you are using LEDC driver directly from ESP-IDF, so I am not sure how it's related to the ESP32 Arduino core? Is it just about missing info in docs?

zackees, 2024-02-27:
Oh, this is actually ESP-IDF. I didn't realize this was the arduino-esp32 repo. Should I refile the bug in the proper repo?

P-R-O-C-H-Y, 2024-02-27:
@igrr @Alvin1Zhang Can we transfer this issue to the ESP-IDF repository? Thanks

zackees, 2024-02-27:
I've gone ahead and moved this to the esp-idf repo.

https://github.com/espressif/esp-idf/issues/13278

closed 2024-02-27 by zackees