espressif/arduino-esp32 #9293: Native IDF websocket reconnecting problem with TLS

hitecSmartHome, 2024-02-27:
### Board

ESP32-Wrover

### Device Description

Nothing fancy

### Hardware Configuration

ETH,SPI

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Weird behaviour when the websocket client disconnects and wants to reconnect.
`TLS error 0x7F00`

I got a system message which says that it can't allocate 16717 byte but i got plenty of ram.
**36852** byte contignous memory and **65844** free. In external ram i got a bunch of ram.

```cpp
[SYSTEM] - [HEAP][1]
Free:   65844
Min:    65844
Max:    65844
Contig: 36852
Diff:   18728

[SYSTEM] - [EXT][1]
Free:   3992815
Min:    3992815
Max:    3992815
Contig: 3932148
Diff:   12684
```

The websocket connects fine at first and there can be a couple of disconnections before this happens. Let's say like 4 disconnections in a row and it can not reconnect anymore, just floods this debug message. Always with the same message and always with this `16717 bytes` allocation.

### Sketch

```cpp
esp_websocket_client_handle_t client;
esp_websocket_client_config_t config = {};

config.cert_pem = hshServer.server_pem.c_str();
config.headers = hshKeyHeader;
config.uri = "wss://testNodejsServer.onrender.com";
config.port = 443;
config.path = "/testSocket";
config.transport = WEBSOCKET_TRANSPORT_OVER_SSL;
config.task_stack = 6000;
config.keep_alive_enable = true;

client = esp_websocket_client_init(&config);
esp_websocket_register_events(
    client,
    WEBSOCKET_EVENT_ANY,
    reinterpret_cast<esp_event_handler_t>(eventHandler),
    static_cast<void*>(this)
);
esp_websocket_client_start(client);
```

### Debug Message

```plain
[HsH_Server] (websocket) - Disconnected

[SYSTEM] - [HEAP][1]
Free:   65804
Min:    65804
Max:    65804
Contig: 36852
Diff:   72

[SYSTEM] - [EXT][1]
Free:   3992591
Min:    3992591
Max:    3992591
Contig: 3932148
Diff:   1776

[SYSTEM] - heap_caps_calloc was called but failed to allocate 16717 bytes with 0x804 capabilities.
E (9173226) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (9173227) esp-tls: create_ssl_handle failed
E (9173227) esp-tls: Failed to open new connection
E (9173229) TRANSPORT_BASE: Failed to open a new connection
E (9173236) TRANSPORT_WS: Error connecting to host hitecsmarthome.onrender.com:443
E (9173242) WEBSOCKET_CLIENT: Error transport connect
[HsH_Server] (websocket) - Disconnected
```

### Other Steps to Reproduce

Just fire up the native ESP IDF websocket client example and connect to a wss endpoint and trigger some disconnections.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-02-28:
caps 0x804 = MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT
You are out of 8bit memory.

hitecSmartHome, 2024-02-28:
Oh god... is there a workaround?
Why doesnt websocket client lib frees ram on disconnect so it is able to reconnect again?

closed 2024-08-07 by hitecSmartHome