espressif/arduino-esp32 #9297: esp32 wroom arduino ide/ beacon

meymi19666, 2024-02-27:
### Related area

I programmed the esp32 wroom in the Arduino IDE, but it does not appear on the mobile phone except for the advertising data and the biandroid application.

### Hardware specification

esp32wroom

### Is your feature request related to a problem?

I programmed the esp32 wroom in the Arduino IDE, but it does not appear on the mobile phone except for the advertising data and the biandroid application.

### Describe the solution you'd like

Does esp32 wroom card not support beacon? If it is supported, what should I do?

### Describe alternatives you've considered

Is firmware enough?

### Additional context

[image]
[image]
android ekran görüntüsü.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2024-02-27:
Hi @meymi19666, can you please post the sketch + debug output?

meymi19666, 2024-02-29:
First of all, thank you for your reply, I was a little annoyed by the late
reply.
In the Arduino IDE, there is no error report, I cannot see the problem
devices of nRF Connect, but I cannot see the advertising broadcast.
I want to do it, I want to open the menu of the cafe, ESP32's advertising
to all mobiles in a cafe, whatever your contribution to me was, it was
enough for you to answer, parting with your work, success.
I am a bit old and have eyes on the ground, I am sorry if there is anything
missing regarding the correspondence rules in the forums.
Mehmet
attachment: beacon installed on esp32 without any problems

Jan Procházka ***@***.***>, 27 Şub 2024 Sal, 23:49 tarihinde
şunu yazdı:

P-R-O-C-H-Y, 2024-02-29:
Hello, as it seems that there is no bug in the code on our side, probably you have some issue in your sketch. I can't tell much more without having more informations about your code and logs. So I am closing this issue.
Feel free to ask on our [Gitter channel](https://gitter.im/espressif/arduino-esp32), where you can get help from the community and maybe someone was facing the same problem as you in the past and can share some tips.

closed 2024-02-29 by P-R-O-C-H-Y

meymi19666, 2024-02-29:
I looked at the translation of the message I wrote to you, the words were
written wrong, I'm sorry. I am grateful for your answers and contributions.
Thank you very much. Good luck to all my friends. Best regards.
Mehmet

Jan Procházka ***@***.***>, 29 Şub 2024 Per, 16:00 tarihinde
şunu yazdı: