espressif/arduino-esp32 #9300: Pulse to low when pin mode set to OUTPUT_OPEN_DRAIN

davidk88, 2024-02-28:
### Board

ESP32-S3 WROOM-1 M0N16R8

### Device Description

AI-S3 devkit

### Hardware Configuration

none

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

First time after external reset, a short pulse to low is generated when pin mode set to output open drain. I have external pull-up. Pulse is not generated if device is reset with ESP.restart().

### Sketch

```cpp
pinMode(1, OUTPUT_OPEN_DRAIN);
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

davidk88, 2024-02-29:
Hello,

my problem is that I need to have reset line connected to another micro. I don't want to be reset when esp is reset. When gpio is set it should stay in hi-z by default.

Here is a waveform:
[image]

lbernstone, 2024-02-29:
Do you have the same drop if you call the config directly?
```
  int mypin = 2;
  gpio_config_t conf = {
      .pin_bit_mask = (1ULL<<mypin),
      .mode = (gpio_mode_t)(GPIO_MODE_OUTPUT | GPIO_MODE_DEF_OD),
      .pull_up_en = GPIO_PULLUP_DISABLE,
      .pull_down_en = GPIO_PULLDOWN_DISABLE,
      .intr_type = GPIO_INTR_DISABLE
  };
  gpio_config(&conf);
```

davidk88, 2024-02-29:
With your code is like this:
[image]

1: reset button pressed
2: reset button released (short pulse to low)
3: gpio initialized with your code

seems that default value for output open drain is low?

lbernstone, 2024-02-29:
Or my code is bad :)
In any case, it shows that the issue is at the hardware level, either physical init or the bootloader.  I'd open an issue at [ESP-IDF](https://github.com/espressif/esp-idf/issues) and see if they have an answer for you.  It's not something that could be fixed through arduino-esp32.

SuGlider, 2024-03-18:
External reset like pressing RESET button?
Not sure if this can be changed.

Please give it a try with [gpio_hold_en(gpio_num_t gpio_num)](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/gpio.html#_CPPv412gpio_hold_en10gpio_num_t)  and [gpio_hold_dis(gpio_num_t gpio_num)](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/gpio.html#_CPPv413gpio_hold_dis10gpio_num_t)

closed 2025-01-15 by Parsaabasi