espressif/arduino-esp32 #9308: Instant crash with BLE-MIDI examples with 3.0.0.alpha3

metaview, 2024-02-28:
### Board

ESP32-S3 Dev Board

### Device Description

This board: https://www.amazon.de/dp/B0CQNB74R9?psc=1&ref=ppx_yo2ov_dt_b_product_details

### Hardware Configuration

Nothing connected, just the computer on USB

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

All BLE examples I've tried lead into a reset loop. They work fine with 2.0.11 but not with 3.0.0.alpha3 anymore.

### Sketch

```cpp
None of the Midi BLE examples worked, regardless of library I used. For instance try this with 3.0.0.alpha3 and 2.0.11 and see the difference:
MidiBle_Client from https://github.com/lathoub/Arduino-BLE-MIDI
```

### Debug Message

```plain
ELF file SHA256: c75e3ce9cd42ea34

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377429
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
[   145][D][esp32-hal-tinyusb.c:711] tinyusb_enable_interface2(): Interface CDC enabled
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 2
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-02-29:
please show a log with the reboot loop and [decode any backtrace](https://github.com/me-no-dev/EspExceptionDecoder)

metaview, 2024-03-01:
Now, for making things easier, I used the BLE Client example from Espressif:
Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\examples\Client
I set "Core debug level" to debug and put it on my ESP32S3 device and got the output below. The Exception Decoder just says:

>PC: 0x40057067
>EXCVADDR: 0x00000000
>
>Decoding stack results

Log output:

ELF file SHA256: 6ce855741ef63b4d

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376949
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
[   200][D][esp32-hal-tinyusb.c:711] tinyusb_enable_interface2(): Interface CDC enabled
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 2
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   361468 B ( 353.0 KB)
  Free Bytes        :   326496 B ( 318.8 KB)
  Allocated Bytes   :    29876 B (  29.2 KB)
  Minimum Free Bytes:   321448 B ( 313.9 KB)
  Largest Free Block:   294900 B ( 288.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Mar  1 2024 12:30:31
[... 66 lines cut ...]
[  2272][D][BLEClient.cpp:180] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
[  2283][D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
[  2295][D][BLEClient.cpp:180] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
[  2306][D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
[  2317][D][BLEClient.cpp:180] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_RES_EVT
[  2328][D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_CMPL_EVT
[  2339][D][BLEClient.cpp:180] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_CMPL_EVT
[  2351][D][BLEDevice.cpp:150] gattClientEventHan - Found oudr serviceler(): gattClientEventHandle
r [esp_gatt_if: 3] ... ESP_GATTC_C[G M3U1E[T][BLERemoteService.cpp:196] retrieveCharacteristics(): Found a characteristic: Handle: 114, UUID: 7772e5db-3868-4112-a1a9-f2669d106bf3
[  2375][D][BLEClient.cpp:180] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_CFG_MTU_EVT
[  2375][D][BLERemoteCharacteristic.cpp:298] retrieveDescriptors(): Found a descriptor: Handle: 115, UUID: 00002902-0000-1000-8000-00805f9b34fb
 - Found our characteristic
[  2419][D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_CHAR_EVT
[  2430][D][BLEClient.cpp:180] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_CHAR_EVT
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40057067  PS      : 0x00060f30  A0      : 0x8200d6f7  A1      : 0x3fca6650  
A2      : 0x3fcb0008  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3fcaab20  
A6      : 0x00000005  A7      : 0x00000020  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x3fcb0478  A12     : 0x00000000  A13     : 0x3fca8780  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffff9  

Backtrace: 0x40057064:0x3fca6650 |<-CORRUPTED

me-no-dev, 2024-03-01:
@metaview please decode the exception anyways. the PC address should give a hint. We will also need a minimal example sketch to reproduce the issue.

metaview, 2024-03-01:
The decoder doen't show any more infos. But I got the example (Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\BLE\examples\Client) working after removing these lines:

>    // Read the value of the characteristic.
>    if(pRemoteCharacteristic->canRead()) {
>      String value = pRemoteCharacteristic->readValue();
>      Serial.print("The characteristic value was: ");
>      Serial.println(value.c_str());
>    }

Now I can receive Midi data from my keyboard. I will work on it a bit now.
Next week I'll try the BLE Midi libraries again. If I get any more exceptions I will report again.

lucasssvaz, 2024-03-04:
If you find anything else, please reopen this issue

closed 2024-03-04 by lucasssvaz