espressif/arduino-esp32 #9312: Need it to support esp32-C2 core

boschLining, 2024-03-01:
### Related area

Board support

### Hardware specification

ESP32-8684

### Is your feature request related to a problem?

Now this version can not support ESP32-C2 board, do you have any solutions.

### Describe the solution you'd like

Please provide a solution to support esp32-C2 on Arduino, thanks.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2024-03-01:
See https://github.com/espressif/arduino-esp32/issues/6701

Arduino Core branch master does support esp32-c2, you have to compile the needed libs yourself or use the GH actions fragments for the c2 from the CI of the espressif Arduino lib builder https://github.com/espressif/esp32-arduino-lib-builder/actions/workflows/push.yml

VojtechBartoska, 2024-03-11:
as mentioned above, C2 is supported version 3.0.0 but you need to build libs on your own.

There is more explanation in mentioned issue.

I am closing this ticket as duplicate.

closed 2024-03-11 by VojtechBartoska