espressif/arduino-esp32 #9314: WebServer.h/cpp don't seem to represent the actual implementation in the library

wishwamithra, 2024-03-01:
### Board

The repo itself

### Device Description

No need of a device description. I am questioning the content of this repo.

### Hardware Configuration

No need of hardware configuration. I am questioning the content of this repo.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

9600

### Description

I don't think the repo shows the true implementation of the library, especially w.r.t. WebServer (https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.h). Here is the proof:

1. I see only two references to WebServer::handleClient() function in this repo.  First, in the declaration in WebServer.h file, and then in the WebServer.cpp file. Where is this function used (get called)?
2. Check what the uses of variables _firstHandler, _lastHandler and _currentHandler are. _firstHandler and _lastHandler are set, but  never used. In contrast, _currentHandler is used, but never set.

### Sketch

```cpp
No need of a sketch. I am questioning the content of this repo.
```

### Debug Message

```plain
No need of a debug messaage. I am questioning the content of this repo.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JAndrassy, 2024-03-01:
https://github.com/espressif/arduino-esp32/blob/ebca505fef55a2978db7607ced21a2700ef340e1/libraries/WebServer/examples/HelloServer/HelloServer.ino#L72

wishwamithra, 2024-03-01:
> https://github.com/espressif/arduino-esp32/blob/ebca505fef55a2978db7607ced21a2700ef340e1/libraries/WebServer/examples/HelloServer/HelloServer.ino#L72
> 
> https://github.com/espressif/arduino-esp32/blob/ebca505fef55a2978db7607ced21a2700ef340e1/libraries/WebServer/src/WebServer.cpp#L356-L365

Thank you. So, for the first question, the answer is that the application should call `handleClient()`. But for the second question, where is _firstHandler and _lastHandler used. I see the places where its values are set, but can't see a place they are used for anything. Even in your answer, it shows where the values are set. What is the use of these variables. And as for _currentHandler, it is only used, but not set in any place of the code. Given these variables are protected variables, an application cannot use them directly either, when they create WebServer object. Appreciate if you could shed some light on this too.

JAndrassy, 2024-03-01:
_addRequestHandler uses _lastHandler
and this is missing in my previous comment
https://github.com/espressif/arduino-esp32/blob/ebca505fef55a2978db7607ced21a2700ef340e1/libraries/WebServer/src/Parsing.cpp#L126