espressif/arduino-esp32 #9320: Issue with documentation

kov-serg, 2024-03-02:
### Board

any

### Device Description

any

### Hardware Configuration

any

### Version

v2.0.14

### IDE Name

any

### Operating System

ubuntu 22

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

115200

### Description

Code lines are overlapping page boundaries:

[image]

### Sketch

```cpp
// any
```

### Debug Message

```plain
// none
```

### Other Steps to Reproduce

git clone --depth=1 https://github.com/espressif/arduino-esp32.git
cd arduino-esp32/docs/
build-docs -l en -bs latex

[arduino-esp32doc-2.0.14.pdf](https://github.com/espressif/arduino-esp32/files/14469449/arduino-esp32doc-2.0.14.pdf)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-02:
It looks weird in the [web format](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/ble.html), but does not mess up the page boundaries. Seems like a latex issue.

kov-serg, 2024-03-02:
No this problem in documentation formatting. It should be more smart or simply fix source codes to prevent formatting issues. 
[image]

lucasssvaz, 2024-03-05:
There are mixed spaces and tabs in the indentation. I'll open a PR to fix it.

PR #9343 (2024-03-05): Fix indentation for UART BLE example
## Description of Change
Fix UART example indentation that was breaking the documentation.

## Related links
Closes #9320
closed 2024-03-05 by me-no-dev