espressif/arduino-esp32 #9321: Preprocessor Macros lead to compiler errors

insunaa, 2024-03-02:
### Board

LilyGo TTGO LoRa32

### Device Description

Just a TTGO

### Hardware Configuration

Not relevant

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Linux

### Flash frequency

not relevant

### PSRAM enabled

yes

### Upload speed

not relevant

### Description

In `cores/esp32/binary.h`
```cpp
#define B1 1
```
Causes compilation issues when Arduino.h is included in any project where the literal `B1` already means something.

Example: using&including fmtlib/fmt where it's used as a template typename
```cpp
template <bool B = false> constexpr auto count() -> size_t { return B ? 1 : 0; }
template <bool B1, bool B2, bool... Tail> constexpr auto count() -> size_t {
  return (B1 ? 1 : 0) + count<B2, Tail...>();
}
```

### Sketch

```cpp
#inlcude <fmt/core.h>
#include "Arduino.h"
```

### Debug Message

```plain
~/.platformio/packages/framework-arduinoespressif32/cores/esp32/binary.h:31:12: error: expected '>' before numeric constant
.pio/libdeps/TTGO-MQTT-lora32-v1/fmt/include/fmt/core.h:1118:58: error: redefinition of 'template<bool <anonymous> > constexpr size_t fmt::v8::detail::count()'
```
```

### Other Steps to Reproduce

Could this be handled by an enum instead?
```cpp
namespace arduino {
    enum Binary : uint32_t {
        B0 = 0,
        B00 = 0,
        B000 = 0,
        B0000 = 0,
        B00000 = 0,
        .
        .
        .
    };
};
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-03-02:
For what is this file there in general?

insunaa, 2024-03-02:
I have absolutely no idea, but it's there in the reference Arduino API, too. Tho they use namespaces and enums appropriately:
https://github.com/arduino/ArduinoCore-API/blob/master/api/Binary.h

Jason2866, 2024-03-02:
Why not open a PR and replace the current with the one you linked?

Jason2866, 2024-03-02:
There is a reference from Arduino. This is an Arduino implementation. So the change is getting closer to the API :-)

PR #9323 (2024-03-05): Update binary.h
uses binary.h from https://github.com/arduino/ArduinoCore-API/blob/master/api/Binary.h
fixes #9321
closed 2024-03-05 by me-no-dev