espressif/arduino-esp32 #9324: ESP32-S2 DAC2 Requires Load Resistor?

biomurph, 2024-03-02:
### Board

Adafruit ESP32-S2 Feather

### Device Description

na

### Hardware Configuration

Oscilloscope

### Version

v2.0.14

### IDE Name

Arduino IDE 2.3.2

### Operating System

Mac

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

DAC2 output is not 0 to 3.3V (Vdd). The lower half of the byte value (0 to 128 or so) is not registering the correct output voltage. I have the DAC1 connected to the Yellow Channel and DAC2 connected to the Pink Channel in the photo here. You can see the DAC1 trace makes a nice sawtooth /| faithfully presenting the correct voltage on the pin. DAC2 on the other hand is trying to make a sawtooth |\ but has distorted values that range from 0 to about mid range (128). I confirm that when I make DAC2 draw a sawtooth /| style it has the same Voltage value correlated to the byte value.

Why is this happening? Is this expected? Is there a preferred circuit to connect to the DAC outputs? 
I tried to search the forum, but the search is broken, at least for me.

I plan to buffer the DAC output signals with an OpAmp.

[image]

### Sketch

```cpp
void setup() {
  pinMode(DAC1,OUTPUT); dacWrite(DAC1,127);
  pinMode(DAC2,OUTPUT); dacWrite(DAC2,127);

  Serial.begin(230400);
  while(!Serial){}
  delay(1000);
  Serial.println("DAC tester");
}

void loop() {

  for(uint8_t b=0; b<255; b++){
    dacWrite(DAC2,256-b);
    dacWrite(DAC1,b);
    delay(5);
  }

}
```

### Debug Message

```plain
na
```

### Other Steps to Reproduce

In order to get a better, more complete DAC2 output, I added a load resistor to GND. Here are examples of various R Load values.

`R Load = 100R`
[image]

`R Load = 220R`
[image]

`R Load = 330R`
[image]

`R Load = 470R`
[image]

You can see that the range expands with increase of R value until there is some weird foldover of the values at 470R.
In the next image, a 390R value is used and the distortion appears equally at either end of the range.

`R Load = 390R`
[image]

I first saw this distortion when I was trying to produce two sine waves at different 
frequencies using a lookup table of pre-calculated values. Here's the initial scope view with no load resistor on DAC2 (Pink Channel).
[image]

With a 390R load resistor connected from DAC2 to GND, The output of the sine values is better, but there is some strange clipping at the range limits.
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-02:
This has been dacWrite has been rewritten in 3.0.0 using the IDF hal.  Can you please test with the [development version](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide)?

biomurph, 2024-03-03:
I installed the development version 3.0.0 alpha3 using the Arduino board manager.
I get the exact same distortion of DAC2 with no load resistor as shown in oscilloscope trace in previous post.
Now, however, the optimal R Load is 470R, and while it appears to be able to produce 0V, it still won't go all the way up to 3.3V. Here is a pic of the same code running with a 470R Load Resistor.

`470R Load with v3.0.0 alpha3 board files`
[image]

biomurph, 2024-03-04:
Can anyone replicate this, please?

P-R-O-C-H-Y, 2024-03-04:
Let me test :)

closed 2024-03-04 by P-R-O-C-H-Y

biomurph, 2024-03-04:
Thanks @P-R-O-C-H-Y 
I will take it up with the kids at Adafruit.

biomurph, 2024-03-07:
@P-R-O-C-H-Y 
I am comenting here, because this issue thread has all of the content in regard to the problem that I found.

I acquired an ESP32-S2 Soala board, and found the same issue with DAC2. The exact same distortion of the lower half of the signal range values were distorted in the same way as my scope pictures above. The signal trace reacted the same way to the application of a load resistor.

I found the schematics of the Soala board here:
https://dl.espressif.com/dl/schematics/ESP32-S2-SAOLA-1_V1.1_schematics.pdf

It clearly shows that on the Soala GPIO-18 is used to drive the neopixel on the board. There is a 0 ohm jumper, R40, from GPIO-18 to the neopixel. GPIO-18 also has a 10K pull-up resistor, R42. I removed R40 and R42 from the Soala, and now I get the proper wave shape on DAC2. That is good, but how were you able to get a clean signal on DAC2 of your Soala board with the 10Kpullup and neopixel hanging off of it?

Further, the Adafruit Feather ESP32-S2 board uses the Espressif ESP32-S2 Mini 1 radio module.
There is nothing on the Adafruit designed circuit that is connected to DAC1 or DAC2. Here is a link to their schematic:
https://learn.adafruit.com/assets/110822
NOTE that the schematic says "ESP32-S3" but Adafruit says that the board layout between the S2 and S3 feather boards are exactly similar. Adafruit does not have any circuitry connected to the DAC1 or DAC2, however, I can measure a 10K resistance between DAC2 and 3V3. DAC1 resistance to 3V3 is in the tens of magaohms. 

It does look like the radio module ESP32-S2 Mini 1 has a 10K pullup resistor under the can. Can that be verified? Is there a schematic of the radio module?

ColonelThirtyTwo, 2026-08-27:
I just ran into this too, again on an Adafruit ESP32-S2 feather that does not have anything except a header pin connected to the DAC2 pin GPIO18. Setting the pin to input with no pull reads 3.26v, and with the internal pulldown reads 2.48V. The DAC can get it down to ~1.5v without a load.

The ESP32-S2 Mini 1 module that it uses does show a pullup in the schematic:

[image]

And while it's labeled as NC, it being populated would explain what I'm seeing.

I guess this is more of a hardware issue though.

viric, 2026-08-28:
I think it's a mistake in the ESP32 *reference design* so every board manufacturer follows it.