espressif/arduino-esp32 #9326: Enable "USB CDC On Boot" for S2 not working with ESP IDF component project

dzungpv, 2024-03-03:
### Board

Lolin S2 Mini

### Device Description

Default

### Hardware Configuration

Default

### Version

v2.0.14

### IDE Name

ESP IDF 4.4.6

### Operating System

Macos 13, Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I enable USB CDC On Boot for a project building as ESP IDF component but it is not working.
The steps to enable the project here: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

The way I use to enable it mention here: https://esp32.com/viewtopic.php?t=30926#p115761.
I have tried both with release 2.0.14 and latest 2.x branch with the same result .
I add a definition in root project CMakeLists.txt:
```
add_definitions(
-DARDUINO_USB_MODE
-DARDUINO_USB_CDC_ON_BOOT
)
```
It can build and flash to the chip, but no output console like when I build the sketch with Arduino with option : "USB CDC On Boot: Enabled"

### Sketch

```cpp
#include <Arduino.h>

static void test_taskfunc(void * p)
{
    uint32_t tasknum = (uint32_t)p;

    while (true) {
        log_i("This is task instance %u running...", tasknum);
        
        delay(10 * random(1, 10));
    }
}

void setup()
{
    Serial0.begin(115200);
    Serial0.setDebugOutput(true);
    delay(2000);
    Serial0.println("\n==============\nStarting...\n");
    log_e("Starting with LOGs.");
    Serial0.println("==============\n\n");
    delay(750);

    for (auto i = 0; i < 5; i++) {
        TaskHandle_t th;

        // Created tasks all need to have different priorities in order to trigger lockup
        UBaseType_t taskPrio = 1 + 2 + i;

        if (pdPASS == xTaskCreate(test_taskfunc, "test_taskfunc", 4096, (void *)(i + 1), taskPrio, &th)) {
            log_i("Task instance %u created with priority %u", i+1, 3 + i);
        } else {
            log_e("Task instance %u creation FAILED", i+1);
        }
    }
}

void loop()
{
    log_i("This is the main loop running...");

    delay(1000);
}
```

### Debug Message

```plain
No output because it is not working
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-03-04:
@dzungpv - Please use it this way:
ARDUINO_USB_MODE=0 ==>for USB OTG (S2 and S3)
ARDUINO_USB_MODE=1 ==> for USB HW JTAG Serial CDC (for S3, C3, C6 and H2)
ARDUINO_USB_CDC_ON_BOOT=1 ==> it will make `Serial` to be the USB CDC port
ARDUINO_USB_CDC_ON_BOOT=0 ==> it will make `Serial` to be the UART0 port

```
add_definitions(
-DARDUINO_USB_MODE=0
-DARDUINO_USB_CDC_ON_BOOT=1
)
```

SuGlider, 2024-03-04:
Another detail, I only see the example sketch using `Serial0` which is for UART0, not USB CDC.
`Serial` will be the USB CDC when `USB CDC On Boot=1".

dzungpv, 2024-03-05:
> @dzungpv - Please use it this way: ARDUINO_USB_MODE=0 ==>for USB OTG (S2 and S3) ARDUINO_USB_MODE=1 ==> for USB HW JTAG Serial CDC (for S3, C3, C6 and H2) ARDUINO_USB_CDC_ON_BOOT=1 ==> it will make [...]

It is not working, could not build , it show error:

```
/Users/vietdzung/xxx//components/arduino-esp32/cores/esp32/main.cpp: In function 'void app_main()':
/Users/vietdzung/xxx/components/arduino-esp32/cores/esp32/main.cpp:58:5: error: 'Serial' was not declared in this scope
     Serial.begin();
     ^~~~~~
/Users/vietdzung/xxx/arduino-esp32/cores/esp32/main.cpp:58:5: note: suggested alternative: 'Serial1'
     Serial.begin();
     ^~~~~~
     Serial1
[1118/1252] Building CXX object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/WString.cpp.obj
ninja: build stopped: subcommand failed.
```

> Another detail, I only see the example sketch using `Serial0` which is for UART0, not USB CDC. `Serial` will be the USB CDC when `USB CDC On Boot=1".

There is no Serial when USB CDC ON Boot define: https://github.com/espressif/arduino-esp32/blob/44da992b774f76777bb2e931dd76cfcf12b9fe70/cores/esp32/HardwareSerial.cpp#L101

You can create a test project with component and test with version 2.x, idf 4.4.6 and you will see the issue.

SuGlider, 2024-03-05:
>There is no Serial when USB CDC ON Boot define:

It is defined in https://github.com/espressif/arduino-esp32/blob/44da992b774f76777bb2e931dd76cfcf12b9fe70/cores/esp32/USBCDC.cpp#L449-L452

But it depends on 
https://github.com/espressif/arduino-esp32/blob/44da992b774f76777bb2e931dd76cfcf12b9fe70/cores/esp32/USB.cpp#L16

Please use the sdkconfig file that has all necessary Arduino Settings for the ESP32-S2
https://github.com/espressif/arduino-esp32/blob/44da992b774f76777bb2e931dd76cfcf12b9fe70/tools/sdk/esp32s2/sdkconfig

This will enable TinyUSB necessary for USB CDC.

https://github.com/espressif/arduino-esp32/blob/44da992b774f76777bb2e931dd76cfcf12b9fe70/tools/sdk/esp32s2/sdkconfig#L259C1-L272C1

```
#
# Arduino TinyUSB
#
CONFIG_TINYUSB_ENABLED=y

#
# Serial (CDC) driver
#
CONFIG_TINYUSB_CDC_ENABLED=y
CONFIG_TINYUSB_DESC_CDC_STRING="Espressif CDC Device"
CONFIG_TINYUSB_CDC_RX_BUFSIZE=64
CONFIG_TINYUSB_CDC_TX_BUFSIZE=64
# end of Serial (CDC) driver
```

dzungpv, 2024-03-05:
> It is defined in
>
> https://github.com/espressif/arduino-esp32/blob/44da992b774f76777bb2e931dd76cfcf12b9fe70/cores/esp32/USBCDC.cpp#L449-L452
>
> But it depends on [...]

It still not build.
The following steps I create a sample project:
1, Use hello world arduino ESP IDF project
2, Replace main.cpp with sketch in first report
3, Use SDK Config like you mention
4, Add:
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)
add_definitions(
 -DARDUINO_USB_MODE=0
 -DARDUINO_USB_CDC_ON_BOOT=1
)
```
to the root CMakeLists.txt 
5, Checkout 2.x branch: not working, exact commit you comment above: not working, checkout 2.0.14 release: not working. Same error in the core/main.cpp: 
```
arduino-esp32/cores/esp32/main.cpp:58:5: error: 'Serial' was not declared in this scope
```

Do I missed somethings? the IDF version I use 4.4.6 on the MacOS Soloma 14.2

SuGlider, 2024-03-07:
@dzungpv - Please edit the  CMaketLists.txt from **Arduino Component folder**, instead of  using `add_definitions(...)`:
Find this entry `target_compile_options(${COMPONENT_TARGET} PUBLIC` in the file and add all necessary definitions there:
```
target_compile_options(${COMPONENT_TARGET} PUBLIC
    -DARDUINO=10812
    -DARDUINO_${idf_target_caps}_DEV
    -DARDUINO_ARCH_ESP32
    -DARDUINO_BOARD="ESP32S2_DEV"                           <<<<<<=== Board Name (Any one, here is set as ESP32 S2 Dev Kit)
    -DARDUINO_VARIANT="esp32s2"                                 <<<<<<=== Variant "folder" must match "/variants/folder" name
    -DESP32
    -DBOARD_HAS_PSRAM                                                    <<<<<=== Does the ESP32-S2 have PSRAM? Remove it if not.
    -DARDUINO_USB_MODE=0                                             <<<<<=== Activate USB OTG Mode
    -DARDUINO_USB_CDC_ON_BOOT=1                               <<<<<=== Enable Serial as USB CDC Interface
)
```

SuGlider, 2024-03-07:
@dzungpv - Please not that I have edited the comment above (https://github.com/espressif/arduino-esp32/issues/9326#issuecomment-1983576064)

The way to make it work is by editing the `CMakeLists.txt` file from `/main` folder to add `target_compile_options(${COMPONENT_TARGET} PUBLIC )` with the definition list for this project.

SuGlider, 2024-03-07:
>4, Add:
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)
add_definitions(
 -DARDUINO_USB_MODE=0
 -DARDUINO_USB_CDC_ON_BOOT=1
)
```
>to the root CMakeLists.txt

This shall be added to the `/main/CMakeLists.txt` instead of the root folder file.

SuGlider, 2024-03-11:
@dzungpv - Please let me know if the issue is solved.

SuGlider, 2024-03-14:
@dzungpv - I got it to work correctly using UART0 and USB CDC ports.
It is necessary to make some changes to `CMakeLists.txt` from Arduino Component.

Instead of changing `CMakeLists.txt` from `main/` or the from `/`, it is necessary to apply a change to the Line 224 of the `components/arduino-esp32/CMakeLists.txt` file:

```
target_compile_options(${COMPONENT_TARGET} PUBLIC
    -DARDUINO=10812
    -DARDUINO_${idf_target_for_macro}_DEV
    -DARDUINO_ARCH_ESP32
    -DARDUINO_BOARD="${idf_target_caps}_DEV"
    -DARDUINO_VARIANT="${CONFIG_ARDUINO_VARIANT}"
    -DARDUINO_USB_MODE=0
    -DARDUINO_USB_CDC_ON_BOOT=1
    -DCORE_DEBUG_LEVEL=4
    -DCONFIG_TINYUSB_CDC_ENABLED=1
    -DCONFIG_TINYUSB_ENABLED=1
    -DESP32)
```

This is the diff in `CMakeLists.txt`:
```
@@ -227,6 +227,11 @@ target_compile_options(${COMPONENT_TARGET} PUBLIC
     -DARDUINO_ARCH_ESP32
     -DARDUINO_BOARD="${idf_target_caps}_DEV"
     -DARDUINO_VARIANT="${CONFIG_ARDUINO_VARIANT}"
+    -DARDUINO_USB_MODE=0
+    -DARDUINO_USB_CDC_ON_BOOT=1
+    -DCORE_DEBUG_LEVEL=4
+    -DCONFIG_TINYUSB_CDC_ENABLED=1
+    -DCONFIG_TINYUSB_ENABLED=1
     -DESP32)
```
It is also necessay to add a component to the project in order to have TinyUSB available.
This shall be added as `components/arduino_tinyusb/`.
This shall be a copy of https://github.com/espressif/esp32-arduino-lib-builder/tree/master/components/arduino_tinyusb

After adding `arduino_tinyusb` component and changing the `components/arduino-esp32/CMakeLists.txt` file, it is possbible to build the project executing `idf.py build`.

SuGlider, 2024-03-14:
I think that this is the necessary trick... there could be a better way to get it done, but this is how I made it.

dzungpv, 2024-03-14:
> I think that this is the necessary trick... there could be a better way to get it done, but this is how I made it.

Yes it work, Just follow you step in https://github.com/espressif/arduino-esp32/issues/9326#issuecomment-1996282136
But additional step to make it work:

`Get tinyusb lib from https://github.com/hathach/tinyusb/releases and place in root folder off arduino_tinyusb components`

This very hard process, you must add this to the document.

SuGlider, 2024-03-14:
>`Get tinyusb lib from https://github.com/hathach/tinyusb/releases and place in root folder off arduino_tinyusb components`

I didn't have to clone `hathach/tinyusb`. I've just cloned `https://github.com/espressif/esp32-arduino-lib-builder` and copied its `esp32-arduino-lib-builder/components/arduino_tinyusb/` folder to `myProj/components`.

dzungpv, 2024-03-14:
> I didn't have to clone `hathach/tinyusb`. I've just cloned `https://github.com/espressif/esp32-arduino-lib-builder` and copied its `esp32-arduino-lib-builder/components/arduino_tinyusb/` folder to [...]

It show me the error not found tinyusb lib in the `arduino_tinyusb/tinyusb/` , event there is a tinyusb component in the IDF 4.4.6. So I search around and fine @Jason2866 [sample](https://github.com/Jason2866/Arduino_IDF_TinyUSB), so I get the fix like above.

SuGlider, 2024-03-14:
@dzungpv - I'll close this ticket as it seems solved. Please feel free to open it again.

closed 2024-03-14 by SuGlider