espressif/arduino-esp32 #9334: LWIP requires ipv6 when arduino as a component in esp-idf

dizcza, 2024-03-04:
### Board

esp32

### Device Description

TTGO T8

### Hardware Configuration

Build error

### Version

other

### IDE Name

idf

### Operating System

Ubu 22.04

### Flash frequency

doesn't matter

### PSRAM enabled

yes

### Upload speed

doesn't matter

### Description

Hey come on can't you set up a template for build errors

----

using git commit 090bc7242cfd597a370d5eb711618b1dd3a0bee8

Arduino as a component

I have 

```
CONFIG_LWIP_IPV6=n
```

in my sdkconfig.defaults and everything worked fine till I added arduino as a component in esp-idf v5.1.2-691-g7380f96017

Now I cannot compile the project. Removing the abusing line in the settings resolves the issue.

----

I opening the issue here because the error occurs while building arduino-esp32

### Sketch

```cpp
doesn't matter
```

### Debug Message

```plain
/home/dizcza/Projects/Embedded/eclipse-workspace/esp32-sdpsensor/components/arduino/libraries/AsyncUDP/src/AsyncUDP.cpp: In member function 'IPv6Address AsyncUDP::listenIPv6()':
/home/dizcza/Projects/Embedded/eclipse-workspace/esp32-sdpsensor/components/arduino/libraries/AsyncUDP/src/AsyncUDP.cpp:813:33: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
  813 |     if(!_pcb || _pcb->remote_ip.type != IPADDR_TYPE_V6){
      |                                 ^~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/esp32-sdpsensor/components/arduino/libraries/AsyncUDP/src/AsyncUDP.cpp:816:40: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
  816 |     return IPv6Address(_pcb->remote_ip.u_addr.ip6.addr);
```

### Other Steps to Reproduce

Related https://github.com/espressif/esp-idf/issues/6935 https://github.com/espressif/esp-idf/issues/2231

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-03-04:
Upcoming Arduino 3.0.0 needs IPv6 support enabled.

dizcza, 2024-03-04:
all right

Koxx3, 2024-05-21:
this is really annoying since IPv6 use 7KB of RAM as described by Espressif documentation :
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/lwip.html#lwip-ram-usage

really no plan to make it optionnal ?