espressif/arduino-esp32 #9336: New version 3.x.x new ledc / analogWriteResolution / analogWriteFrequency crossed interferences

fanfanlatulipe26, 2024-03-05:
### Related area

ESP32 familly

### Hardware specification

New version 3.x.x new ledc / analogWriteResolution / analogWriteFrequency 

### Is your feature request related to a problem?

. Interested in PWM on ESP32. Just looking at the code in the master branch for 3.0.x
```
void analogWriteFrequency(uint8_t pin, uint32_t freq)
void analogWriteResolution(uint8_t pin, uint8_t resolution)
```

In fact analogWriteFrequency change also the resolution for the pin (using the previous result of an analogWriteResolution even if it was not for the same pin ) , and analogWriteResolution changes also,the frequency (using the previous result of an analogWriteResolution even if it was not for the same pin).
 
I know it was more or less the same in 2.0.xx

Why not changing only the frequency or the resolution ?
So it will be easy to choose for each pin the convenient frequency/resolution using only the analogXYXY interface.

If not we must use an explicit ledcChangeFrequency(). It should be easier for some user to only know the analogXYXY interface  without digging in the full  ledc API

And only use the 
```
static uint8_t analog_resolution = 8;
static int analog_frequency = 1000;
```
as default value for the first time a channel is allocated by a analogWrite(...)

### Describe the solution you'd like

void analogWriteFrequency(uint8_t pin, uint32_t freq) should only change frequency for the pin
void analogWriteResolution(uint8_t pin, uint8_t resolution) should only change the resolutionfor the pin                            

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-03-05:
They are rather intimately [tied together](https://docs.espressif.com/projects/esp-idf/en/v4.4.6/esp32s3/api-reference/peripherals/ledc.html#supported-range-of-frequency-and-duty-resolutions)
Would it be better if you got a message saying "Frequency not supported at current resolution" and vice-versa?

fanfanlatulipe26, 2024-03-05:
Of course some combinations of frequency/resolution are illegal and library generates an error if debug is enabled. It could be nice to have a return code for some of the analogXXX call ...

In my opinion the way analogWrite/analogWriteFrequency/analogWriteResolution handle the pwm chanel is questionable:

```
analogWrite(16, 128);		// PWM on  pin 16 is at 1khz, resolution 8 bit   (the default)
analogWriteFrequency(16, 2000); // PWM on  pin 16 is now 2khz, resolution 8 bit

analogWrite(17, 128);		// PWM on  pin 17 is at 2khz, resolution 8 bit . I would prefer the default 1khz/8 bits
analogWriteResolution(17, 10) ; // PWM on  pin 17 is now 2khz, resolution 10 bit .

analogWrite(18, 128);		// PWM on  pin 18 is at 2khz, resolution 10 bit . I would prefer the default 1khz/8 bits
```

Don't know if people play a lot with analogWriteFrequency and analogWriteResolution but if they don't want to play with ledcXYZ API the best may be to always make both in a row for setting up the channel
```
analogWrite(pinNbr,0);
analogWriteResolution(pinNbr, myRes);
analogWriteFrequency(pinNbr, myFreq);
```

Anyhow, I don't want to think about the mess with timers shared by two channels/pins when the user try to set different frequency/resolution on each pins  ;-) 
There is no longer the function `analogGetChannel(uint8_t pin)` and so it may be not easy to know the links between pins/channels/timers ....

P-R-O-C-H-Y, 2024-03-11:
Hello @fanfanlatulipe26, I agree that a return code of the `analogWrite` functions would be good, but we are following Official Arduino API, where all functions are without any return.

Also the `analogWriteResolution` and `analogWriteFrequency` are following the API, so it needs to set the default value for all new channels that would be created, same as Arduino API does. But in order to also change the already used pins, we added the parameter pin to the function, so it will also change the pin directly.

From my point of view using LEDC API is the best option there to have the best control and its really simple to use.
It also has a return so it can be checked for an error.
All you need is just call:
```
ledcAttach(pin, frequency, resolution);
ledcWrite(duty);
```
And you can always change the frequency and resolution by calling:
```
ledcChangeFrequency(pin, frequency, resolution);
```

Let me know your opinion, if all is clear feel free to close this issue.

fanfanlatulipe26, 2024-03-11:
I agree that LEDC API is the way to go if we need best control of the pwm
But with ESP32 2.x the user has the burden of managing pins and explicit channel. ESP 3.x will be best on this point.

I am writing a pwm motor driver and want it to run on many platforms without too much conditional compilation.
Using the analogWrite is simple in this case has it is supported on most processors.

Unfortunatly there is at least 1 discrepancy for setting the frequency: some platforms (ESP32 ...) use **analogWriteFrequency** when others (ESP8266, RP2040) use **analogWriteFreq**
Not a big deal but well ...

Closing the issue. Thanks for the exchange.

closed 2024-03-11 by fanfanlatulipe26