espressif/arduino-esp32 #9341: 'disableInterrupt' was not declared in this scope

Ismola, 2024-03-05:
### Board

ESP32-C6

### Device Description

ESP32-C6-DevKitC-1

### Hardware Configuration

n/a 

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

n/a

### Description

I am trying to use the **_Zigbee_Light_Switch_** example code, but the code does not compile. This is the error: 
`'enableInterrupt' was not declared in this scope; did you mean 'detachInterrupt'`

I have the board configured correctly. The **_Zigbee_Light_Bulb_** example compiles correctly for me and is uploaded to my board

### Sketch

```cpp
I have two ESP32-C6 boards

One with the code of the Zigbee_Light_Bulb example, and the other with the Zigbee_Light_Switch example.
```

### Debug Message

```plain
FQBN: esp32:esp32:esp32c6:DebugLevel=info,EraseFlash=all,ZigbeeMode=zczr
Using board 'esp32c6' from platform in folder: C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3
Using core 'esp32' from platform in folder: C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3

cmd /c if exist "D:\\Codigo\\ESP32\\arduino-esp32\\libraries\\ESP32\\examples\\Zigbee\\Zigbee_Light_Switch\\partitions.csv" COPY /y "D:\\Codigo\\ESP32\\arduino-esp32\\libraries\\ESP32\\examples\\Zigbee\\Zigbee_Light_Switch\\partitions.csv" "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\partitions.csv"
cmd /c if not exist "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\partitions.csv" if exist "C:\\Users\\Ismael\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\variants\\esp32c6\\partitions.csv" COPY "C:\\Users\\Ismael\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\variants\\esp32c6\\partitions.csv" "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\partitions.csv"
cmd /c if not exist "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\partitions.csv" COPY "C:\\Users\\Ismael\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\tools\\partitions\\default.csv" "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\partitions.csv"
        1 archivo(s) copiado(s).
cmd /c IF EXIST "D:\\Codigo\\ESP32\\arduino-esp32\\libraries\\ESP32\\examples\\Zigbee\\Zigbee_Light_Switch\\bootloader.bin" ( COPY /y "D:\\Codigo\\ESP32\\arduino-esp32\\libraries\\ESP32\\examples\\Zigbee\\Zigbee_Light_Switch\\bootloader.bin" "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\Zigbee_Light_Switch.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\Ismael\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\variants\\esp32c6\\bootloader.bin" ( COPY "C:\\Users\\Ismael\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\variants\\esp32c6\\bootloader.bin" "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\Zigbee_Light_Switch.ino.bootloader.bin" ) ELSE ( "C:\\Users\\Ismael\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.6/esptool.exe" --chip esp32c6 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\Zigbee_Light_Switch.ino.bootloader.bin" "C:\\Users\\Ismael\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32c6\\bin\\bootloader_qio_80m.elf" ) )
esptool.py v4.6
Creating esp32c6 image...
Merged 2 ELF sections
Successfully created esp32c6 image.
cmd /c if exist "D:\\Codigo\\ESP32\\arduino-esp32\\libraries\\ESP32\\examples\\Zigbee\\Zigbee_Light_Switch\\build_opt.h" COPY /y "D:\\Codigo\\ESP32\\arduino-esp32\\libraries\\ESP32\\examples\\Zigbee\\Zigbee_Light_Switch\\build_opt.h" "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\build_opt.h"
cmd /c if not exist "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\build_opt.h" type nul > "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\\build_opt.h"
cmd /c type nul > "C:\\Users\\Ismael\\AppData\\Local\\Temp\\arduino\\sketches\\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1/file_opts"
Detecting libraries used...
C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302/bin/riscv32-esp-elf-g++ -c @C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10607 -DARDUINO_ESP32C6_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32C6_DEV" -DARDUINO_VARIANT="esp32c6" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=default,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=info,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=zczr" -DESP32 -DCORE_DEBUG_LEVEL=3 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DZIGBEE_MODE_ZCZR @C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/flags/defines -ID:\Codigo\ESP32\arduino-esp32\libraries\ESP32\examples\Zigbee\Zigbee_Light_Switch -iprefix C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/include/ @C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/flags/includes -IC:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/qio_qspi/include -IC:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32 -IC:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\variants\esp32c6 @C:\Users\Ismael\AppData\Local\Temp\arduino\sketches\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1/build_opt.h @C:\Users\Ismael\AppData\Local\Temp\arduino\sketches\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1/file_opts C:\Users\Ismael\AppData\Local\Temp\arduino\sketches\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\sketch\Zigbee_Light_Switch.ino.cpp -o nul
Generating function prototypes...
C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302/bin/riscv32-esp-elf-g++ -c @C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10607 -DARDUINO_ESP32C6_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32C6_DEV" -DARDUINO_VARIANT="esp32c6" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=default,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=info,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=zczr" -DESP32 -DCORE_DEBUG_LEVEL=3 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DZIGBEE_MODE_ZCZR @C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/flags/defines -ID:\Codigo\ESP32\arduino-esp32\libraries\ESP32\examples\Zigbee\Zigbee_Light_Switch -iprefix C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/include/ @C:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/flags/includes -IC:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-3662303f31/esp32c6/qio_qspi/include -IC:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32 -IC:\Users\Ismael\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\variants\esp32c6 @C:\Users\Ismael\AppData\Local\Temp\arduino\sketches\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1/build_opt.h @C:\Users\Ismael\AppData\Local\Temp\arduino\sketches\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1/file_opts C:\Users\Ismael\AppData\Local\Temp\arduino\sketches\E2311A0DCDE5F4FA0B9BFBBF2C68E1D1\sketch\Zigbee_Light_Switch.ino.cpp -o C:\Users\Ismael\AppData\Local\Temp\3979930044\sketch_merged.cpp
[... cut ...]
  232 |             enableInterrupt((button_func_pair[i]).pin);
      |             ^~~~~~~~~~~~~~~
      |             detachInterrupt

exit status 1

Compilation error: 'enableInterrupt' was not declared in this scope; did you mean 'detachInterrupt'?
```

### Other Steps to Reproduce

I use Windows, and Arduino IDE 2.32

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-03-05:
There is no support for the C6 with Arduino Core 2.0.14.

lbernstone, 2024-03-05:
Yup, [this](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-gpio.h#L81-L82) is a 3.0 thing.

Ismola, 2024-03-05:
> Yup, [this](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-gpio.h#L81-L82) is a 3.0 thing (as is all C6 support).

Sorry, but I'm new to programming in ESP32-C6 and C. How should I implement your solution?

Ismola, 2024-03-06:
> You would need to do a [manual installation](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation). As a new user, I would strongly recommend you just wait a couple weeks until 3.0.0 is released (or at least a RC). Zigbee is still very rough around the edges.

Thank you so much. Compiling and flashing worked using manual installation.

The switch still doesn't turn the light bulb on and off, but at least it compiles

SuGlider, 2024-03-18:
@P-R-O-C-H-Y - Could you please help with the last comment about "The switch still doesn't turn the light bulb on and off"?

SuGlider, 2024-03-24:
It seems solved due to non response. Closing.

closed 2024-03-24 by SuGlider

ghmeteo, 2024-04-11:
Hi everyone,...I don't know if it is appropriate to place the problem in this closed issue.
I have the same problem as the ISMOLA user's initial one with the compilation of the Zigbee_Light_Switch example code for the ESP32-C6-DEV-KIT-N8 card
Arduino IDE ESP32 core updated to 3.0.0-alpha3 installed.
ESP32-C6 Dev Module selected card.
I have also done the manual installation

Zigbee_Light_Switch example code.
The code does not compile. This is the error:
'enableInterrupt' was not declared in this scope; did you mean 'detachInterrupt'

Zigbee_Light_Bulb compile without problems, logically by not including the "enableInterrupt" and "disableInterrupt" function in the code.

Ismola, 2024-04-11:
> Hi everyone,...I don't know if it is appropriate to place the problem in this closed issue. I have the same problem as the ISMOLA user's initial one with the compilation of the Zigbee_Light_Switch example code for the ESP32-C6-DEV-KIT-N8 card with ESP32-C6-WROOM-1. Arduino IDE ESP32 core updated to 3.0.0-alpha3 installed. ESP32-C6 Dev Module selected card. I have also done the manual installation
> 
> Zigbee_Light_Switch example code. The code does not compile. This is the error: 'enableInterrupt' was not declared in this scope; did you mean 'detachInterrupt'
> 
> Zigbee_Light_Bulb compile without problems, logically by not including the "enableInterrupt" and "disableInterrupt" function in the code.

Entiendo que has comentado esas lineas, y por eso compila el codigo. Pero, ¿funciona?, hace lo que debería hacer el ejemplo¿

ghmeteo, 2024-04-11:
No,... I have not commented on those lines. If I comment on the lines that cause the error, of course there is no error and the compilation ends, but the code will not do what it has to do, which is turn on the LED of the card configured as ENDDEVICE when the button installed on the COORDINATOR card is activated. .

Ismola, 2024-04-11:
> No,... I have not commented on those lines. If I comment on the lines that cause the error, of course there is no error and the compilation ends, but the code will not do what it has to do, which is turn on the LED of the card configured as ENDDEVICE when the button installed on the COORDINATOR card ( GPIO_NUM_9) is activated. .

I didn't find a solution. I simply used esp-idf without arduino, with vs-code and its plugin. The examples work the first time, but it is more difficult to understand what it does. Arduino is simpler but fails.

ghmeteo, 2024-04-11:
Ok... so the problem initially described is not solved.

P-R-O-C-H-Y, 2024-04-11:
The functions `enableInterrupt` and `disableInterrupt` were added in this commit: https://github.com/espressif/arduino-esp32/commit/c7a109bac4dda7cc76b8ed7e6aaf07f36bd15660 in December 2023. It's not included in the 3.0.0-aplha3 version. We are releasing the new 3.0.0-rc1 version soon on you can use manual install and use the master branch.

ghmeteo, 2024-04-12:
Ok... perfect, I'll wait for the new version. Thanks!

P-R-O-C-H-Y, 2024-04-12:
> Ok... perfect, I'll wait for the new version. Thanks!

@ghmeteo 3.0.0-rc1 is out! https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-rc1

ghmeteo, 2024-04-15:
Hi... with 3.0.0-rc1 compiles and switch-bulb example works OK!! Thanks!!!...

ghmeteo, 2024-04-29:
Hello again,...I am trying to obtain the Zigbee transmission power with the function: void esp_zb_get_tx_power(int8_t *power) . This function gets the tx power value in dB.

Using the Light_Switch/Light_bulb example I call the esp_zb_get_tx_power() function when I press the button connected to the Coordinator. The result is that the system reboots...
[image]
Anybody knows what could be the problem..?