espressif/arduino-esp32 #9347: Add LittleFS update to HTTPUpdate and HTTPUpdateServer

lucasssvaz, 2024-03-06:
### Related area

HTTPUpdate

### Hardware specification

Any

### Is your feature request related to a problem?

Requested in #6482.

Currently, HTTPUpdate and HTTPUpdateServer can't handle LittleFS.

### Describe the solution you'd like

Add LittleFS support to HTTPUpdate and HTTPUpdateServer.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

tobiasfaust, 2024-03-12:
Hi,
thank you so much for taking care of this feature. Is existing a plan when we can expect U_LITTLEFS Feature?

HugoML, 2025-07-18:
Any roadmap for this feature to be done? Hope to see this implemented soon. Otherwise, it is a big roadblock for LittleFS to replace SPIFFS.

jblazeg, 2025-09-10:
here is my patch suggestion

**Update.h**
add this line after `U_AUTH 200`
`#define U_LITTLEFS   101`

**Updater.cpp**
add this block after this block _else if (command == U_SPIFFS) {}
`else if (command == U_LITTLEFS) {
    _partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_LITTLEFS, label);
    _paroffset = 0;
    if (!_partition) {
      _error = UPDATE_ERROR_NO_PARTITION;
      return false;
    }
  }`

now I were able to update my partition with:
`Update.begin(UPDATE_SIZE_UNKNOWN, updatetype,-1,0, "web")`

web is my webserver partition defined in my partitions.csv
`web,      data, littlefs,   [START], [SIZE],`