espressif/arduino-esp32 #9348: Enable SDKonfig options for RGB displays using bounce buffer

pauleilio, 2024-03-06:
### Board

ESP32-S3R8 SoC custom board

### Device Description

Custom board with RGB display using PSRAM frame buffer

### Hardware Configuration

RGB display, touch controller, RTC clock, I2C gpio expander

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

During WiFi and BT operations a frame shift relating to bounce buffer out of sync occurs. This effect can be minimized by enabling this options

CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
CONFIG_LCD_RGB_RESTART_IN_VSYNC=y

We know we can pre-compile a library with a custom configuration but if it does not harm any other aspect of the board and have no detremental impact in general ESP-S3 use could they be enabled by default. The result in doing this on an RGB display is just a slight flash rather than a permanent shift of frame.

### Sketch

```cpp
N/A various tasks replicate the issue
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

The enabling of these options is a well known ESP-IDF solution for RGB displays using bounce buffers

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-03-06:
`CONFIG_LCD_RGB_RESTART_IN_VSYNC=y` has negative side effects.
See https://espressif-docs.readthedocs-hosted.com/projects/esp-faq/en/latest/software-framework/peripherals/lcd.html#why-do-i-get-drift-overall-drift-of-the-display-when-driving-an-rgb-lcd-screen

Jason2866, 2024-04-12:
I had this options enabled and had crashes with S3 QIO Flash and QIO PSRAM
```
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
```
the seems to be faulty with some hardware configs and not useable for a general config.

VojtechBartoska, 2024-09-12:
Hello, closing this as wontfix. Thanks for understanding.

closed 2024-09-12 by VojtechBartoska