espressif/arduino-esp32 #9351: S3 QIO 120Mhz Flash speed not possible

Jason2866, 2024-03-06:
### Board

every s3 board with QIO Flash capable 120Mhz

### Device Description

Setting Flash speed 120Mhz for S3 not possible

### Hardware Configuration

none

### Version

latest master (checkout manually)

### IDE Name

ArduinoIDE, Platdormio

### Operating System

MacOS Ventura

### Flash frequency

120Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Using the setting 120Mhz flash speed from `boards.txt`
esp32s3.menu.FlashMode.qio120.build.boot_freq=120m
This setting is nowhere used in `platform.txt` when the firmware is merged
https://github.com/espressif/arduino-esp32/blob/4123e20a2ca28db70ffeb4db3d5157c8e5b1a058/platform.txt#L159
nor when flashed with esptool
https://github.com/espressif/arduino-esp32/blob/4123e20a2ca28db70ffeb4db3d5157c8e5b1a058/platform.txt#L278
and
https://github.com/espressif/arduino-esp32/blob/4123e20a2ca28db70ffeb4db3d5157c8e5b1a058/platform.txt#L286

### Sketch

```cpp
none
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-03-14:
Hi @Jason2866, we are building the boot loader for 120MHz QIO correctly, double checked that and this is printed on build:
`Note: HPM is enabled for the flash, force the ROM bootloader into DOUT mode for stable boot on`
More about the HPM in the [docs](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32s3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#high-performance-mode). So that means that boot loader flash will run in 120 MHz speed.

The 120MHz boot loader is selected here in the platform.txt:
https://github.com/espressif/arduino-esp32/blob/4123e20a2ca28db70ffeb4db3d5157c8e5b1a058/platform.txt#L117-L121

The `build.boot_freq` is at the end of the line, selecting the right bootloader.

Is there any way you figured out, that the boot loader is not running flash at 120MHz?

P-R-O-C-H-Y, 2024-03-14:
But maybe according to the note from build, we should use DOUT mode instead of DIO for 120 MHz.

Jason2866, 2024-03-14:
@P-R-O-C-H-Y Thx for the investigation. Overlooked the entry in platform.txt.
I was trying with Platformio my attempts with 120Mhz ended with errors in build process with esptool.py. There is nothing regarding `build.boot_freq` in the Platformio scripts.
So looked in platform.txt and didn't find it there, i opened this issue. 
Since `build.boot_freq` is there (overlooked, sorry) it probably works with 120Mhz when using ArduinoIDE.
Leaving one question, can it be tested if the flash speed is 120Mhz?

Jason2866, 2024-03-14:
@P-R-O-C-H-Y Thx again. Will enable logging (as much as possible in IDF) and build the Arduino Libs with verbose logging. 
The Infos from you are very Helpful! With them i think i can adopt the Platformio build script(s) to support the 120Mhz mode.

P-R-O-C-H-Y, 2024-03-14:
@Jason2866 You are welcome :) Let us know the results and I think we can rename this issue to be related to PIO which can be closed if you successfully adopt the PIO build script to support 120MHz mode.

Jason2866, 2024-03-14:
@P-R-O-C-H-Y  Yes, will come back when a solution for Pio is found.

TD-er, 2024-03-14:
You could mirror the SPI clock pin to some other pin and then simply measure the frequency.
Like I did for the ESP32-C2 to double check the SPI frequency.

Jason2866, 2024-03-15:
@P-R-O-C-H-Y  Opened PR https://github.com/espressif/arduino-esp32/pull/9380

P-R-O-C-H-Y, 2024-03-15:
@Jason2866 Great! Already checked and looking good.

Jason2866, 2024-03-15:
> But maybe according to the note from build, we should use DOUT mode instead of DIO for 120 MHz.

@P-R-O-C-H-Y  Recommended in IDF 5.1 esptool to use DOUT when using 120Mhz https://github.com/espressif/esp-idf/blob/release/v5.1/components/esptool_py/project_include.cmake