espressif/arduino-esp32 #9356: Crashing on Read/Write with SD library functions

KeiOmni, 2024-03-10:
### Board

Esp32 Dev Module

### Device Description

Dev Kit C

### Hardware Configuration

HSPI - GPIO 12, 13,14, 27 (SS) connected to sdcard. VSPI ( 15 (SS) , 2 ( LCD DC), 4 ( LCD RST),18 (CLK) ,19 (MISO), 23 (MOSI) connected to only Spi Screen.

GPIO 36, 39, 32,33,34,35 connected to Analog Inputs.

### Version

v2.0.14

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 11

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

11

### Description

the SD library read and write functions  causes crashing in Loop() and any RTOS Task created with e.g xTaskCreateUniversal or others. It only works fine in Setup().

### Sketch

```cpp
#include "SD.h"
#include "SPI.h"

#define HSPI_MISO   12
#define HSPI_MOSI   13
#define HSPI_SCLK   14
#define HSPI_SS     27

SPIClass * hspi = NULL;
static const uint32_t hspiClk = 8000000; // 8 MHz
char* DBuffer = (char*)malloc(1024);

bool  StorageMounted = false;
void setup(){ 
 
   Serial.begin(115200);

   hspi = new SPIClass(HSPI);
    SPIClass mSPI = *hspi;
    pinMode(HSPI_SS, OUTPUT);    
    //digitalWrite(HSPI_SS, HIGH);
    hspi->begin(HSPI_SCLK,HSPI_MISO,HSPI_MOSI,HSPI_SS);
   // hspi->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));
    bool SD_OK =  SD.begin(HSPI_SS,mSPI,hspiClk,"/sd",10,false);
   
    if(!SD_OK){
      StorageMounted = false;
        Serial.println("Card Mount Failed");
        return;
    }
    else {  StorageMounted = true ; Serial.println("Card Mounted. "); }
    
     uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

    listDir(SD, "/", 0);
}

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void readFile(fs::FS &fs, const char * path, char* FileBuffer, uint32_t FileSize){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

    Serial.print("Read from file: ");

    while(file.available()){
        file.readBytes(FileBuffer, FileSize);
    }
    file.close();
}

bool disableREAD =false;

void loop(){
 
   if(StorageMounted && !disableREAD)
  {
     readFile(SD, "/test.txt",DBuffer,8);
     Serial.println("Done Reading");
     disableREAD =true;
  }
}
```

### Debug Message

```plain
19:50:45.378 -> Reading file: /test.txt
19:50:45.378 -> 
19:50:45.378 -> assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))
19:50:45.378 -> 
19:50:45.378 -> 
19:50:45.378 -> Backtrace: 0x400835dd:0x3ffb1ae0 0x4008984d:0x3ffb1b00 0x4008e749:0x3ffb1b20 0x4008a5ad:0x3ffb1c50 0x400d4e06:0x3ffb1c90 0x400d2539:0x3ffb1cc0 0x400e35b2:0x3ffb1cf0 0x400e38b1:0x3ffb1d10 0x400e584b:0x3ffb1d50 0x400e706e:0x3ffb1fc0 0x400e09c2:0x3ffb2130 0x400f0ec0:0x3ffb2150 0x400d490e:0x3ffb2170 0x400d3856:0x3ffb2200 0x400d188b:0x3ffb2230 0x400d19c5:0x3ffb2270 0x400d8141:0x3ffb2290
19:50:45.422 -> 
19:50:45.422 -> 
19:50:45.422 -> 
19:50:45.422 -> 
19:50:45.422 -> ELF file SHA256: 6a1c002771b20c20
19:50:45.422 ->
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-11:
You are using a local variable as part of a static object.  Don't do that.  I already showed you how to fix it.

sailingfree, 2024-06-13:
Yep, found this too after using the example in the SD library. The SPI object is local and goes out of scope when setup finishes. trying to use the SD object in the loop finds an invalid semaphore. My setup now looks like this so the SPI object stays around after setup exits.

SPIClass myspi = SPIClass(HSPI);

void setup(void) {
    myspi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
    SD.begin(SD_CS, myspi, 80000000);

.....

VojtechBartoska, 2024-09-12:
Closing this as answered.

closed 2024-09-12 by VojtechBartoska