espressif/arduino-esp32 #9357: use char instead of String in SerialBT.begin()

BlackWolfDEsign, 2024-03-11:
### Related area

SerialBT.begin()

### Hardware specification

ESP32 DevKit c3

### Is your feature request related to a problem?

no

### Describe the solution you'd like

i want to use a char array instead of String because i read String is more insecure then char[]

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-03-11:
how about `SerialBT.begin(String(char_array));`? And what do you mean by "insecure"?

BlackWolfDEsign, 2024-03-11:
currently i use String().  I read String can cause memory (overflow) problems

me-no-dev, 2024-03-11:
Sorry, never heard of the issue. Do you have any links to such info/discussion? String is the default class for handling char arrays in Arduino. It takes a bit more memory then a char array, but not much and it offers many features for that price.

BlackWolfDEsign, 2024-03-11:
[lnk](https://forum.arduino.cc/t/difference-between-char-array-and-string/603608/10)

> It took me a while to unlearn using the String class, but thanks to some very patient experts here, I finally grok.I had no choice because I was running out of RAM in a project and it was full of Strings (note the upper-case "S"). Strings (String class) are a memory hog, but using c-strings (lower-case "s"), saved my project.

[lnk2](https://forum.arduino.cc/t/understanding-the-evil-string/427528/11)

SuGlider, 2024-03-14:
> [lnk](https://forum.arduino.cc/t/difference-between-char-array-and-string/603608/10)
>
>
> [lnk2](https://forum.arduino.cc/t/understanding-the-evil-string/427528/11)

I understand that using many `String` objects that change along time (or C++ sdt:: many things that "mallocs") in a MCU that has just 16KB or 32KB of SRAM may be an issue (AVR) because of memory fragmentation, in very specific use cases.

But ESP32-C3 has 400KB SRAM. The smallest memory case may be the C2 that has 272KB SRAM.
For most Arduino sketches it won't make any difference.

closed 2024-03-18 by BlackWolfDEsign