espressif/arduino-esp32 #9358: TWAI receiver RX issue

hicham-vv, 2024-03-11:
### Board

az-delivery-devkit-v4

### Device Description

development  with SN65HVD230 CAN driver

### Hardware Configuration

CAN RX_PIN 19
CAN TX_PIN 18

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

hello guys 
I was working with a library to sniff data CAN J1939 of  miwagner/ESP32CAN @ ^0.0.1 . 
i am able to Read Data with out any problems, i was able to receive many IDs, as you can see in debug message .

when i switch the TWAI library receiver , i only sniff and  got ID cf00203 at all, as you can see in debug message

i tried to configure it to accept all data , no filter added but with out success.

it's the same hardware for both code.

any help thank you 

Sketch 1 is for miwagner Library 
Sketch 2 is for TWAI 

### Sketch

```cpp
/**** *Sketch 1 ******/

CAN_device_t CAN_cfg;               // CAN Config
unsigned long previousMillis = 0;   // will store last time a CAN Message was send
const int interval = 1000;          // interval at which send CAN Messages (milliseconds)
const int rx_queue_size = 5;       // Receive Queue size

void setup() {
  Serial.begin(115200);
  while(!Serial);
  delay(1500);
  Serial.println("Basic Demo - ESP32-Arduino-CAN");
  #ifdef FUSOCANTER
[... 198 lines cut ...]
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
Message is in Extended Format  ID: cf00203 Byte: fc 00 00 ff ff ff ff ff
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2024-03-11:
Could you please try if this issue occurs when using ESP-IDF only ?

SuGlider, 2024-03-11:
This may be an issue relaed to the external library miwagner/ESP32CAN.
Maybe someone from the community may help...

aaarnas, 2024-03-12:
In mode `TWAI_MODE_LISTEN_ONLY` ESP32 only listens to CAN bus network traffic and does not acknowledge received messages. Packet is getting retransmitted, because no one answers to it.

To make it work - connect additional CAN bus device with mode "NORMAL", or change to `TWAI_MODE_NORMAL`.

miwagner library works, because it uses `TWAI_MODE_NORMAL`.

hicham-vv, 2024-03-12:
> Could you please try if this issue occurs when using ESP-IDF only ?

I will try that, and let you know

hicham-vv, 2024-03-12:
> In mode `TWAI_MODE_LISTEN_ONLY` ESP32 only listens to CAN bus network traffic and does not acknowledge received messages. Packet is getting retransmitted, because no one answers to it.
>
> To make it [...]

I am sniffing CAN data from my vehicule, and i think there is many Master/Slave connected to the bus line to receive and transmit messages , I use TWAI_MODE_LISTEN_ONLY to only  receive and i don't want to transmit nothing in the BUS.

But i will try that mode and let you know

hicham-vv, 2024-03-13:
> In mode `TWAI_MODE_LISTEN_ONLY` ESP32 only listens to CAN bus network traffic and does not acknowledge received messages. Packet is getting retransmitted, because no one answers to it.
>
> To make it [...]

SOLUTION :
you are right , i changed it to mode Normal 
to close

hicham-vv, 2024-03-13:
thanks for your help

closed 2024-03-13 by hicham-vv