espressif/arduino-esp32 #9360: HWCDC & periman broken

lbernstone, 2024-03-11:
### Board

esp32-c6

### Device Description

devkitc

### Hardware Configuration

usb connected (to usb-otg)

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

linux

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

usb

### Description

https://github.com/espressif/arduino-esp32/pull/9331 has issues.  Please test on a esp32-c3 or c6.  Pins do not get set properly. The pin assignments on 249&258 are wrong, and even if pins are set properly I don't get USB-CDC starting.

### Sketch

```cpp
Any with serial output
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2024-03-12:
@SuGlider PTAL

SuGlider, 2024-03-12:
@lbernstone - Correct. I'll fix it.

SuGlider, 2024-03-12:
> Not sure if related, but HWCDC on the ESP32-S3 is also working really bad. Not as broken as on the C3 (doesn't even show up as serial port anymore in Windows) but it does skip quite a lot of bytes sent to any connected terminal (I use it as console in ESPEasy) I used to send it per byte from a buffer to the serial port, but I noticed it is now doing quite a lot of locking, so I adapted the code to flush in chunks. Now it just flushes my chunk of data and then doesn't output anything for quite a while. Just like the `availableForWrite` is reporting 0. (have to check to be sure this is causing it)

@TD-er - please open a new issue reporting this problem with a sketch that I can use to reproduce it.

PR #9365 (2024-03-14): Fixes HW Serial attaching pins, PHY initialization, crash after end()
## Description of Change
Fixes:
- pin number error
- PHY initialization that prevented HW Serial to start correctly and get enumerated after end() or detaching pins
- crash after executing end() 
- force USB Host enumeration in order to set the right state to detect first time sending data to the HOST (CDC Connection) 

## Tests scenarios
Tested with ESP32-C3, ESP32-S3, ESP32-C6, ESP32-H2

``` cpp
// HW Serial CDC mode with CDC on boot enabled

void setup() {
  Serial.begin();
  Serial.end();
  Serial.begin();
  // waits unt [... cut]
closed 2024-03-14 by me-no-dev