espressif/arduino-esp32 #9361: esp32c3 esp_core_dump_flash???

kangz543g, 2024-03-12:
### Board

esp32c3

### Device Description

E (191) esp_core_dump_flash: Core dump data check failed: ????

### Hardware Configuration

no

### Version

v2.0.13

### IDE Name

arduino esp32

### Operating System

windows10

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

no

### Sketch

```cpp
no
```

### Debug Message

```plain
no
```

### Other Steps to Reproduce

no

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kangz543g, 2024-03-12:
E (191) esp_core_dump_flash: Core dump data check failed: ????

lucasssvaz, 2024-03-12:
Please provide more information about what are you trying to do and the code use. We can't help with only the error shown.

closed 2025-01-15 by Parsaabasi