espressif/arduino-esp32 #9362: When will v3.0.0 be released?

Zhu-jiatong, 2024-03-12:
### Related area

N/A

### Hardware specification

N/A

### Is your feature request related to a problem?

I am curious of when v3.0.0 of the Arduino Core be released? I am making a library and is looking forward to using some C++17 features.

### Describe the solution you'd like

An estimated time will do.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2024-03-12:
Hello,

we are currently working on this release. The last left task is Networking refactoring, we will try to release RC1 version as soon as possible and if everything is okay, we will follow with the stable release.

Unfortunately I am not able to give you exact date.

Closing this ticket as answered.

VojtechBartoska, 2024-03-12:
Relates to https://github.com/espressif/arduino-esp32/issues/8796