espressif/arduino-esp32 #9363: ESP32S2 could not reassign Serial pins

dzungpv, 2024-03-12:
### Board

Lolin S2 Mini

### Device Description

Lolin S2 Mini

### Hardware Configuration

GPIO 9 and 11 as UART pins

### Version

v2.0.14

### IDE Name

Arduino, IDF 4.4.6

### Operating System

Macos 13, Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I use Serial1.begin(9600, SERIAL_8E1, 11, 9) it work but not with
Serial.begin(9600, SERIAL_8E1, 11, 9)

### Sketch

```cpp
Serial.begin(9600, SERIAL_8E1, 11, 9)
```

### Debug Message

```plain
No debug
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-03-13:
@dzungpv - The Lolin S2 Mini has a USB port that is actually the USB OTG CDC peripheral.
 When this board is selected, it will automatically enable the "CDC on Boot", therefore, `Serial` will be actually the USB CDC Serial.
 
 In order to use UART0, please use `Serial0` instead of `Serial` (USB CDC).
 `Serial` will be the USB CDC port.
 `Serial0` will be the UART0 port.
 `Serial1` will be the UART1 port.

dzungpv, 2024-03-13:
@SuGlider as mention in in https://github.com/espressif/arduino-esp32/issues/9326 I still could not enable USB CDC on ESP EDF, so it only have Serial and Serial1. the S2 too limit compare to ESP32 and ESP32S3.
So this is a bug of the Core 2.0.

SuGlider, 2024-03-14:
Are you using the Arduino IDE?

SuGlider, 2024-03-14:
It works fine with the Arduino IDE, but enabling "USB CDC on Boot" in the tools menu.

If you need to use Arduino as IDF Component, please follow the instructions in https://github.com/espressif/arduino-esp32/issues/9326#issuecomment-1996282136

SuGlider, 2024-03-14:
>So as you say I must enable USB CDC on Boot to use UART0 on the ESP32S2 with Arduino? I mean I can assign other pin for it? I use ESP IDF with Arduino component.

No, ESP32-S2 Arduino has 2 possible configurations:
1) USB CDC on Boot is **Enabled**: 
- `Serial` is the USB CDC port and the pins can't be changed.
- UART0 is `Serial0` and UART1 is `Serial1`. The pins can be changed for both.
2) USB CDC on Boot is **Disabled**: 
- UART0 is the  `Serial` and UART1 is `Serial1`. The pins can be changed for both.

dzungpv, 2024-03-14:
> No, ESP32-S2 Arduino has 2 possible configurations:
>
>     1. USB CDC on Boot is **Enabled**:
>
>
>     * `Serial` is the USB CDC port and the pins can't be changed.
>
>     * UART0 is `Serial0` and UART1 is [...]

Yes, but my problem: could not re-assign UART0 Pin with Arduino, I test with Arduino and ESP IDF with Kaluga dev board and Lolin S2 mini. So it is impossible running 2 UART at same time, My previous design running ESP32 without any problem because it has three UART

SuGlider, 2024-03-14:
Ok, the issue is about reassigning UART0 pins using Arduino Core 2.0.14.
There is an improvement/fix in the PR #9176 about pin attachment.

Could you please apply this PR or just replace all 4 files: `HardwareSerial.cpp`, `HardwareSerial.h`, `esp32-hal-uart.c` and `esp32-hal-uart.h` within your Arduino Core installation?

I think that it will solve the issue. This shall be released soon within the Arduino Core 2.0.15.
Please let me know.

dzungpv, 2024-03-14:
> Ok, the issue is about reassigning UART0 pins using Arduino Core 2.0.14. There is an improvement/fix in the PR #9176 about pin attachment.
>
> Could you please apply this PR or just replace all 4 [...]

I will pull this branch and it will be OK: https://github.com/espressif/arduino-esp32/tree/release/v2.x

SuGlider, 2024-03-14:
@dzungpv - Please let me know if the code from `release/v2.x` solves this issue. Thanks!

dzungpv, 2024-03-14:
> @dzungpv - Please let me know if the code from `release/v2.x` solves this issue. Thanks!

It seem not working, I tested same code but with Serial1 it work but everytime I change to Serial (cdc not enable) it not work again, same pins, same code. You should retest the code on real device.

SuGlider, 2024-03-14:
@dzungpv - Issue confirmed. There is a problem with UART0 pin setup. I'll work on a PR and submit it by today.

SuGlider, 2024-03-14:
@dzungpv - Fix by #9373 - Please apply the fix to your local develoment environment and let me know if this works fine now.

dzungpv, 2024-03-15:
> @dzungpv - Fix by #9373 - Please apply the fix to your local develoment environment and let me know if this works fine now.

It is OK now. I tested and it work fine.

SuGlider, 2024-03-18:
Closed by #9373  - ready for Core 2.0.15 release.

PR #9373 (2024-03-18): fix: UART0 boot pin setup
## Description of Change

UART0 pins are set by ROM Boot to default values. 
This must be reflected into Arduino HardwareSerial in order to allow it to correctly detach it and then attach it to something else.

Summary:
UART0 constructor sets default RX/TX pins as done in boot time.

## Tests scenarios
Tested with DevKits ESP32, ESP32-S3, ESP32-S2.

``` cpp
// necessary to crossconnect 
// TX0(4) <--> Original TX pin, in order to see the output in the Serial Monitor - testing TX0 pin
// RX0(2) <--> TX1(17) for testing RX0 pin
void setup() {
  Serial.begin(115200, SERI [... cut]
closed 2024-03-18 by SuGlider