espressif/arduino-esp32 #9369: Rapid NeoPixel updates cause apparent hang on ESP32-S3

einglis, 2024-03-13:
### Board

Lolin S3 mini

### Device Description

Bare Lolin S3 mini board.  Nothing attached but USB and a serial monitor.

### Hardware Configuration

FTDI cable attached to TX (43), RX (44), and Ground.

### Version

v2.0.13

### IDE Name

Arduino IDE (also seen with PlatformIO)

### Operating System

N/A  (but actually Linux Mate)

### Flash frequency

240 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have a sketch that blinks a status LED at a rate of 10 Hz (from a Ticker).  If the CPU gets busy, the updates get bunched together in a burst and the system locks up.  This lock up has not been seen on boards with a conventional status LED.
I've replicated this with the example provided.  If the `digitalWrite` in the second loop is commented out, the program continues and the status LED blinks forever. The same effect is seen with the `neopixelWrite` variant.

### Sketch

```cpp
void setup() 
{
  Serial.begin(115200);
  Serial.println("");
  Serial.println("");
}

void loop() 
{
  Serial.println("blink slow");
  for (auto i = 0; i < 10; i++)
  {
     digitalWrite( LED_BUILTIN, i & 1 );
     //neopixelWrite( 47, i & 1, i & 1, i & 1);  // 47 is Lolin S3 onboard LED
     delay( 100 );
  }
    
  Serial.println("blink fast");
  for (auto i = 0; i < 10; i++)
  {
     digitalWrite( LED_BUILTIN, i & 1 );
     //neopixelWrite( 47, i & 1, i & 1, i & 1);
  }
}
```

### Debug Message

```plain
18:25:59.490 -> ESP-ROM:esp32s3-20210327
18:25:59.490 -> Build:Mar 27 2021
18:25:59.490 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
18:25:59.490 -> SPIWP:0xee
18:25:59.490 -> mode:DIO, clock div:1
18:25:59.490 -> load:0x3fce3808,len:0x44c
18:25:59.490 -> load:0x403c9700,len:0xbd8
18:25:59.490 -> load:0x403cc700,len:0x2a80
18:25:59.490 -> entry 0x403c98d0
18:25:59.755 -> 
18:25:59.755 -> 
18:25:59.755 -> blink slow
18:26:00.748 -> blink fast
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

einglis, 2024-03-13:
In fact, we can get even simpler:
```
  Serial.println("blink fast");
  digitalWrite( LED_BUILTIN, 0 );
  //digitalWrite( LED_BUILTIN, 1 ); // uncomment to cause hang
```

In case it's of significance:  most of the time, the LED stays solid white, but sometimes it goes solid blue.

SuGlider, 2024-03-14:
Good news: it works correctly within master (3.0.0).
Bad news: it fils within `version/2.x` branch. Therefore it fails with v2.0.13.

SuGlider, 2024-03-14:
Work around found (it may have relationship with the issue ~ FreeRTOS related)

``` cpp
void setup() 
{
  Serial.begin(115200);
  Serial.println("");
  Serial.println("");
}

void loop() 
{
  Serial.println("blink slow");
  for (auto i = 0; i < 10; i++)
  {
     digitalWrite( LED_BUILTIN, i & 1 );
     //neopixelWrite( 47, i & 1, i & 1, i & 1);  // 47 is Lolin S3 onboard LED
     delay(100);
  }
    
  Serial.println("blink fast");
  for (auto i = 0; i < 10; i++)
  {
     digitalWrite( LED_BUILTIN, i & 1 );
     delay(1);   // <<<<<<<<<<<<<<<<<<  this delay fixes it.
     //neopixelWrite( 47, i & 1, i & 1, i & 1);
  }
}
```

SuGlider, 2024-03-14:
Root cause found... problem is in the `neopixelWrite()` function. I'll post a PR for `release/2.x` branch.

SuGlider, 2024-03-14:
@einglis - Please apply the PR #9374 into your development environment and let me know if it fixes this issue.

einglis, 2024-03-15:
> let me know if it fixes this issue.

Confirmed.  Thanks, @SuGlider.

SuGlider, 2024-03-15:
>ps. I also tried the fix in PlatformIO. I don't know whether I need to raise the same issue with those guys, or if you share fixes.

No need to raise any issue with PlatformIO. They use the code from this repository.
It may be necessary for them to update their packages.

I think that @Jason2866 maintains a set of PIO packages that use to be up to date with the changes we add here.

Jason2866, 2024-03-15:
For core 2.0.x it is easy to use the latest version from here. Just change your setup like this:
```
platform = espressif32 @ 6.5.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#release/v2.x
```

SuGlider, 2024-03-18:
Closed by #9374 - ready for Core 2.0.15 release.

PR #9374 (2024-03-18): Fixes issue related to writing RGB LED fast in a row
## Description of Change

When writing to the same GPIO using RMT, like in a loop, it is necessary to use `rmtWriteBlocking()` instead of `rmtWrite()` to avoid issues with FreeRTOS and IDF 4.4.x.

This PR fixes `neopixelWrite()` (as well as digitalWrite(RGBLED, state)`) in order to avoid such issue.

## Tests scenarios
Tested with ESP32-S3

``` cpp
void setup() 
{
  Serial.begin(115200);
  Serial.println("");
  Serial.println("");
}

void loop() 
{
  Serial.println("blink slow");
  for (auto i = 0; i < 10; i++)
  {
     digitalWrite( LED_BUI [... cut]
closed 2024-03-18 by SuGlider