espressif/arduino-esp32 #9370: Does v3.0.0 support std::variant and other RTTI-dependent features?

Zhu-jiatong, 2024-03-14:
### Related area

language features

### Hardware specification

N/A

### Is your feature request related to a problem?

I am making a JSON-related programme and would likely be using std::variant and other RTTI-related features. I would like to know if RTTI is enabled by default in v3.0.3. Thank you.

### Describe the solution you'd like

Would be great of it is enabled by default.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2024-03-14:
@me-no-dev Can you please help with triage? Thanks

Jason2866, 2024-03-14:
No, it is not enabled by default. Can be seen here https://github.com/espressif/esp32-arduino-libs/blob/idf-release/v5.1/esp32/platformio-build.py
The file is used for Platformio but the settings are taken 1:1 from the build process with the Arduino Lib Builder when compiling the Arduino Libs.

Zhu-jiatong, 2024-03-14:
So 'std::variant' and 'std::any' would be unusable?

Jason2866, 2024-03-14:
yes, but you can enable rtti with`-frtti` by overwriting the default setting (`-fno-rtti`), at least with Platformio.
Idk if this is possible with ArduinoIDE.

SuGlider, 2024-03-14:
For the Arduino IDE it would be necessary to change the `platform.txt` settings by using `platform.local.txt`
For more information: https://arduino.github.io/arduino-cli/0.35/platform-specification/

`platform.local.txt` can be placed in the same folder where a `platform.txt` already is.
1. In the [Arduino15_folder](https://support.arduino.cc/hc/en-us/articles/360018448279-Open-the-Arduino15-folder)/packages/esp32/hardware/esp/<Core_Version>/platform.local.txt
2. In the local cloned Arduino Core used as a `board in sketchbook` at [sketchbook_folder](https://support.arduino.cc/hc/en-us/articles/4412950938514-Open-the-Sketchbook)/hardware/espressif/<Core_Version>/platform.local.txt

These are the default extra flags settings from ESP32 Arduino `platform.txt`.
It is possible to override anything from `platform.txt`, including recipes.
```
# These can be overridden in platform.local.txt
compiler.c.extra_flags=-MMD -c
compiler.cpp.extra_flags=-MMD -c
compiler.S.extra_flags=-MMD -c -x assembler-with-cpp
compiler.c.elf.extra_flags="-Wl,--Map={build.path}/{build.project_name}.map" "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/ld" "-L{compiler.sdk.path}/{build.memory_type}" "-Wl,--wrap=esp_panic_handler"
compiler.ar.extra_flags=
compiler.objcopy.eep.extra_flags=
compiler.elf2hex.extra_flags=
compiler.libraries.ldflags=
```

Zhu-jiatong, 2024-03-15:
Is there any reason as to why this feature is not supported by default?

me-no-dev, 2024-03-15:
There is no particular reason apart from not being enabled by default in IDF and you being the first to request it for the last 8 years :) we will assess any performance/size consequences of enabling it (there are some) and will enable it if everything is OK. Can not give you timeframe, because we need to improve our tooling a bit first

Zhu-jiatong, 2024-03-15:
Should a separate issue for adding default support of RTTI be opened? If so, should it be opened in the IDF repo or this repo?

SuGlider, 2024-03-15:
> Should a separate issue for adding default support of RTTI be opened? If so, should it be opened in the IDF repo or this repo?

This issue will be tracked for Arduino Core 3.0.1. All set.

me-no-dev, 2024-05-31:
Increases flash size with 2-7KB depending on chip and features.

VojtechBartoska, 2024-06-05:
Hello, after discussion we agreed that we will not implement this. Above mentioned PR have been closed, I am closing this ticket as wont do.

closed 2024-06-05 by VojtechBartoska