espressif/arduino-esp32 #9375: Rebooting and Unable to use two ADS1115 in single I2C port

Murugesh-Hobbyist, 2024-03-15:
### Board

ESP32S3

### Device Description

ESP32S3 connected with 2-ADS1115 @ 39 and 40th pins for I2C
one with (0x48) another with (0x49)

### Hardware Configuration

for I2C,
#define SDA0_Pin 39
#define SCL0_Pin 40

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows-11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

I can point the address of those ADC boards and read the analog values. but if I call them in a if statement, it throws error for the second ADS1115.
address for the first one, im using (0x48) and the seond with, (0x49)
if i use the board with (0x48), i get no issues but if i fetch for the (0x49)
im getting 

_Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420021cf  PS      : 0x00060830  A0      : 0x8200207c  A1      : 0x3fcebdf0  
A2      : 0x003c0301  A3      : 0x3fc95392  A4      : 0x00000003  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x820024a4  A9      : 0x3fcebdd0  
A10     : 0x3fc91e7c  A11     : 0x3c03302c  A12     : 0x00000002  A13     : 0xffffffff  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x003c030d  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x420021cc:0x3fcebdf0 0x42002079:0x3fcebe10 0x420020ab:0x3fcebe30 0x4200215a:0x3fcebe50 0x42001b9d:0x3fcebe70 0x42001c9e:0x3fcebe90 0x42003f2d:0x3fcebeb0

ELF file SHA256: a29005a4d446cba5
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403770f0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0_
[... 288 lines cut ...]
### Debug Message

```plain
**and when decoded to human readable through exception decoder,**

_PC: 0x420021cf: Adafruit_I2CDevice::begin(bool) at C:\Users\Vasanth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\Wire\src/Wire.h line 92
EXCVADDR: 0x003c030d

Decoding stack results
0x420021cc: Adafruit_I2CDevice::begin(bool) at C:\Users\Vasanth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\Wire\src/Wire.h line 92
0x42002079: Adafruit_ADS1X15::writeRegister(unsigned char, unsigned short) at C:\Users\Vasanth\Documents\Arduino\libraries\Adafruit_ADS1X15\Adafruit_ADS1X15.cpp line 383
0x420020ab: Adafruit_ADS1X15::startADCReading(unsigned short, bool) at C:\Users\Vasanth\Documents\Arduino\libraries\Adafruit_ADS1X15\Adafruit_ADS1X15.cpp line 344
0x42001b9d: readADC() at C:\Users\Vasanth\Desktop\11_Un_Activated_Dual_Core_7inch/11_Un_Activated_Dual_Core_7inch.ino line 178
0x42001c9e: readADC() at C:\Users\Vasanth\Desktop\11_Un_Activated_Dual_Core_7inch/11_Un_Activated_Dual_Core_7inch.ino line 209
0x42003f2d: uartFlushTxOnly at C:\Users\Vasanth\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\cores\esp32\esp32-hal-uart.c line 455_
```

### Other Steps to Reproduce

I tried swapping the boards and changing the address by rewiring and rewriting the codes.
except the board with (0x48), all other addresses causes the same rebooting

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-03-15:
can you please try to make the sketch smaller and just enough to reproduce the problem?

TD-er, 2024-03-15:
Not stating there can't be a problem in Espressif/Arduino code....
However when reading about issues with ADS1x15, my first Pavlov reaction is that it currently is quite hard to get the exact chip you think you ordered or getting genuine ones. (when ordering those on the "purple" boards as sold on sites like AliExpress etc.)
And on top of that most boards also have pull-up resistors mounted for both I2C pins.
If you add multiple of those boards, you essentially make the pull-up stronger and not all I2C devices can pull down hard enough for stable communications.  (you could lower the I2C clock speed as a quick test to see if it changes anything)

One simple test to detect quite a lot of fake chips is to set a fixed voltage on each ADC input.
For example using a few resistors:
`3V3 --R-- Vin --R-- Vin --R-- Vin --R-- Vin --R-- GND`
And then try sampling each input pin sequentially.
You would expect to see the same voltage (+ some noise) being measured on each pin.
A lot of fake ones will sometimes give you the voltage of the wrong pin.

Another thing that happens a lot is that ADS1015 is swapped for ADS1115.
One is 12 bit at higher sample rate, the other is 16 bit.
If you expect the 16 bit version, check the last few bits of the data.
Or if you expect the 12 bit, try to see if there are more bits present not set to 0.

So unless you bought the boards from sellers like Adafruit or parts directly from large distributors like Mouser, Digikey, etc., my first bet would be to double check you have a genuine one.
There really is a lot of garbage out there and I have lost way too many hours debugging my code only to find out it was fake garbage someone bought.

TD-er, 2024-03-15:
And about your code:

```c++
    while (dwinSerial.available() > 0)  // empty the buffer into our array
    {
      char incomingByte = dwinSerial.read();  // read 1 character from the dwin display
      delay(2);
      incomingData[i] = incomingByte;
      i++;
    }
    if (incomingData[3] == (byte)0x83)  // VP read instruction is in the returned char array
    {
      int k = 7;
      int j = 0;

      while (incomingData[k] != 0xFF) {
        receivedFloatArray[j] = incomingData[k];
        k++;
        j++;
      }
```
I don't see any check on array boundaries.

Murugesh-Hobbyist, 2024-03-16:
Mates, Thanks for the long response! 

Actually in this code, Im connecting DWIN display to show 8 linear displacement sensor values and getting the user inputs to ESP32S3.
unable to believe how it is resolved.

when the user resets in the display, HEX value from the display for zero should be 
5A A5 08 83 20 00 02 30 FF FF 00 --> text "0"
Instead i received 
5A A5 08 83 20 00 02 29 30 FF FF --> text ")0"

now its modified and no reboot happening.
my doubt is, the MC handled it when single ADC is connected but not at 2.
It should failed when i connecting single ADC itself.

closed 2024-06-02 by Murugesh-Hobbyist