espressif/arduino-esp32 #9376: OTA aborted (IDFGH-12341)

hitecSmartHome, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.6 2nd stage bootloader

### Espressif SoC revision.

ESP-32 Wrover-E revision v3.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP-32 Wrover-E

### Power Supply used.

External 3.3V

### What is the expected behavior?

OTA should start normally.

### What is the actual behavior?

OTA does not start, instead it aborts.

### Steps to reproduce.

Basic Arduino OTA sketch.

### Debug Logs.

```plain
Uploading .pio\build\esp-wrover-kit\firmware.bin
13:39:45 [DEBUG]: Options: {'esp_ip': 'esp', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 35818, 'auth': 'test', 'image': '.pio\\build\\esp-wrover-kit\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
13:39:45 [INFO]: Starting on 0.0.0.0:35818
13:39:45 [INFO]: Upload size: 2115056
Sending invitation to hsh 
Authenticating...OK
13:39:46 [INFO]: Waiting for device...
13:39:56 [ERROR]: No response from device
*** [upload] Error 1

[1870023][D][Updater.cpp:133] begin(): OTA Partition: app1

[OTA] - Starting.
[1873026][I][WiFiClient.cpp:260] connect(): select returned due to timeout 3000 ms for fd 54

[1873027][E][Updater.cpp:284] end(): premature end: res:0, pos:0/2115056

[1873040][E][ArduinoOTA.cpp:356] _runUpdate(): Update ERROR: Aborted
```

Jason2866, 2024-03-16:
Example sketch is missing to reproduce.

hitecSmartHome, 2024-03-18:
Sorry, it was a firewall issue.

closed 2024-03-18 by hitecSmartHome