espressif/arduino-esp32 #9377: USB USBHIDKeyboard::pressRaw() modifier key not working

EmileSpecialProducts, 2024-03-15:
### Board

ESP32-S3

### Device Description

ESP32-S3

### Hardware Configuration

USB connection on ESP32-S3

### Version

latest master (checkout manually)

### IDE Name

PlatformIO / not relavant

### Operating System

Win

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

USB

### Description

The modifiers are not working for the USBHIDKeyboard::pressRaw(uint8_t k)/ USBHIDKeyboard::releaseRaw(uint8_t k)
The modifiers are between 0xe0 and 0xe8  this will represent the Bit in the _keyReport.modifiers.
Then you will need to subtract 0xe0 and not 0x80

 Original code:
```
    if (k >= 0xE0 && k < 0xE8) {
        // it's a modifier key
        _keyReport.modifiers |= (1<<(k-0x80));

```
Needs to be 

```
    if (k >= 0xE0 && k < 0xE8) {
        // it's a modifier key
        _keyReport.modifiers |= (1<<(k-0xE0));

```

### Sketch

```cpp
not provided.
```

### Debug Message

```plain
There is not debug information
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

EmileSpecialProducts, 2024-03-15:
I issue is in this file:
https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/src/USBHIDKeyboard.cpp

P-R-O-C-H-Y, 2024-03-15:
Thanks @EmileSpecialProducts for reporting. I will take a look.

SuGlider, 2024-04-09:
>The modifiers are not working for the USBHIDKeyboard::pressRaw(uint8_t k)/ USBHIDKeyboard::releaseRaw(uint8_t k)
The modifiers are between 0xe0 and 0xe8 this will represent the Bit in the _keyReport.modifiers.
Then you will need to subtract 0xe0 and not 0x80

This is correct. The modifiers are marked in a bitmap in the HID Report Descriptor. 
```
Bit:        7   6   5   4   3   2   1   0
          +---+---+---+---+---+---+---+---+
Byte 0    | RG| RA| RS| RC| LG| LA| LS| LC|  Modifier bits (LC=Left Control, LS= Left Shift, etc)
          +---+---+---+---+---+---+---+---+
Byte 1    |        Reserved byte          |
          +---+---+---+---+---+---+---+---+
Byte 2    |        Key 1                  |
          +---+---+---+---+---+---+---+---+
Byte 3    |        Key 2                  |
          +---+---+---+---+---+---+---+---+
Byte 4    |        Key 3                  |
          +---+---+---+---+---+---+---+---+
Byte 5    |        Key 4                  |
          +---+---+---+---+---+---+---+---+
Byte 6    |        Key 5                  |
          +---+---+---+---+---+---+---+---+
Byte 7    |        Key 6                  |
          +---+---+---+---+---+---+---+---+
```

SuGlider, 2024-04-09:
It shall be fixed with https://github.com/espressif/arduino-esp32/pull/9473

PR #9473 (2024-04-10): fix: USB HID Keyboard raw report
## Description of Change
Fixes key modifier bitmap when using `USBHIDKeyboard::pressRaw(uint8_t k)` and `USBHIDKeyboard::releaseRaw(uint8_t k)`.

The modifiers are marked in a bitmap in the HID Report Descriptor. 
```
Bit:        7   6   5   4   3   2   1   0
          +---+---+---+---+---+---+---+---+
Byte 0    | RG| RA| RS| RC| LG| LA| LS| LC|  Modifier bits (LC=Left Control, LS= Left Shift, etc)
          +---+---+---+---+---+---+---+---+
Byte 1    |        Reserved byte          |
          +---+---+---+---+---+---+---+---+
Byte 2    |        Key 1          [... cut]
closed 2024-04-10 by me-no-dev