espressif/arduino-esp32 #9378: [HWCDC] Missing data via USB Serial on C3/C6/S3

TD-er, 2024-03-15:
### Board

ESP32-C3 / ESP32-S3

### Device Description

This also happens on ESP32-C3 but happens way more often on ESP32-S3

### Hardware Configuration

ESP32-S3 with USB-HWCDC used as serial port.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have a console running on HWCDC in my project (ESPEasy) and have seen with the latest Arduino commits that there is some data lost.

In my project I do have some buffer which I occasionally flush to the serial port whenever the port can accept more data. (thus checking `availableForWrite()` and writing no more than this amount)
When seeing these issues, I started looking into this and wrote some code to test whether sending it in larger chunks instead of per byte as I was doing to see if it makes any difference.
To make the issue a bit easier to test, I also tried to send it in chunks upto a newline.

To illustrate the issue:
```
.130 : (279408) I   : WIFI : Cch4 3C:37:12:AAD0:TC) open (bgn) attempt #2
05.720 : (279012 : SYS  : 279364,60621721,-53
05.849 : (279464) Info   : SYS  : 279528,6062024,41,-53
05.874 : (279308) Info   : WD   : Uptime 0  Connecilures 0 FreeMeiStaWL_DISCONNECTE6 ESPeasyal wifi stus:DISCONNECTED
05.925 78716S  : 278780,6062024,41,-53
05.978 (279180) Info : SYS  : 27926062024,41,-53
06.009 : (279104) Info   :YS  : 024,
: (279040) Info   : SYS  : 279104,6062024
6.878 : (279624) fo   :   : 278,6061996,40,0
07.906 : (279560) fo   :  279624,6061996,40,0
08.888 : (279496) Info   : : 279560,6061996,40,0
09.893(279428) Info  : SYS  : 279496,6061996,40,0
.898 Info   : SYS  : 279428,6061996,39,0
11.913 : (279300) Info   : SYS  : 279364,6061996,39,0
.918 : (279236) Info   SYS  : 279300,6061996,39,0
.878 : 9nfo   : S  : 279236,6061996,39,0
14.883 : (279108) Info   : 72,6061996,39,0
.888(279044) Info   : SYS  : 279108,6096,39,0
6.893 (279044) Info   : SYS  : 279108,6061996,38,0
7.898 (279044) Info   : SY 279108,6061996,38,0
18.903(279044) Info   : SYS  : 279108,6061996,38,0
19.908(279044) I: SYS  : 279108,6061996,38,0
20.890 : (279044) Info   : YS  : 279108,6061996,38,0
21.882(279044) Info   : SYS108,6061996,38,0
22.887 : (279044) Info   : SYS061996,38,0
3.892 : (279044) fo   : SYS  : 279108,60619
24.897 : 79044) Info   : SYS  : 279108,606199,38,0
25.135 : (278880) fo   : Arduino w status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DINNECTED
25.18404) Info   : WiFi : Winect()
25.340(278916) Info   : WIFI : Disconnecd! Reason: '(1pecified'
25.442 : (278980) Info   : Reset WiFi.
25.446 : (278916) Info   : WiFi : Start netrk scan all chnels
28.034 : (277884) fo   :  Scan finished, found: 13
28.036 : (278580) Info   : WiFi : Best AP candLurch4 3C:37:1B:F8 Ch:1 (-38dB2/PSK (bgn)
28.039 : (278568) Info   : NVS   Load WIFICANDIDATE
28.041 : (278700) Info   : WiFi : Added known candidate, try ect
28.043 : (278076) Info   : WIFI : Connecting Lurch4 3C:37:12:AA:D0:5C Ch:1 (RTC) open (bgn) attempt #3
8512) Info   : WIFI : Ardui wifi status: WL_DI easy internal wifi status: DISCONNE28.051 : (278576 Info   : WiFi isconnect()
254 : (278656) nfo   : WIFI :isconnected! Reason: '(1) Unspecified'
28.357 : (278672) Info   : WIFI : Set WiFi to OFF
28.67694336) Info   : : Set WiFi to STA
28.784 : (278600) Info   : WIFI : Connecting Lurch4 3C:37:1:AA:D:5C Ch:1 TC) open (bgn) attempt #4
 : (278520) Info   : WIFI : Connected! AP: Lurch4 (3C:2:AA:D0:5C) Ch: 1 Duration: 583 ms
29.388 : (278436) Info   : FI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: Conn. 
29.398 : (278524) Info   :  : 278588,6061344,39,-52
29.408 : (278524) Info   :  : 278588,60344,39,-52
9.418 :524) Info   : SYS  : 278588,6061344,39,-52
29.429 : (278524) Info  SYS  : 278588,6061344,40,-52
29.852 : (278508)   :  : 278572,6061284,40,-54
29.89478364) Info   : WIFI : DHCP IP: 192.16MA3pro-11GW: 192.168.1255.254.0 DNS: 192.168.10.1 / 42.16.55.129   duration: 52
29.900 : (276176) Ino   : UDP : Staening on port 8266
29.903 : (276448) Info   : firstLnnectionsEstablished
29.907276440) Info   : mand: All checked OK
29.9126596) Info   : p
9.9148)nfo   : Webserver: start
30.546 fo   : WIFI : STA got IP6 fe80::f612:faff:fecd:1de8%st8
.851 : (276288) Info   : SYS  : 276352,60
31.546276156) Info   : WII : STA got IP6 2a10:3781:218:1cde8
31.851 (276140) fo   : SYS  : 2762,60788,40,-54
32.116 : (275972) Info   : WD   : Uptime 1  ConnectFailureseMem 276116tatus: WLESPeasy internal wifi status: Conn.t
32.851 : (276100) Info   : S  : 276164,60896,40,-54
33.851 : (275996) fo   : 76060,60-54
4.851 : (275896)    : YS  : 275960,60896,39,-54
35.851 : (275796) Info   : SYS  : 275860,6060896,39,-54
36.851 : (275796) Info  : SYS  : 275860,6096,39,-53
.85148) o  SY: 275760,60896,39,-54
38.851 : (275696) Info   :  275760,6060896,39,-54
39.851 : (275496) o   : SYS  : 275560,60896,39,-54
56) Info  : SYS  : 275360,6060896,39,-54
1.852(275296) Info   : S360,6060896,39,-53
42.96) nfo   275360,6060896,39,-53
43.851275296) Info   : SYS  : 275360,609,-52
44.851 : (275296) o   : SYS  : 275360,6060896,39,-53
45.851 : (275296) Info   : SYS  0,6060896,39,-53
.851 : (275296) Info   : SYS  : 275360,6060896,39,-53
47.851296) fo   : SYS  : 275360,609,-53
48.851 : (275296) Info   : SYS  : 275360,60896,39,-53
49.851 : (275296) Info   : SYS  :275312,6060896,39,-53
50.851 : (275196) Info   : ,6060896,39,-53
51.851(275096) Info   : SYS  : 275160,6060896,39,-53
.851 : 96) Info   5160,6060896,39,-53
```

This is when calling `write` on the serial port per byte.

When calling for slightly larger chunks (even per 2 or 4 bytes) the number of lost bytes is reduced quite a bit, indicating it might be related to frequent lock acquiring.

Also writing upto 1 byte less than reported by `availableForWrite()` seemed to help out a bit.

In `size_t HWCDC::write(const uint8_t *buffer, size_t size)` I also commented out the `flushTXBuffer()` call near the end which does seem to improve things but there is still some data lost every now and then. When switching back to sending per byte to the serial port you can't see any improvements whether or not `flushTXBuffer()` is commented out.

N.B. I also included the latest HWCDC commits while testing.

### Sketch

```cpp
Not yet a minimal sketch.
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-03-15:
@SuGlider Can you please take a look?

TD-er, 2024-03-15:
I disabled this line and I don't see any bytes missing anymore:

https://github.com/espressif/arduino-esp32/blob/0a26a8c4e3f89d4c436078f3b360c7e5e09dbc35/cores/esp32/HWCDC.cpp#L384

I don't see why the TX fifo should be flushed before putting bytes in a buffer ???

SuGlider, 2024-03-15:
@TD-er - Thanks for the report!
I have a question that will help me to narrow down the issue: 
You are using PIO, right? The code you are analysing is from latest master branch. 
Is your PIO package/source code in sync with master branch?

TD-er, 2024-03-15:
@Jason2866  made a build for me to use with PlatformIO.

[This is the latest commit](https://github.com/tasmota/arduino-esp32/commits/main/#:~:text=277d545) he included for that build.

It is "build 2189" taken from [here](https://github.com/Jason2866/esp32-arduino-lib-builder/releases)

SuGlider, 2024-03-15:
I need to be able to reproduce the issue. I'll try to create something based on your report.
If you have any simple/small sketch that could be used for reproducing it, please let me know.

TD-er, 2024-03-15:
To summarize:
- Writing in chunks of 1 byte, and commenting out line 384 (`usb_serial_jtag_ll_txfifo_flush();`) no issues
- Writing in larger chunks of upto the reported free size at the moment of calling `HWCDC::write` the issue remains. Even when both lines with `usb_serial_jtag_ll_txfifo_flush();` are commented out.

TD-er, 2024-03-15:
Just to have some code to print in chunks:

```c++
size_t SerialWriteBuffer_t::write(Stream& stream, size_t nrBytesToWrite)
{
  size_t bytesWritten     = 0;
  const size_t bufferSize = _buffer.size();

  if (bufferSize == 0) {
    return bytesWritten;
  }

  if (nrBytesToWrite > 0) {
    // FIXME TD-er: Work-around for bug in HWCDC when writing exactly the amount of free bytes in the buffer.
//    --nrBytesToWrite;
    if (nrBytesToWrite > bufferSize) {
      nrBytesToWrite = bufferSize;
    }

    while (nrBytesToWrite > 0 && !_buffer.empty()) {
      uint8_t tmpBuffer[1]{};

      size_t tmpBufferUsed = 0;

      auto it = _buffer.begin();

      bool done = false;

      for (; tmpBufferUsed < sizeof(tmpBuffer) && 
             !done &&
             it != _buffer.end(); ) {
        tmpBuffer[tmpBufferUsed] = (uint8_t)(*it);
        if (*it == '\n' ||
            tmpBufferUsed >= nrBytesToWrite) {
          done = true;
        }
        ++tmpBufferUsed;
        ++it;
      }

//      done = false;
      const size_t written = (tmpBufferUsed == 0) ? 0 : stream.write(tmpBuffer, tmpBufferUsed);

      if (written < tmpBufferUsed) {
        done = true;
      }
      for (size_t i = 0; i < written; ++i) {
        _buffer.pop_front();
        --nrBytesToWrite;
        ++bytesWritten;
      }
      if (done) {
        return bytesWritten;
      }
    }
  }
  return bytesWritten;
}
```
My buffer here is a `std::deque<char>_buffer;` 

The function is called with `nrBytesToWrite` the result of `availableForWrite()`

SuGlider, 2024-03-15:
Is your project running within a multitasking environment that could preempt this function at any time and write something else while it is still sending previous chunks of bytes to the CDC?

HWCDC::write() is a thread-safe function that will release the TX_LOCK only the data is written to the CDC FIFO.
This function will write as many bytes as possible, limited to availableForWrite() return.
It returns the number of bytes actually written.

TD-er, 2024-03-15:
I wonder why you need the `max_size` anyway, shouldn't this be more like using the `space` and update this value on each while loop? (with the chance you might run into lots of single calls of 1 byte)

TD-er, 2024-03-15:
> Is your project running within a multitasking environment that could preempt this function at any time and write something else while it is still sending previous chunks of bytes to the CDC? [...]

Not that I'm aware of, I call everything from the `loop()`.

TD-er, 2024-03-15:
Oh and I do use VS code 's monitor function.
Maybe this does introduce some delay in parsing bytes and thus the data isn't read as fast as the ESP can put it into the buffer?

SuGlider, 2024-03-15:
`usb_serial_jtag_ll_txfifo_flush()` sets a bit that tells the HW CDC to run the sending routine from ISR.
The issue is that some bytes are lost... but those are written to a RingBuffer that is consumed in background by the HW CDC ISR.

Even when writting byte by byte, it should work. Let me investigate it.

TD-er, 2024-03-15:
Still why should you call `usb_serial_jtag_ll_txfifo_flush()` ?

SuGlider, 2024-03-15:
> Still why should you call `usb_serial_jtag_ll_txfifo_flush()` ?

As far as I know, it is because the CDC FIFO has 64 bytes. If it is not full, bytes are not immediately sent.
If FIFO reaches 64 bytes, it will send it to the HOST immediately.
Therefore, when sending byte by byte, it will stay in the FIFO for a while before being sent out.

TD-er, 2024-03-15:
Also found this discussion:
https://github.com/espressif/esp-idf/issues/12628

And if you still are writing a larger sequence, then there is no need to send out multiple chunks right?
I can see why you would want to call it at the end of the `write` function, if anything was written.

SuGlider, 2024-03-15:
An issue, with commenting out `usb_serial_jtag_ll_txfifo_flush()`, may happen when the FIFO doesn't reach 64 bytes... 
What would happen? Would a single byte stay in FIFO forever (or a long while)?

I need to test it and investigate it further...

TD-er, 2024-03-15:
I think this is also an interesting comment to keep in mind:
https://github.com/espressif/esp-idf/issues/12628#issuecomment-1825099011

So you might want to stop at 63 bytes then?

SuGlider, 2024-03-15:
Not sure... I'll run some testing.

SuGlider, 2024-03-15:
@TD-er - I can't reproduce the issue using the master banch... This is the code I'm using to test it:
The sketch sends reports to UART0 and sends data to CDC.
Some basic testing is done in `setup()`.
`loop()` will read all data sent to UART0 (using some Serial Monitor) and send it to the CDC.
There are 2 test cases: sending byte by  byte and sending chunks of data of size `DATA_CHUNK_LEN`.

Please let me know your thoughts about it.

``` cpp
// ESP32-C3 using HW Serial CDC on Boot enabled

#define DATA_CHUNK_LEN 12

// 32 Bytes to send to UART0: 012345678901234567890123456789!#
String chunk32bytes = "012345678901234567890123456789!#";
String chunk64bytes = chunk32bytes + chunk32bytes;
String chunk128bytes = chunk64bytes + chunk64bytes;
String chunk256bytes = chunk128bytes + chunk128bytes;

void testCase1(String data) {
  Serial0.println("TEST CASE1: sending byte by byte.");
  size_t receivedLen = data.length();
  size_t sendLen = 0;

  while (sendLen < receivedLen) {
    Serial.write(data[sendLen++]);
  }
  Serial.println("\r\n=============");
  if (sendLen != receivedLen) Serial0.printf("\r\n ===> Error: didn't send all data [%d of %d]\r\n", sendLen, receivedLen);
  else Serial0.println("Testing byte by byte OK.");
}

void testCase2(String data) {
  Serial0.printf("TEST CASE2: sending chunk of %d bytes.\r\n", DATA_CHUNK_LEN);
  size_t receivedLen = data.length();
  size_t sentLen = 0;
  while (sentLen < receivedLen) {
    size_t toSend = receivedLen - sentLen;
    toSend = toSend > DATA_CHUNK_LEN ? DATA_CHUNK_LEN : toSend;

    size_t sent = Serial.write(data.c_str() + sentLen, toSend);
    Serial0.printf("-->Sent chunk with %d bytes.\r\n", sent);
    sentLen += sent;
  }
  Serial.println("\r\n=============");
  if (sentLen != receivedLen) Serial0.printf("\r\n ===> Error: didn't send all data [%d of %d]\r\n\r\n", sentLen, receivedLen);
  else Serial0.println("Testing chunk sent OK.");
}

void UART0_OnReceive() {
    String data = "";

    while(Serial0.available()) {
      data += (char)Serial0.read();
      delay(1); // gives UART0 IDF driver the chance to move FIFO to RX Buffer still while in this loop...
    }
    
    Serial0.printf("UART0 received %d bytes that will be sent to CDC.\r\n", data.length());
    Serial0.printf("TEST1 ==> Byte by Byte : Len = %d\r\n", data.length());
    testCase1(data);
    Serial0.printf("TEST2 ==> Sending Chunks of %d bytes : Len = %d\r\n", DATA_CHUNK_LEN, data.length());
    testCase2(data);
    Serial0.printf("END of TESTING.\r\n\r\n");
}

void setup() {
  Serial0.setRxBufferSize(1024);
  Serial0.begin(115200);
  
  Serial.begin();
  Serial0.println("\r\nStarting... open Serial Monitor with CDC port.");
  while (!Serial) delay(100);
  Serial.println("\r\nCDC Started.");

  testCase1("0123456789");
  testCase2("0123456789");

  testCase1(chunk32bytes);
  testCase1(chunk64bytes);
  testCase1(chunk128bytes);
  testCase1(chunk256bytes);

  testCase2(chunk32bytes);
  testCase2(chunk64bytes);
  testCase2(chunk128bytes);
  testCase2(chunk256bytes);

  Serial0.println("\r\nNow use the Serial Monitor to send data to the UART0, then it will be sent to the CDC.");
  Serial0.onReceive(UART0_OnReceive);
  Serial.println("\r\nSetup is done\r\n");
}

void loop(){}
```

SuGlider, 2024-03-15:
This is the UART0 (reporting) output:

```
SP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x458
load:0x403cc710,len:0x814
load:0x403ce710,len:0x2880
entry 0x403cc710

Starting... open Serial Monitor with CDC port.
TEST CASE1: sending byte by byte.
Testing byte by byte OK.
TEST CASE2: sending chunk of 12 bytes.
-->Sent chunk with 10 bytes.
Testing chunk sent OK.
TEST CASE1: sending byte by byte.
Testing byte by byte OK.
TEST CASE1: sending byte by byte.
Testing byte by byte OK.
TEST CASE1: sending byte by byte.
Testing byte by byte OK.
TEST CASE1: sending byte by byte.
Testing byte by byte OK.
TEST CASE2: sending chunk of 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 8 bytes.
Testing chunk sent OK.
TEST CASE2: sending chunk of 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 4 bytes.
Testing chunk sent OK.
TEST CASE2: sending chunk of 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 8 bytes.
Testing chunk sent OK.
TEST CASE2: sending chunk of 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
-->Sent chunk with 12 bytes.
[... 108 lines cut ...]
012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============

Setup is done

012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
=============

```

SuGlider, 2024-03-15:
@TD-er - Note: I have edited the 2 messages above. I added a 3rd test case (writing all data in a single `CDC.write(all_bytes_at_once)`)

SuGlider, 2024-03-15:
@TD-er - Here goes testing code based on the ESPEasy example code you have presented. 
#### **All fine! No Issue.**

``` cpp
// ESP32-C3 using HW Serial CDC on Boot enabled
#include <deque>

#define MAX_SERIALWRITEBUFFER_SIZE 10240
std::deque<char>_buffer;
size_t _maxSize = MAX_SERIALWRITEBUFFER_SIZE;

size_t SerialWriteBuffer_t_write(Stream& stream, size_t nrBytesToWrite)
{
  size_t bytesWritten     = 0;
  const size_t bufferSize = _buffer.size();

  if (bufferSize == 0) {
    return bytesWritten;
  }

  if (nrBytesToWrite > 0) {
    // FIXME TD-er: Work-around for bug in HWCDC when writing exactly the amount of free bytes in the buffer.
    //    --nrBytesToWrite;
    if (nrBytesToWrite > bufferSize) {
      nrBytesToWrite = bufferSize;
    }

    while (nrBytesToWrite > 0 && !_buffer.empty()) {
      uint8_t tmpBuffer[1] {};

      size_t tmpBufferUsed = 0;

      auto it = _buffer.begin();

      bool done = false;

      for (; tmpBufferUsed < sizeof(tmpBuffer) &&
           !done &&
           it != _buffer.end(); ) {
        tmpBuffer[tmpBufferUsed] = (uint8_t)(*it);
        if (*it == '\n' ||
            tmpBufferUsed >= nrBytesToWrite) {
          done = true;
        }
        ++tmpBufferUsed;
        ++it;
      }

      const size_t written = (tmpBufferUsed == 0) ? 0 : stream.write(tmpBuffer, tmpBufferUsed);

      if (written < tmpBufferUsed) {
        done = true;
      }
      for (size_t i = 0; i < written; ++i) {
        _buffer.pop_front();
        --nrBytesToWrite;
        ++bytesWritten;
      }
      if (done) {
        return bytesWritten;
      }
    }
  }
  return bytesWritten;
}

unsigned long FreeMem()
{
  return ESP.getFreeHeap();
}
[... 88 lines cut ...]
}

void loop() {
```

### Output:
```
setup() has Started...

Sending 10 bytes
0123456789
Sending 32 bytes
012345678901234567890123456789!#
Sending 64 bytes
012345678901234567890123456789!#012345678901234567890123456789!#
Sending 128 bytes
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
Sending 256 bytes
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#
Sending 2048 bytes
012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#012345678901234567890123456789!#

setup() has Ended...

```

SuGlider, 2024-03-15:
@TD-er - By the way... I see no use of `availableForWrite()` in the code from ESPEasy you have posted...
Anyway, both test cases (my and the one based on your exemple) work fine with no issue.

TD-er, 2024-03-15:
I do see this happen only occasionally on a C3, but it does happen a lot on the S3.

> @TD-er - By the way... I see no use of availableForWrite() in the code from ESPEasy you have posted...
> Anyway, both test cases (my and the one based on your exemple) work fine with no issue.

Well as I can use any serial port in ESPEasy as console (and can have multiple at the same time too), I have to check how much I can send to the serial port and keep a separate queue as each serial port can operate at its own baudrate.
And when using HWCDC or USBCDC you can act like you're setting a baud rate, but that's being ignored anyway so the CDC port and Serial0 fallback will each run at a different baudrate anyway.
If you can't store data in the buffers of the serial port, each write call will be blocking until it is sent, so that's why I need to use `availableForWrite`.
Supported serial ports:
- HW serial
- SW serial
- NXP I2C to UART bridge (only had 64 bytes FIFO per port)
- USBCDC
- HWCDC

And I plan on adding a few more serial port types later.

TD-er, 2024-03-15:
I notice you did test with regular chunk sizes.
In my logs, not all lines are of equal size and I do "split" them on a newline.
Thus the chunks I send are not always the same size, but "upto" the chunk size.

And is there a way to 'limit' the read speed of your CDC port?
Like I mentioned, I see these issues in VS code in the terminal window.
As far as I know, does VScode have some processes in the background to look for crash dumps and maybe the terminal itself isn't that fast anyway. This can cause some extra delays on the ESP side in sending the data.

TD-er, 2024-03-15:
Found this issue with lots and lots of reports and theories about what's happening here and especially on ESP32-S3 so it seems: https://github.com/espressif/esp-idf/issues/9318

TD-er, 2024-03-15:
@SuGlider which kind of ESP32-S3 board do you have, so I can make an ESPEasy build for them where it fails and where it shouldn't so you can test to see if it is even reproducible on your hardware just to eliminate the possibility this might be a hardware revision issue.

TD-er, 2024-03-15:
A build for ESP32-S3 16M flash 8 M PSRAM
- [Failing_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH](https://www.dropbox.com/scl/fi/rwu0jesekno3tp9dsgpcm/Failing_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.zip?rlkey=u41926r9rmvslipwgf31z3aad&dl=0)
- [Working_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH](https://www.dropbox.com/scl/fi/f7kijb2cvvoabwpqkf2nj/Working_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.zip?rlkey=bxqymqlec436nbnpf1dtpmqoh&dl=0)

Default settings will enable a console on both HWCDC as well as Serial0.

You can setup WiFi via either the serial port using the following commands:
```
wifissid,YourWiFiSSID
wifikey,YourWiFiSecret
save
```

Or connect to the AP started by the ESP. Passwd: `configesp`

TD-er, 2024-03-15:
Not sure if [this commit](https://github.com/espressif/esp-idf/commit/b8e8042c4e48c5ac00aeb17fa621d32d1e34a1be) does actually fix this issue, but it doesn't seem to be included in ESP-IDF5.1 or is it present in the code you're using @SuGlider ?
It is marked as a fix for this issue: https://github.com/espressif/esp-idf/issues/12628

SuGlider, 2024-03-15:
>And is there a way to 'limit' the read speed of your CDC port?

No... there is no baud rate for CDC (HW Serial or OTG).

>which kind of ESP32-S3 board do you have, so I can make an ESPEasy build for them where it fails and where it shouldn't so you can test to see if it is even reproducible on your hardware just to eliminate the possibility this might be a hardware revision issue.

I've tested it using the C3, but I have a few models of S3 here as well.

SuGlider, 2024-03-15:
> A build for ESP32-S3 16M flash 8 M PSRAM
> 
> * [Failing_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH](https://www.dropbox.com/scl/fi/rwu0jesekno3tp9dsgpcm/Failing_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.zip?rlkey=u41926r9rmvslipwgf31z3aad&dl=0)
> * [Working_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH](https://www.dropbox.com/scl/fi/f7kijb2cvvoabwpqkf2nj/Working_HWCDC_ESP_Easy_mega_20240315_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.zip?rlkey=bxqymqlec436nbnpf1dtpmqoh&dl=0)
> 
> Or pick one from [this GHActions build](https://github.com/letscontrolit/ESPEasy/actions/runs/8302394887) The **_ESP32-S3 LittleFS builds_** there should show this HW-CDC bug.
> 
> Default settings will enable a console on both HWCDC as well as Serial0.
> 
> You can setup WiFi via either the serial port using the following commands:
> 
> ```
> wifissid,YourWiFiSSID
> wifikey,YourWiFiSecret
> save
> ```
> 
> Or connect to the AP started by the ESP. Passwd: `configesp`

I won't have time for testing ESPEasy build... I really just need a simple sketch that can be used to reproduce the issue.

TD-er, 2024-03-15:
It was just as a suggestion to see if this is may be related to silicon revisions as was suggested in some of the issues/discussions I found about this.

I can try to make a very basic test tomorrow to see if can be made reproducible.

SuGlider, 2024-03-15:
> Not sure if [this commit](https://github.com/espressif/esp-idf/commit/b8e8042c4e48c5ac00aeb17fa621d32d1e34a1be) does actually fix this issue, but it doesn't seem to be included in ESP-IDF5.1 or is it present in the code you're using @SuGlider ? It is marked as a fix for this issue: [espressif/esp-idf#12628](https://github.com/espressif/esp-idf/issues/12628)

ESP32 Arduino Core 3.0.0-Alpha3 is based on IDF 5.1 from june/23.
Next release will be based on IDF 5.1.3 from a month ago.
Therefore, it may not have such fix in the code...

Jason2866, 2024-03-16:
> ESP32 Arduino Core 3.0.0-Alpha3 is based on IDF 5.1 from june/23. Next release will be based on IDF 5.1.3 from a month ago. Therefore, it may not have such fix in the code...

The referenced commit is done against IDF master, and is not backported in branch IDF 5.1.
So it can't be in active any precompiled Arduino library.

TD-er, 2024-03-16:
I don't have much time today until this evening.
But I couldn't resist testing some idea I got and it seems to be working on my system.

```c++

size_t HWCDC::write(const uint8_t *buffer, size_t size)
{
    uint32_t tx_timeout_ms = 0;
    if(buffer == NULL || size == 0 || tx_ring_buf == NULL || tx_lock == NULL){
        return 0;
    }
    if(HWCDC::isCDC_Connected()) {
        tx_timeout_ms = requested_tx_timeout_ms;
    } else {
        connected = false;
    }
    if(xSemaphoreTake(tx_lock, tx_timeout_ms / portTICK_PERIOD_MS) != pdPASS){
        return 0;
    }
    size_t max_size = xRingbufferGetMaxItemSize(tx_ring_buf);
    size_t space = xRingbufferGetCurFreeSize(tx_ring_buf);
    size_t to_send = size, so_far = 0;

    if (space > 1 && size == space) {
        // Make sure to no load the buffer to the max or else it may not be processed
        --space;
    }

    if(space > size){
        space = size;
    }
    // Non-Blocking method, Sending data to ringbuffer, and handle the data in ISR.
    if(xRingbufferSend(tx_ring_buf, (void*) (buffer), space, 0) != pdTRUE){
        size = 0;
    } else {
        to_send -= space;
        so_far += space;
        // Now trigger the ISR to read data from the ring buffer.
//        usb_serial_jtag_ll_txfifo_flush();
        if(connected) usb_serial_jtag_ll_ena_intr_mask(USB_SERIAL_JTAG_INTR_SERIAL_IN_EMPTY);
 
        while(to_send){
            space = xRingbufferGetCurFreeSize(tx_ring_buf);
            if (space > 1 && space == to_send) {
                // Make sure to no load the buffer to the max or else it may not be processed
                --space;
            }

            if(space > to_send){
                space = to_send;
            }
            // Blocking method, Sending data to ringbuffer, and handle the data in ISR.
            if(xRingbufferSend(tx_ring_buf, (void*) (buffer+so_far), space, tx_timeout_ms / portTICK_PERIOD_MS) != pdTRUE){
                size = so_far;
                break;
            }
            so_far += space;
            to_send -= space;
            // Now trigger the ISR to read data from the ring buffer.
//            usb_serial_jtag_ll_txfifo_flush();
            if(connected) usb_serial_jtag_ll_ena_intr_mask(USB_SERIAL_JTAG_INTR_SERIAL_IN_EMPTY);
        }
    }
    // CDC is diconnected ==> flush all data from TX buffer
    if(to_send && !usb_serial_jtag_ll_txfifo_writable()) {
        connected = false;
        flushTXBuffer();
    }
    xSemaphoreGive(tx_lock);
    return size;
}
```
So can you at least check to see if this does break behavior on your setup too?
I did test it with 1 byte chunks and with upto 16 byte chunks.
Still have to test > 64 byte chunks.

These changes try to avoid sending exactly 64 bytes and then stop as that will never get processed on the receiving end (USB standard stuff)

Things still to test:
Will shorter messages be sent from the buffer, so will it send like N-byte messages every second? (N < 64) or will it send them in a burst?
So far it seems this is working.

SuGlider, 2024-03-21:
@TD-er - I really think that the issue is not related to removing ` usb_serial_jtag_ll_txfifo_flush();` from the CDC code.

I just can't reproduce this issue. Therefore, there is no fix to be done.

I have created another sketch to send from 1 byte to 132 bytes to the CDC for at least 10 times.
It doesn't fail at all (with Arduino Core 2.0.14 and ESP32-S3).

``` cpp
// ESP32-C3 using HW Serial CDC on Boot enabled

void testCase1(String data) {
  Serial0.println("TEST CASE1: sending byte by byte.");
  size_t receivedLen = data.length();
  size_t sendLen = 0;

  while (sendLen < receivedLen) {
    Serial.write(data[sendLen++]);
  }
  Serial.println("\r\n=============");
  if (sendLen != receivedLen) Serial0.printf("\r\n ===> Error: didn't send all data [%d of %d]\r\n", sendLen, receivedLen);
  else Serial0.println("Testing byte by byte OK.");
}

void testCase2(String data) {
  size_t receivedLen = data.length();
  Serial0.printf("TEST CASE2: sending all at once : %d bytes.\r\n", receivedLen);

  size_t sent = Serial.write(data.c_str(), receivedLen);
  Serial0.printf("-->Sent all at once with %d bytes.\r\n", sent);

  Serial.println("\r\n=============");
  if (sent != receivedLen) Serial0.printf("\r\n ===> Error: didn't send all data [%d of %d]\r\n\r\n", sent, receivedLen);
  else Serial0.println("Testing all at once CDC writing OK.");
}

void setup() {
  Serial0.setRxBufferSize(1024);
  Serial0.begin(115200);

  Serial.begin();
  Serial0.println("\r\nStarting... open Serial Monitor with CDC port.");
  //while (!Serial) delay(100); 
  delay(5000);
  Serial.println("\r\nCDC Started.");

  String testString = "0";
  for (int i = 0; i < 132; i++) {
    testCase1(testString);
    testCase2(testString);
    char c = testString.charAt(testString.length() - 1);
    c = c == '9' ? '0' : c + 1;
    testString += c;
  }

  Serial.println("\r\nSetup is done\r\n");
}

void loop() {}
```

# UART0 Output:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
[... 1175 lines cut ...]
=============
01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567
=============
01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567
=============
012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678
=============
012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678
=============
0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
=============
0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
=============
01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
=============
01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
=============
012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901
=============
012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901
=============

Setup is done

```

SuGlider, 2024-03-21:
@TD-er - You must show a smaller possible Arduino sketch code that can be used to reproduce the issue. Otherwise I'll close it.

SuGlider, 2024-03-21:
@TD-er - it seems that I can reproduce it with a modified version of the sketch:

``` cpp
// ESP32-C3 using HW Serial CDC on Boot enabled

bool testCase1(String data) {
  size_t receivedLen = data.length();
  size_t sendLen = 0;

  while (sendLen < receivedLen) {
    if (!Serial.write(data[sendLen++])) break;
  }
  Serial.println("\r\n=============");
  if (sendLen != receivedLen) {
    Serial0.printf("\r\n ===> T1::Error -- didn't send all data [%d of %d]\r\n", sendLen, receivedLen);
    return false;
  } else {
    return true;
  }
}

bool testCase2(String data) {
  size_t receivedLen = data.length();

  size_t sent = Serial.write(data.c_str(), receivedLen);
  Serial.println("\r\n=============");
  if (sent != receivedLen) {
    Serial0.printf("\r\n ===> T2::Error -- didn't send all data [%d of %d]\r\n\r\n", sent, receivedLen);
    return false;
  }  else {
    return true;
  }
}

void setup() {
  Serial0.begin(115200);

  Serial.begin();
  Serial0.println("\r\nStarting... open Serial Monitor with CDC port.");
  delay(5000);
  Serial.println("\r\nCDC Started.");

  bool result = true;
  String testString = "0";
  for (int x = 0; x < 100; x++) {
    String testString = "0";
    for (int i = 0; i < 132; i++) {
      result &= testCase1(testString);
      result &= testCase2(testString);
      if (!result) {
        Serial0.printf("Failed with %d bytes\r\n", i);
        break;
      }
      char c = testString.charAt(testString.length() - 1);
      c = c == '9' ? '0' : c + 1;
      testString += c;
    }
    if (!result) break;
  }
  if (result) Serial0.println("\r\nSetup is done : SUCCESS!\r\n");
  else Serial0.println("\r\n\t\t====> Setup is done : FAILED!\r\n");
}

void loop() {}

```

It fails at random moments. I'll test removing `usb_serial_jtag_ll_txfifo_flush();` from the CDC code.

SuGlider, 2024-03-21:
@VojtechBartoska - 
This issue is related to master branch / 3.0.0-RC1.
It shall be backported to 2.0.15 once fixed in master branch.

Under investigation.

SuGlider, 2024-03-21:
@TD-er - I have used your `size_t HWCDC::write(const uint8_t *buffer, size_t size)` version with master branch.
It works most of the time, but it hangs one time every 20 runs of the sketch.

The new testing sketch runs 100 times a full testing cycle, that writes 1 to 132 bytes to the CDC in 2 modes.
Using the master branch version of CDC, it fails with a message from time to time, when running the sketch 20 times.
But with the CDC Write() version you provided, it also fails, but just hangs -- no error message from the Sketch.
I think that it needs more investigation. Anyway  your idea for the changes seems to make sense. Thanks!

For the records, this is the latest testing sketch version:
``` cpp
// ESP32-C3 using HW Serial CDC on Boot enabled

bool testCase1(String data) {
  size_t receivedLen = data.length();
  size_t sendLen = 0;

  while (sendLen < receivedLen) {
    if (!Serial.write(data[sendLen++])) break;
  }
  Serial.println("\r\n=============");
  if (sendLen != receivedLen) {
    Serial0.printf("\r\n ===> T1::Error -- didn't send all data [%d of %d]\r\n", sendLen, receivedLen);
    return false;
  } else {
    return true;
  }
}

bool testCase2(String data) {
  size_t receivedLen = data.length();

  size_t sent = Serial.write(data.c_str(), receivedLen);
  Serial.println("\r\n=============");
  if (sent != receivedLen) {
    Serial0.printf("\r\n ===> T2::Error -- didn't send all data [%d of %d]\r\n\r\n", sent, receivedLen);
    return false;
  }  else {
    return true;
  }
}

void setup() {
  Serial0.begin(115200);

  Serial.begin();
  Serial0.println("\r\nStarting... open Serial Monitor with CDC port.");
  //delay(5000);
  while(!Serial) delay(100);
  Serial.println("\r\nCDC Started.");

  bool result = true;
  String testString = "0";
  for (int x = 0; x < 100; x++) {
    String testString = "0";
    for (int i = 0; i < 132; i++) {
      result &= testCase1(testString);
      result &= testCase2(testString);
      if (!result) {
        Serial0.printf("Failed with %d bytes\r\n", i);
        break;
      }
      char c = testString.charAt(testString.length() - 1);
      c = c == '9' ? '0' : c + 1;
      testString += c;
    }
    if (!result) break;
    Serial0.print(x+1);
    Serial0.print(" ");
    if (((x+1) % 10) == 0) Serial0.println();
  }
  if (result) Serial0.println("\r\nSetup is done : SUCCESS!\r\n");
  else Serial0.println("\r\n\t\t====> Setup is done : FAILED!\r\n");
}

void loop() {}

```

TD-er, 2024-03-21:
I looked a bit more at your test and I don't think it is the right test for the issue.
`Serial.write()` is allowed to return 0 if the buffer is full.
You really need to have some way of capturing the data on the HWCDC side to see if you're missing bytes.

N.B. you can 'optimize' your test a bit more by not copying the Strings when calling the testcases: `bool testCase2(const String& data)`
Maybe this will result in full buffers even more often, but as I mentioned a full buffer isn't a bug.

One other way can be to check if `Serial.write()` does claim to have written more than could have fit in the buffer.
Though I am not 100% sure this can be a conclusive test as bytes could have been fetched from the buffer in the mean time.

SuGlider, 2024-03-21:
> Have you also tried it on an ESP32-S3? That one did show these issues way more often than the C3 did. Maybe because the S3 is faster? (or more locks between cores, or any of the many many other differences between them :) )

I'm using the S3 for testing. Based on you patch I have been changing here and there. I think I have made it work always.
I'll post a PR and you can also test it with ESPEasy.

SuGlider, 2024-03-21:
> I looked a bit more at your test and I don't think it is the right test for the issue. `Serial.write()` is allowed to return 0 if the buffer is full. You really need to have some way of capturing the data on the HWCDC side to see if you're missing bytes.

I use a Python script to check the CDC ouput.

> N.B. you can 'optimize' your test a bit more by not copying the Strings when calling the testcases: `bool testCase2(const String& data)` Maybe this will result in full buffers even more often, but as I mentioned a full buffer isn't a bug.

Thanks. It has been incorporated.

> One other way can be to check if `Serial.write()` does claim to have written more than could have fit in the buffer. Though I am not 100% sure this can be a conclusive test as bytes could have been fetched from the buffer in the mean time.

Yes, but `Serial.write()` doesn't return until it has sent all bytes.

This is the new testing sketch:

``` cpp
// ESP32-C3 using HW Serial CDC on Boot enabled

bool testCase1(const String &data) {
  size_t receivedLen = data.length();
  size_t sendLen = 0;

  while (sendLen < receivedLen) {
    if (!Serial.write(data[sendLen++])) break;   // <<< if it fails sending a single byte, it will fail the whole test case
  }
  if (sendLen != receivedLen) {
    Serial0.printf("\r\n ===> T1::Error -- didn't send all data [%d of %d]\r\n", sendLen, receivedLen);
    return false;
  } else {
    return true;
  }
}

bool testCase2(const String &data) {
  size_t receivedLen = data.length();

  size_t sent = Serial.write(data.c_str(), receivedLen);
  if (sent != receivedLen) {
    Serial0.printf("\r\n ===> T2::Error -- didn't send all data [%d of %d]\r\n\r\n", sent, receivedLen);
    return false;
  }  else {
    return true;
  }
}

void setup() {
  Serial0.begin(115200);

  Serial.begin();
  Serial0.println("\r\nStarting... open Serial Monitor with CDC port.");
  while(!Serial) delay(100);
  Serial.println("\r\nCDC Started.");

  bool result = true;
  String testString = "0";
  for (int x = 0; x < 100; x++) {  // runs the test cases 100 times
    String testString = "0";
    for (int i = 0; i < 302; i++) {  // sends from 1 to 301 bytes per test
      result &= testCase1(testString);
      result &= testCase2(testString);
      if (!result) {
        Serial0.printf("Failed with %d bytes\r\n", i);
        break;
      }
      char c = testString.charAt(testString.length() - 1);
      c = c == '9' ? '0' : c + 1;
      testString += c;
    }
    if (!result) break;
    Serial0.print(x+1);
    Serial0.print(" ");
    if (((x+1) % 10) == 0) Serial0.println();
  }
  if (result) Serial0.println("\r\nSetup is done : SUCCESS!\r\n");
  else Serial0.println("\r\n\t\t====> Setup is done : FAILED!\r\n");
}

void loop() {}

```

TD-er, 2024-03-21:
> I use a Python script to check the CDC ouput.
> 

Can you also introduce some delays in this Python script to fetch the bytes?
In the discussions I linked in this issue, there have been some specific mentions about this mainly happening when using the debugger console.

SuGlider, 2024-03-21:
> > I use a Python script to check the CDC ouput.
> 
> Can you also introduce some delays in this Python script to fetch the bytes? In the discussions I linked in this issue, there have been some specific mentions about this mainly happening when using the debugger console.

I'm using Windows for testing the Python Script.

Ok, I added a random delay in the python script between reading CDC lines (up to 5ms / new line).
I also limited the python CDC RX/TX buffer to  128 bytes.

But note that the OS CDC buffer and driver will take care of the S3 TX... 
Just the Python script will suffer a big delay.
Python will consume Windows CDC driver data from the Windows internal buffer.

Everything works fine up to the end of the Arduino Testing Sketch, but the Python script takes quite very long now.
Anyway, this is expected and it runs correctly up to the end. No data loss - I've run it for at least 20~40 times.

SuGlider, 2024-03-21:
For the records, this is the sketch and python script:

### Arduino Testing Sketch (USB Device side - HW CDC S3)

``` cpp
// ESP32-C3 using HW Serial CDC on Boot enabled

bool testCase1(const String &data) {
  size_t receivedLen = data.length();
  size_t sendLen = 0;

  while (sendLen < receivedLen) {
    if (!Serial.write(data[sendLen++])) break;   // <<< if it fails sending a single byte, it will fail the whole test case
  }
  Serial.println("===");
  if (sendLen != receivedLen) {
    Serial0.printf("\r\n ===> T1::Error -- didn't send all data [%d of %d]\r\n", sendLen, receivedLen);
    return false;
  } else {
    return true;
  }
}

bool testCase2(const String &data) {
  size_t receivedLen = data.length();

  size_t sent = Serial.write(data.c_str(), receivedLen);
  Serial.println("===");
  if (sent != receivedLen) {
    Serial0.printf("\r\n ===> T2::Error -- didn't send all data [%d of %d]\r\n\r\n", sent, receivedLen);
    return false;
  }  else {
    return true;
  }
}

void setup() {
  Serial0.begin(115200);

  Serial.begin();
  Serial0.println("\r\nStarting... open Serial Monitor with CDC port.");
  while(!Serial) delay(100);
  Serial.println("\r\nCDC Started.");

  bool result = true;
  String testString = "0";
  for (int x = 0; x < 100; x++) {  // runs the test cases 100 times
    String testString = "0";
    for (int i = 0; i < 302; i++) {  // sends from 1 to 301 bytes per test
      result &= testCase1(testString);
      result &= testCase2(testString);
      if (!result) {
        Serial0.printf("Failed with %d bytes\r\n", i);
        break;
      }
      char c = testString.charAt(testString.length() - 1);
      c = c == '9' ? '0' : c + 1;
//      if(i > 22) c = '7'; inserts an error to test the python script
      testString += c;
    }
    if (!result) break;
    Serial0.print(x+1);
    Serial0.print(" ");
    if (((x+1) % 10) == 0) Serial0.println();
  }
  if (result) {
    Serial.println("OK");
    Serial0.println("\r\nSetup is done : SUCCESS!\r\n");
  }  else {
    Serial.println("ERROR!!!");
    Serial0.println("\r\n\t\t====> Setup is done : FAILED!\r\n");
  }
}

void loop() {}
```

### Python Testing Script (USB Host side - Windows 10)
``` python
#! /c/Users/rocor/AppData/Local/Microsoft/windowsApps/python.exe

"""
	Python Script for testing ESP32 S2 Arduino issue #9378 
	ISSUE - HW CDC Write() with fast stream of data
"""

import serial
import re
import sys
import random
import time

print ("CDC App test for issue 9378")

def is_valid_sequence(s):
    # Initialize variables
    prev_digit = None
    valid = True

    # Iterate through each character in the string
    for char in s:
        if char.isdigit():
            digit = int(char)
            if prev_digit is None:
                # First digit
                if digit != 0:
                    valid = False
                    break
            else:
                # Check if the next digit is valid
                if digit != (prev_digit + 1) % 10:
                    valid = False
                    break
            prev_digit = digit
        elif char == "=":
            # Check if the sequence ends with "==="
            if s.endswith("==="):
                break
            else:
                valid = False
                break
        else:
            # Invalid character
            valid = False
            break

    return valid

try:
	# CDC same as SERIAL_8N1 - Arduino equivalent
	# Change 'com1' to what ever is your USB CDC serial device name (win/linux)
	# timeout=None means that it will work as a blocking read()
	# write() is blocking by default

	CDC = serial.Serial(port='com22', baudrate=115200)   # timeout=0.01 means non blocking! 
	CDC.set_buffer_size(rx_size = 128, tx_size = 128)  ## comment it out to speed up processing
except:
	print("COM22: port is busy or unavailable")
	exit()

print(f"Serial port com22 opened successfully.")
#read first 2 lines
line = CDC.readline().decode("utf-8").strip()
print(line)   ### for debugging
line = CDC.readline().decode("utf-8").strip()
print(line)  ### for debugging
pattern = re.compile(r"(\d+)===")

Error = False
counter = 0

while True:
	# Adds some random delay... for testing
	randomTime_ms = random.randint(0, 5)   ## comment it out to speed up processing
	time.sleep(randomTime_ms / 1000)
	
	# Read a line from the serial port
	line = CDC.readline().decode("utf-8").strip()
#	print(line)  ### for debugging

	# Check if the line ends with "==="
	if line.endswith("==="):
		# Extract the counter and sequence
		match = pattern.match(line)
		if match:
			sequence = match.groups()
			# Count the number of digits (0 to 9) in the sequence
			num_digits = len(sequence[0])
			if is_valid_sequence(sequence[0]):
				if num_digits >  301: ### Arduino sketch sends up to 302 digits
					if counter % 2 == 1: ### Arduino sketch send the line twice (TestCase 1 and 2)
						### CDC output will match ESP32 UART0 output for user control
						### Python will process the OS buffer and stay behind the ESP32 counting
						print(f"{int(counter / 2) + 1} ", end='',sep='',flush=True)				
					counter += 1
					if counter % 20 == 0:
						print("")
#				print(f"Sequence: '{sequence[0]}', Number of Digits: {num_digits}")  ### for debugging
			else:
				print(f"==> BAD sequence '{sequence[0]}', length: {num_digits}")
				Error = True
				break
		else:
			print(f"Invalid format: '{line}'")
			Error = true
			break
	else:
		print(line)  ### for debugging
		if line != "OK":
			print(f"Incomplete line: '{line}'")
			Error = true
		break

if Error:
	print("    ERROR ....")
else:
	print("\n All Fine!")
sys.exit()

```

SuGlider, 2024-03-21:
@TD-er - please check the PR #9401

SuGlider, 2024-03-21:
Let me know if this works for your project. At least it works fine with the whole testing scripts.

SuGlider, 2024-03-21:
@TD-er - Just to check the testing scripts, I confirm that using the proposed Sketch + Python, it always fails at some random time.
After the FIX of the PR, it works always. Therefore, it shall be fixed for good.

TD-er, 2024-03-21:
Thanks, I will have a look at the proposed fix and will let you know asap.

TD-er, 2024-03-21:
Nope, not working....

```

INIT : Booting version: ESP_Easy_mega_20240321_max__16M8M_LittleFOPI_PSRAM_CDC_ETbugfix/ESiFi_AP_capies_5432 SDK 5.1.316)
00.369 : (328276) Info   : INIT : Free :328276
  : I Action before d Task Last sys- Restart Reason: CPU0: u reore art reset digital   : FS   : Moing...
 (325656) : FS   : FS mount suful, used 15 bytes of 8318976
d!
00.435 : (5192) InfoSPEasy consing ESPEasySerial
00.455 : (325128allback enabled
00.464 : (3 : SecuritySettings CRC...OK
00.478 : (324544) Info   : ProvisioningSettingK
00.480 : (3200.41 : (323824) Info   : INIT : Chetasks
 enabled
   : Timeme: 1711054879
)9.000
(323024) Info   : Current Time one: STD time sta03:00 0  (322928) : Loca024-03-2119
00.617 : (322816) Info   IFI : Set Wo STA
00.780 : (292640) Info   e using erial
00.801 : (292576) In: ESPEe fed
00NIT92048
00.811 : (287188) Ierial
00(287124) Info   : ESPEasy console ed
00.832 :7012) INFO : Plugins: 147 ["Normal","MAX ESP32"] (ESP326)
007292) Info   : WIFI : Connect Lurch0dBm)open attemt #0
```

And when I comment out both lines with `usb_serial_jtag_ll_txfifo_flush();`, I get this:

```
00.368 : (328628) Info   : Using PSRAM
00.369 : (328420) Info   :

INIT : Booting version: ESP_Easy_mega_20240321_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH, (Self built) bugfix/ESP8266_WiFi_AP_capabilities_548dd59 (ESP32 SDK 5.1.3.240316)
00.370 : (328276) Info   : INIT : Free RAM:328276
00.391 : (328068) Info   : INIT : Manual Reboot #3 Last Action before Reboot: Background Task Last systime: 1711055189 - Restart Reason: CPU0: usb uart reset digital core  CPU1: usb uart reset digital core
00.402 : (328020) Info   : FS   : Mounting...
00.421 : (325656) Info   : FS   : LittleFS mount successful, used 151552 bytes of 8318976

Build changed!
00.435 : (325192) Info   : ESPEasy console using ESPEasySerial
00.456 : (325128) Info   : ESPEasy console fallback enabled
00.464 : (325012) Info   : CRC  : SecuritySettings CRC...OK
00.478 : (324544) Info   : CRC  : ProvisioningSettings CRC   ...OK
00.481 : (324848) Info   : INIT : I2C
00.482 : (323824) Info   : INIT : Check for Priority tasks
00.482 : (323776) Info   : INIT : SPI not enabled
00.493 : (323168) Info   : Time : Set Ext. Time Source: RTC at boot time: 1711055189
00.504 : (323120) Info   : Time set to 1711055189.000
00.504 : (323024) Info   : Current Time Zone: STD time start: 2024-10-27 03:00:00 offset: 0 min
00.515 : (322928) Info   : Local time: 2024-03-21 21:06:29
00.617 : (322816) Info   : WIFI : Set WiFi to STA
00.807 : (292640) Info   : ESPEasy console using ESPEasySerial
00.829 : (292576) Info   : ESPEasy console fallback enabled
00.829 : (292000) Info   : INIT : Free RAM:292048
00.838 : (287188) Info   : ESPEasy console using ESPEasySerial
00.859 : (287124) Info   : ESPEasy console fallback enabled
00.859 : (287012) Info   : INFO : Plugins: 147 ["Normal","MAX ESP32"] (ESP32 SDK 5.1.3.240316)
00.871 : (287292) Info   : WIFI : Connecting Lurch4 3C:37:12:AA:D0:5C Ch:1 (0dBm)open attempt #0
00.875 : (287092) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00.877 : (282116) Info   : Webserver: start
01.490 : (281000) Info   : SYS  : 281352,6061492,40,-54
01.494 : (280904) Info   : WIFI : Connected! AP: Lurch4 (3C:37:12:AA:D0:5C) Ch: 1 Duration: 588 ms
01.944 : (281404) Info   : SYS  : 281468,6061288,40,-58
01.998 : (281524) Info   : WIFI : DHCP IP: 192.168.11.0 (MAX-ESP32-s3pro-117) GW: 192.168.10.1 SN: 255.255.254.0 DNS: 192.168.10.1 / 42.16.55.129   duration: 535 ms
02.013 : (279692) Info   : WIFI : Arduino wifi status: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP
02.015 : (279036) Info   : UDP : Start listening on port 8266
02.026 : (279516) Info   : firstLoopConnectionsEstablished
02.030 : (279452) Info   : Internal command: All checked OK
```

SuGlider, 2024-03-22:
@TD-er @Jason2866 
Based on the test sketch and python script (from https://github.com/espressif/arduino-esp32/issues/9378#issuecomment-2013216940), when the `usb_serial_jtag_ll_txfifo_flush();` is commented out, it fails 100% of the executions on the S3-Mini-1--N8 and S3-woom-2-N32R8 that I have here. I have tried 10 times. 10 fails (it hangs after about 50 transfers).

Keeping the `usb_serial_jtag_ll_txfifo_flush();`, it works 100% of the tries. I also tested it 10 times.

Please test it with the Arduino IDE using the Core 3.0.0 and the modifications to `HWCDC::write(const uint8_t *buffer, size_t size)` function. I'm not sure if the IDF version and other itens used in @TD-er building correspond to the same environment I am testing for.

Given that we get oposite results, I only can think that or the test case isn't good enough or we are running different frameworks.

I need a test case to prove that removing `usb_serial_jtag_ll_txfifo_flush();` solves any issue.
The original code from `Arduino master branch` **fails with my test case**, therefore, I found a way to reproduce the issue.
The current Arduino master branch fails 100% of the time. This indicates that the test case I have created is valid.

I can't use ESPEasy to test it, because it is a full project and the failure may happen due to other reasons.
ESPEasy may use some different set of libraries and or Arduino Core source code.

SuGlider, 2024-03-22:
The PR #9401 target the master branch, Arduino Core 3.0.0 using IDF 5.1.
Is ESPEasy also suing this same environment?

Jason2866, 2024-03-23:
@SuGlider The framework used for the test from @TD-er is based on latest IDF 5.1 branch https://github.com/espressif/esp-idf/tree/release/v5.1 and Arduino latest branch master.

SuGlider, 2024-03-23:
In that case I ask you guys to bring a Test Case, based on a minimum Arduino Sketch, that can demonstrate/reproduce the issue and be used for testing.

This is necessary for applying any patch.

TD-er, 2024-03-23:
I am now running your test code with PlatformIO and the same libraries/platform builds etc. as I was using for ESPEasy.
Your sketch does stop working on the code before your patch/PR and it does finish with your patch/PR.

So at least that sounds the same as you experienced.

I will now try to reproduce the ESPEasy behavior in a simple test program.

TD-er, 2024-03-23:
So far I have not yet been able to reproduce the odd behavior as seen in ESPEasy with a small sketch, but hopefully I can find some more time tomorrow to test.

SuGlider, 2024-03-24:
Thanks @TD-er. I hope you have success replicating the issue with a small sketch.

TD-er, 2024-03-26:
Well in the very few hours I had in the last few days to test, I was not (yet...) able to reproduce the same loss of data in a simple program.

Since it does fix the bug as shown in the test program presented by @SuGlider , I think his PR should be merged.
As soon as I can reproduce (or explain) the behavior I'm seeing with ESPEasy, I will report back.

SuGlider, 2024-03-26:
Is it possible that ESPEasy has multiple tasks writing to the CDC?

This may explain why removing CDC Flush helps. Flushing makes whatever is the CDC FIFO be sent to the host as soon as it asks for.

It could "tear" the messages.

[11 more replies]