espressif/arduino-esp32 #9383: An Example for using ESP32S3 with SD card as Mass Storage Device via SDIO

wbadry, 2024-03-16:
### Related area

USB Mass Storage Device

### Hardware specification

ESP32-S3 DevKitC-1

### Is your feature request related to a problem?

The examples are based on the SPI SD card interface. Is it possible to have an example based on the SD MMC interface via SDIO ?

### Describe the solution you'd like

A simple example to recognize the SD MMC connected to ESP32S3 to be recognized as USB mass storage device via one of the two USB ports

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-03-16:
SD_MMC does not have methods for direct diskio access (reading/writing blocks).  It looks like that would need to be added to use the USBMSC example as is.

martinberlin, 2024-06-14:
Just wanted to know is there any progress on this particular Issue?
So far I could implement  SD card as Mass Storage Device but failed trying to do it in Arduino. Existing examples do not work correctly for me so I'm curious to understand how to get this one working

lbernstone, 2024-06-14:
Yes, sorry.  I missed the issue when I added the PR.
[SD2USBMSC example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SD2USBMSC/SD2USBMSC.ino)
Fixed by #9796

closed 2025-01-15 by Parsaabasi