espressif/arduino-esp32 #9384: WiFi.Begin does not accept specified BSSID for Access Point when connecting to WPA2 Enterprise Network

MrSwirlyPants, 2024-03-16:
### Board

esp32 devkit 1

### Device Description

esp32 devkit 1

### Hardware Configuration

nothing

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Dear all,

My intention is to connect my ESP32 to the strongest AP available in a WPA2 Enterprise network. I have the luxury to test my code as well in a WPA2-PSK version of the same network, however this is only possible for test purposes. A few observations:

Examples | WiFi | WiFiScan  works and shows all available AP's, with RSSI
Examples | WiFi | WiFiClient  works with WPA2-PSK and connects
Examples | WiFi | WiFiClientEnterprise  works with WPA2-Enterprise and connects

Thanks to several forumposts around the internet I have the code as shown below. It searches for the strongest AP and tells WiFi.begin to connect to the AP with the BSSID belonging to the strongest AP.

If I run the code for the WPA2-PSK network, it does exactly as specified: it shows the BSSID of the AP I'm connected to, and it corresponds to the strongest from the list shown by the scan of the AP's.

If I run the code for the WPA2-Enterprise network, it takes a bit longer to connect, but IT NEVER CONNECTS TO THE SPECIFIED BSSID. I almost always get a connection to an AP with lower RSSI.

In the code I have specified where a line should be commented / uncommented to connect to either a regular network or an enterprise network.

I have searched several forums but I can't seem to find anyone mentioning this exact problem. Could this be something in the code supplied by "esp_wpa2.h", or could this be something in the network I'm trying to connect to - though I can't fathom what.

Thank you for reading and much obliged for any help anyone can offer, even if it's just reproducing the problem.

With kind regards,

Swirly

### Sketch

```cpp
#include "WiFi.h"
#include "esp_wpa2.h"

const char* ssid     = "********"; // Change this to your WiFi SSID
const char* username = "********"; // Change this to your WiFi username, if needed for WPA2 enterprise
const char* password = "********"; // Change this to your WiFi password

void setup()
{
  Serial.begin(115200);
  while(!Serial){delay(100);}
  Serial.println("Connecting to WiFi");

  WiFi.persistent(false); //Avoid to store Wifi configuration in Flash
  WiFi.mode(WIFI_STA);
  WiFi.disconnect(); // delete old config
  delay(100);

  int i_strongest = -1;
  int32_t rssi_strongest = -100;
  Serial.printf("Start scanning for SSID %s\r\n", ssid);

  int n = WiFi.scanNetworks(); // WiFi.scanNetworks will return the number of networks found
  Serial.println("Scan done.");

  if (n == 0)
  {
    Serial.println("No networks found!");  
  }
  else
  {
    Serial.printf("%d networks found:\n", n);
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.printf("%d: BSSID: %s  %2ddBm, %3d%%  %9s  %s\r\n", i, WiFi.BSSIDstr(i).c_str(), WiFi.RSSI(i), constrain(2 * (WiFi.RSSI(i) + 100), 0, 100), (WiFi.encryptionType(i) == WIFI_AUTH_OPEN) ? "open" : "encrypted", WiFi.SSID(i).c_str());
      if ((String(ssid) == String(WiFi.SSID(i)) && (WiFi.RSSI(i)) > rssi_strongest))
      {
        rssi_strongest = WiFi.RSSI(i);
        i_strongest = i;
      }
    }
  }

  if (i_strongest < 0)
  {
    Serial.printf("No network with SSID %s found!\r\n", ssid);
  }
  Serial.printf("SSID match found at %d. Connecting...\r\n", i_strongest);

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  
  //Select this line to make connection to WPA2 enterprise network
  WiFi.begin(ssid, WPA2_AUTH_PEAP, username, username, password,NULL,NULL,NULL,0,WiFi.BSSID(i_strongest),true);
  
  //Select this line to make connection to WPA2 personal network / regular wifi network
  //WiFi.begin(ssid,password,0,WiFi.BSSID(i_strongest),true);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(100);
  }

  Serial.println(" connected!");
  Serial.print("BSSID: ");
  Serial.println(WiFi.BSSIDstr());
  Serial.print("RSSI: ");
  Serial.println(WiFi.RSSI());
  Serial.println(WiFi.localIP()); //print LAN IP
}

void loop()
{
}
```

### Debug Message

```plain
I will provide this as soon as I can run the program on the enterprise network.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-03-16:
Can you try with actual development version from branch master?
It is based on IDF 5.1.x. Arduino core 2.0.14 is based on IDF 4.4.x
There are many changes and fixes from IDF 4.4 to IDF 5.1. There is a good chance that the issue you see is solved with using IDF 5.1 as base for Arduino core.

MrSwirlyPants, 2024-03-16:
Yes I can! I just found out how to install the development release, I will install it this weekend and run the test on monday when I'm near the enterprise network. I will report my findings!

TD-er, 2024-03-16:
Have you also tried using the channel when specifying the BSSID to connect to?

MrSwirlyPants, 2024-03-16:
I did, and I think to no avail, but I will retest that on monday.

MrSwirlyPants, 2024-03-18:
I installed v3.0.0 alpha and tried my program, specifying the channel as well. Unfortunately, this does not seem to work.

I loaded the example WiFiClientEnterprise, entered my credentials, and tried it. This does not seem to work either. I enabled the full debug mode, as shown below. I'm terribly sorry but I couldn't get it formatted in any nice way, so it's a dump. If I can do this better, please let me know.

I gather there is an authentication problem. I reinstalled v2.0.14 and loaded up the WiFiClientEnterprise example and copied my credentials. I get a connection fairly quickly. Please advice, I'm willing to run some tests.

A a side note: is there a way to see if i can connected with the ESP-IDF functions? As I use the Arduino environment, I have not read into the API presented by ESP, and the docs seem rather daunting.

With kind regards,

Swirly

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1256
load:0x40078000,len:13832
load:0x40080400,len:4
load:0x40080404,len:3048
entry 0x40080590
[     3][D][esp32-hal-cpu.c:259] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 1.45
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   345248 B ( 337.2 KB)
  Free Bytes        :   314684 B ( 307.3 KB)
  Allocated Bytes   :    23468 B (  22.9 KB)
  Minimum Free Bytes:   309348 B ( 302.1 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : DIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Mar 18 2024 07:57:33
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.2-185-g3662303f31-dirty
  Arduino Version   : 3.0.0
------------------------------------------
[... 152 lines cut ...]
               spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
             coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
  ------------------------------------------
  Software Info:
------------------------------------------
  Compile Date/Time : Mar 18 2024 07:57:33
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.2-185-g3662303f31-dirty
  Arduino Version   : 3.0.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : doitESP32devkitV1
  Arduino FQBN      : esp32:esp32:esp32doit-devkit-v1:UploadSpeed=921600,FlashFreq=80,DebugLevel=debug,EraseFlash=none
============ Before Setup End ============

Connecting to network: MYNETWORK
[   540][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   625][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 2 - STA_START
............................................................E (43619) wifi:NAN WiFi stop
[ 30644][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 30653][W][WiFiGeneric.cpp:1064] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 30660][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 3 - STA_STOP
ets Jul 29 2019 12:21:46

closed 2025-01-15 by Parsaabasi