espressif/arduino-esp32 #9385: I can't upload large program code

Zsolt56, 2024-03-16:
### Board

Arduino Nano ESP32

### Device Description

PLATFORM: Espressif 32 (6.5.0+sha.9b0c95d) > Arduino Nano ESP32

### Hardware Configuration

HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

240 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I run into an upload problem: step by step developing my program, I reached a size limit where i cant upload the full code any more.
While there are a plenty of room for code:
  RAM: [== ] 18.5% (used 60560 bytes from 327680 bytes)
  Flash: [====== ] 55.3% (used 1088161 bytes from 1966080 bytes)
the upload process stops at 95%:
  Download [======================= ] 92% 1007616 bytes
  Download [======================= ] 95% 1040384 bytes failed!
  DFU state(10) = dfuERROR, status(3) = Device is unable to write memory
  **** [upload] Error 74*

I think here should be a barrier, because if I shring the code (omitting a part of it), the upload works fine and the code runs without any problem. I tried looking for issues around the partitions.csv, but I haven't found any solution yet 

### Sketch

```cpp
not relevant
```

### Debug Message

```plain
DFU state(10) = dfuERROR, status(3) = Device is unable to write memory
  **** [upload] Error 74*
```

### Other Steps to Reproduce

I also tried to upload similar size code on Arduino IDE 2.3.2, with the similar results

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-03-16:
Could you please post the `partitions.csv` used in the project?

lbernstone, 2024-03-16:
The whole platformio.ini would be useful.

Zsolt56, 2024-03-16:
Here is the platfomrio.ini file:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:arduino_nano_esp32]
platform = espressif32
board = arduino_nano_esp32
framework = arduino
monitor_speed = 115200
board_build.mcu = esp32s3

build_flags = 
	-DELEGANTOTA_USE_ASYNC_WEBSERVER=1

lib_deps = 
	arduino-libraries/NTPClient@^3.1.0
	jchristensen/Timezone@^1.2.4
    gitlab-communication/VEGA_IRremote@^1.0.0
	https://github.com/me-no-dev/ESPAsyncWebServer.git
    ayushsharma82/ElegantOTA@^3.1.0
	arduino-libraries/Arduino_JSON @ 0.1.0

board_build.filesystem = littlefs
;board_build.partitions = app3M_spiffs9M_fact512k_16MB.csv
board_build.partitions = min_spiffs.csv

;upload_protocol = esptool		;for data uploading
upload_protocol = dfu			;for program uploading

Earlier I tried with 2 partition files, without any effect. Later however I have installed the ESP-IDF and download the hello_world project with it, as a test. It worked smootly. Then I returned to Platformio, built a filesystem image, uploaded it, the recomplied the program and I could upload it without any size problem (it was strange, that the program uploading was working only with esptool, not with dfu, as it was eralier). So, the ESP-IDF reapired something and it I can upload code without size problems. I have no idea, what ESP-IDF could do and the PlatformIO could not. As the Arduino Nano ESP32 is really a problematic board, it is possible, that somewhen and somehow I made a wrong step during the program development...

Jason2866, 2024-03-16:
Do not define any partition scheme yourself for this board. You will encounter issues.
The partition scheme is defined in the boards manifest https://github.com/platformio/platform-espressif32/blob/develop/boards/arduino_nano_esp32.json

Jason2866, 2024-03-16:
The boards design concept is very different compared to other boards. The upload of firmware is ONLY done with the `dfuutil` which is installed automatically from Platformio (when this board is used). So problems with uploads have nothing to do with Platformio or ArduinoIDE. Issues with the uploader `dfuutil` and the board should be directly addressed to the Arduino team

SuGlider, 2024-03-17:
By the way, Arduino.CC (board maker) has its own board definition for the Arduino Nano ESP32, when using the Arduino IDE (I know it's about PlatformIO, but just in case).

[image]

SuGlider, 2024-03-24:
@Zsolt56 - any further question? Otherwise we can close the issue.

Zsolt56, 2024-03-26:
Hi Rodrigo,

Yes, you can close the issue.

Kind Regards,
Zsolt

closed 2024-03-26 by SuGlider