espressif/arduino-esp32 #9388: tcpip_thread crash!

Speedy2k, 2024-03-18:
### Board

Adafruit ESP32 Feather

### Device Description

It is connected tp a custom device.
[image]
[image]

### Hardware Configuration

There is a lot! But nothing that should create a tcp_ip thread crash!

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I got random crash on tcp_ip on the udp_sendTo function after a dns_timeout_cb ?? I Don,t understand at all why this is making my sketch crash.

### Sketch

```cpp
I don't have any specific code to make it happen since it happen not very frequently and i can't see in the debug what is making it crash!
```

### Debug Message

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4012909f  PS      : 0x00060630  A0      : 0x801210e4  A1      : 0x3ffb3880  
A2      : 0x00000001  A3      : 0x3ffccb0c  A4      : 0x3ffc7124  A5      : 0x00000035  
A6      : 0x00000004  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffb38d4  A12     : 0x00000000  A13     : 0x3ffccb90  
A14     : 0x01000100  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000015  LBEG    : 0x4008a928  LEND    : 0x4008a93e  LCOUNT  : 0xffffffff  

Backtrace: 0x4012909c:0x3ffb3880 0x401210e1:0x3ffb38c0 0x401211ce:0x3ffb3900 0x401218b0:0x3ffb3920 0x401218fb:0x3ffb3940 0x401287c1:0x3ffb3960 0x401205f9:0x3ffb3980

Decoding stack results
0x4012909c: udp_sendto at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/udp.c line 538
0x401210e1: dns_send at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/dns.c line 919
0x401211ce: dns_check_entry at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/dns.c line 1175
0x401218b0: dns_tmr at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/dns.c line 1208
0x401218fb: dns_timeout_cb at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/dns.c line 438
0x401287c1: sys_check_timeouts at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/timeouts.c line 411
0x401205f9: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 104
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-03-20:
>Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Usually it is related to a bad pointer with `EXCVADDR: 0x00000015`
More information:

### [LoadProhibited, StoreProhibited](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited)
These CPU exceptions happen when an application attempts to read from or write to an invalid memory location. The address which has been written/read is found in the EXCVADDR register in the register dump. If this address is zero, it usually means that the application has attempted to dereference a NULL pointer. **If this address is close to zero, it usually means that the application has attempted to access a member of a structure, but the pointer to the structure is NULL.** If this address is something else (garbage value, not in `0x3fxxxxxx` - `0x6xxxxxxx` range), it likely means that the pointer used to access the data is either not initialized or has been corrupted.

>I don't have any specific code to make it happen since it happen not very frequently and i can't see in the debug what is making it crash!

It is hard to help any further. One possible reason could be that the application runs out of HEAP memory and some `malloc()` returns NULL, but the application doesn't check it and end up using this NULL pointer, crashing the ESP32.

Speedy2k, 2024-03-21:
Ok i understand that, but why is it happening on the tcp_ip thread ? I don't personnaly have access to anything in this code!?

lbernstone, 2024-03-22:
It is servicing a UDP send.

Jason2866, 2024-04-05:
Github issue is not a forum for general programming problems.

closed 2024-04-05 by Jason2866