espressif/arduino-esp32 #9391: Does this library work with EXT 4 Filesystem?

basementmedia2, 2024-03-19:
### Related area

Which filesystems are supported generally?

### Hardware specification

ESP32 (Arduino IDE)

### Is your feature request related to a problem?

No

### Describe the solution you'd like

I wonder if the SD Card msut be formated with FAT32 or also other (better) filesystems like EXT4 are supported. 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

atanisoft, 2024-03-19:
SD cards are only supported via FAT filesystems. I'm not aware of any Arduino library that supports ext4 (or other Linux compatible file systems).

basementmedia2, 2024-03-19:
html,body{background-color:#fff;color:#333;line-height:1.4;font-family:sans-serif,Arial,Verdana,Trebuchet MS;}Hi Mike,
i found this library
https://github.com/gkostka/lwext4
Developer says:

But I am a beginner so i don't really understand, if this lib can be used (in a good performance) with a ESP32 microcontroller.
So if you say it is not possible, i believe you  :-)
Best regards
Daniel
 
Mike Dunston schrieb am 19.03.2024 13:32 (GMT +01:00):> SD cards are only supported via FAT filesystems. I'm not aware of any Arduino library that supports ext4 (or other Linux compatible file systems).> > —>

atanisoft, 2024-03-19:
That looks like an interesting library that might work, but it is likely not going to be simple or straightforward to integrate.

basementmedia2, 2024-03-19:
My current problem is that sometimes (after logging data on SD card for weeks or month) suddenly the SD card can not be read anymore (single files are "corrupt" i think. After formatting it with FAT32 they work again.

I always close files after writing / appending data with SD.close().
Before restarting the microcontroller i use SD.end() to close the filesystem.

Is there more that i can do to prevent errors?
Do you have any tipps?

Somebody gave me the following tips, but i don't know weather they can be used for my use case (ESP32, Arduino IDE, SD Card):

"Years ago, I spent several months optimizing with a customer who could not be dissuaded from using FAT (vxWorks operating system) in his system. In the end, the solution was

o Use partitioning/formatting adapted to the geometry of the FLASH medium
o (Write) cache off - in the OS and in the FLASH medium
o Switch off unnecessary timestamps ("Access Time")
o 2 large files of unchangeable length in different directories, to which the dynamic data is written alternately and redundantly, always calling fflush() in between
o Always call "dcheck -ry" during the boot process"

Do you understand what he means and can some of those tips be solved with your SD-library?

atanisoft, 2024-03-19:
I'm unsure what might be causing your issues with the SD code. We would need an example that reliably reproduces the issue to be able to assist in debugging and advising how to possibly fix it.

lbernstone, 2024-03-19:
If you are doing a lot of writes to an SD card, you want something with active wear levelling, eg WD purple cards.  This has a whole lot of advantages over trying to manage WL through a file system.  If you really want to go down a rabbit hole instead of buying quality hardware, you could probably adapt the IDF wear levelling library for use on SD.

lbernstone, 2024-03-19:
arduino-esp32 is built on esp-idf, so idf code works in arduino.  It will not be a drop-in replacement for the SD library.
This is not something that will come out-of-the-box ready for use.  You would need to write some code to get the SD to behave as the ffat does in https://github.com/espressif/esp-idf/blob/master/components/fatfs/vfs/vfs_fat_spiflash.c#L117

SuGlider, 2024-03-24:
Closing this issue. Feel free to open a new question in the Discussions session of this GH.

closed 2024-03-24 by SuGlider