espressif/arduino-esp32 #9392: Hi, I'm trying to implement a MQTT broker connection for my ESP32

lekshith, 2024-03-19:
### Board

ESP32-S3 DevkitC-1

### Device Description

DevkitC-1

### Hardware Configuration

it has MODBUS connection to fans and some LED connections.

### Version

other

### IDE Name

Visual Studio

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm trying to make a simple MQTT broker connection to my project. Initially, I was getting the error as:
"esp-tls-mbedtls: No server verification option set in esp_tls_cfg_t structure. Check esp_tls API reference", for which I used CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y, in the menuconfig. Now I'm faced with the following error to establish a network connection.
E (363948) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (363948) esp-tls: create_ssl_handle failed
E (363948) esp-tls: Failed to open new connection
E (363948) transport_base: Failed to open a new connection

### Sketch

```cpp
Below is the code bit for executing MQTT protocol:
void mqtt_api_init(void)
{
    /* Create message queue */
    mqtt_message_queue = xQueueCreate(MQTT_MESSAGE_QUEUE_SIZE, sizeof(mqtt_message_t));
    if (mqtt_message_queue == NULL)
    {
        ESP_LOGE(TAG, "Create message queue fail");
        return;
    }

    uint8_t mac_addr[6];
    ESP_ERROR_CHECK(esp_read_mac(mac_addr, ESP_MAC_WIFI_STA));
    sprintf(gateway_id, "%02X%02X%02X%02X%02X%02X", mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);

    const esp_mqtt_client_config_t mqtt_cfg = {
        .broker = {
            .address.uri = MQTT_BROKER_URI,
            .address.port = 1883,
        },
        // .credentials = {
        //     .username = MQTT_USERNAME,
        //     .client_id = gateway_id,
        //     .authentication = {
        //         .certificate = (const char *)client_cert_pem_start,
        //         .key = (const char *)client_key_pem_start,
        //         .password = MQTT_PASSWORD,
        //     },
        // },
        .session.disable_clean_session = false,
    };

    /* Start mqtt client */
    mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
    ESP_LOGI(TAG, "MQTT client initialized");

    esp_mqtt_client_register_event(mqtt_client, ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
    esp_mqtt_client_start(mqtt_client);

    /* Creat mqtt handle message task */
    BaseType_t result = xTaskCreate(mqtt_handle_message_task, MQTT_TASK_NAME, MQTT_TASK_SIZE,
                                    NULL, MQTT_TASK_PRIORITY, &mqttTaskHandle);
    if (result != pdPASS)
    {
        ESP_LOGE(TAG, "Create mqtt task fail %d", result);
    }
    else
    {
        vTaskSuspend(mqttTaskHandle);
    }

    char topic_cmd[MQTT_TOPIC_MAX_LENGTH];
    sprintf(topic_cmd, MQTT_TOPIC_CMD, gateway_id);
    mqtt_register_callback(topic_cmd, mqtt_handle_message_default);
}
```

### Debug Message

```plain
E (363948) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (363948) esp-tls: create_ssl_handle failed
E (363948) esp-tls: Failed to open new connection
E (363948) transport_base: Failed to open a new connection
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lekshith, 2024-03-19:
[image]
Here is my free heap size for reference.

lbernstone, 2024-03-19:
The libraries included in arduino-esp32 are pre-compiled versions of the ESP-IDF sdk.  You cannot change CONFIG_XX options. You cannot make mqtt insecure this way.  There are 3rd party arduino libraries which will be able to make insecure mqtt connections.

lekshith, 2024-03-19:
I have removed those changed CONFIG options since, potentially, the working broker needs to be a secure mqtt. I tried that cause it was the only related fix I could find online. I'm very new to handling ESP32 projects, so any suggestions are welcome. Also, this is  ESP-IDF hosted on Visual studio code (not Arduino), which contains other functionalities. So, this mqtt connection must publish some of these functionalities to the broker.

SuGlider, 2024-03-20:
It doesn't sound like an Arduino issue/question, but actually an IDF question.
Try using this IDF Component: https://github.com/espressif/esp-mqtt

## Features

- Based on: https://github.com/tuanpmt/esp_mqtt 
- Support MQTT over TCP, SSL with mbedtls, MQTT over Websocket, MQTT over Websocket Secure
- Easy to setup with URI 
- Multiple instances (Multiple clients in one application)
- Support subscribing, publishing, authentication, will messages, keep alive pings and all 3 QoS levels (it should be a fully functional client).

## How to use

[ESP-MQTT](https://github.com/espressif/esp-mqtt) is a standard [ESP-IDF](https://github.com/espressif/esp-idf) component.
Please refer to instructions in [ESP-IDF](https://github.com/espressif/esp-idf)

## Documentation

* Please refer to the standard [ESP-IDF](https://github.com/espressif/esp-idf), documentation for the latest version: https://docs.espressif.com/projects/esp-idf/

* Documentation of ESP-MQTT API: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/mqtt.html

SuGlider, 2024-03-20:
Examples at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#application-examples

lekshith, 2024-03-20:
Thanks for the documentations. I've gone through those and implemented my code based on that. From what I understand,
esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00 is memory allocation failure code. Please suggest some ways to overcome this issue. I've tried freeing up so memory from idf.py size-components, but didn't help with this issue.

SuGlider, 2024-03-20:
> ![image](https://private-user-images.githubusercontent.com/162119147/314106524-66864364-b388-4441-bfb1-355c5c68cad9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

From that report, it seems that there is PSRAM in use. 
But **mbedtls** may request a `malloc()` exclusive within SRAM, threfore it may fail when it is low.
It is hard to tell how to solve it. It depends on the application.

It is necessary to do some memory management and diagnostic.
Try to monitor the HEAP by applying [Heap Debug](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html)

MBEDTLS needs about 40KB free HEAP. An alternative is using WOLFSSL.
https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/protocols/esp_tls.html#comparison-between-mbedtls-and-wolfssl

dzungpv, 2024-03-21:
S3 should run esp-mqtt fine, with Rainmaker use MQTT secure connection and HTTPS to the server fine, you can test Rainmaker first. For Arduino you can try this lib, it is wrapper of ESP-IDF: https://github.com/cyijun/ESP32MQTTClient. I am using it for project running on S1 S2 and S3.

lekshith, 2024-03-21:
Thanks for the suggestion of Rainmaker tool, I'll give that a test. Just to clarify, I'm not using Arduino development. I'm making use espressif - IDF extension in VS code. Rest of the MODBUS functions codes are all written in C.

SuGlider, 2024-03-24:
Closing this issue-question. New questions shall be open in the Discussions session.

closed 2024-03-24 by SuGlider