espressif/arduino-esp32 #9393: ESP32S3-esp_ota_set_boot_partition(_partition) in Update.end() will cause Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

guess-name, 2024-03-19:
### Board

lilygo-t-display-s3

### Device Description

lilygo-t-display-s3
Original sample

### Hardware Configuration

NULL

### Version

latest master (checkout manually)

### IDE Name

arduino

### Operating System

windows10

### Flash frequency

240MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

我想根据LittleFS中的BIN文件进行更新，但是我一直报错

我已经排查了下列信息
1. FS文件和编译文件对比，无差异
2. Being及Wirte无异常
3. 仅仅在esp_ota_set_boot_partition(_partition)报错

I tried to update from the BIN file in LittleFS, but I kept getting errors

I've checked the following information
1. There is no difference between the FS file and the compiled file
2. Being and Wirte are normal
3. Only esp_ota_set_boot_partition(_partition) reports an error

### Sketch

```cpp
#include <Arduino.h>
#include "FS.h"
#include <LittleFS.h>
#include "Update.h"
#include <functional>
#include "esp_ota_ops.h"

// put function declarations here:
uint32_t timesdfsf = 0;
extern uint32_t lenccsa;
bool sfsfdsdfdsf = false;
byte buffs[1500];
int readwwflen;

File filess;

void setup() {
  delay(2000);
  Serial0.begin(115200);
  LittleFS.begin();
  // LittleFS.format();
  Serial0.println("littlefs is init");
  // LittleFS.begin();
  // led.LedOpen();
  // uart485.UartOpen();
  // mcuConfig.McuConfigInit();

  auto appVal = esp_ota_get_running_partition();
  // mcuConfig.RtuConfigReadFs();
  // mcuConfig.RtuCtpReadFs();
  // mcuConfig.RtuCtpAllReadFs();
  // ht16k33.Ht16k33Open(&Wire, 0x70);
  // adc.AdcOpen();
  // keyObject.KeyInit();
  // moto.MotoOpen(100);
  // rtuModeVal.ModeInit();
  // Serial1.begin(115200, SERIAL_8N1, USART_RX, USART_TX);
  Serial0.println("app:" + String(appVal->label));
  delay(10000);
}
esp_ota_handle_t update_handle = 0;              // update的hdl，用于识别当前是哪个OTA，这里应该允许多个OTA同时进行？
const esp_partition_t *update_partition = NULL;  // 拿分区的指针
int binary_file_length = 0;                      // 文件长度
bool image_header_was_checked = false;           // 是否检测到头了

void loop() {
  byte sdfsd[4096];
  byte sdfdfsdfwwww;
  int sdfsdfsdfsdf;
  int filesfs = 0;
  sfsfdsdfdsf = true;
  // ymodem_receive();
  sfsfdsdfdsf = false;
  filess = LittleFS.open("/ssx.bin", FILE_READ);

  update_partition = esp_ota_get_next_update_partition(NULL);

  Serial0.println("appnext:" + String(update_partition->label));
  Update.begin(filess.size());
  Update.printError(Serial0);
  Serial0.println("begin");
  while (1) {
    sdfsdfsdfsdf = filess.read(sdfsd, 4096);
    Update.write(sdfsd, sdfsdfsdfsdf);
    filesfs += sdfsdfsdfsdf;
    if (sdfsdfsdfsdf < 4096) {
      break;
    }
  }
  Update.printError(Serial0);
  Serial0.println("writeStream:" + String(filesfs));
  filess.close();
  Update.printError(Serial0);
  Serial0.println("close");
  Update.end();
  Update.printError(Serial0);
  Serial0.println("end");
  ESP.restart();
}
```

### Debug Message

```plain
1⸮app:app1
No Error
begin
No Error
writeStream:337648
No Error
close
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4037daa9  PS      : 0x00060c33  A0      : 0x8037fab4  A1      : 0x3fcea940  
A2      : 0xffcf1314  A3      : 0xffffffff  A4      : 0x0000abab  A5      : 0x00060c23  
A6      : 0x000001ce  A7      : 0x00000000  A8      : 0x82019fa8  A9      : 0x3fceaa70  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000005  A13     : 0x3fc96190  
A14     : 0xb33fffff  A15     : 0x00000001  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffcf1314  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x4037daa6:0x3fcea940 0x4037fab1:0x3fcea970 0x4037fc0d:0x3fcea990 0x4037737e:0x3fcea9b0 0x403775af:0x3fcea9e0 0x42019159:0x3fceaa00 0x42019fbb:0x3fceaa50 0x4201a011:0x3fceaa70 0x42019fa5:0x3fceaae0 0x42019eff:0x3fceab00 0x42019b5c:0x3fceab40 0x42009748:0x3fceab70 0x4200954a:0x3fceab90 0x42009613:0x3fceabf0 0x420096fd:0x3fceac50 0x42009f96:0x3fceac70 0x4200a12c:0x3fceada0 0x420032a8:0x3fceae
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-19:
Please [decode the backtrace](https://github.com/me-no-dev/EspExceptionDecoder)

guess-name, 2024-03-19:
[image]

guess-name, 2024-03-19:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
app:app0
No Error
No Error
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4037d680  PS      : 0x00060d33  A0      : 0x8037fde0  A1      : 0x3fc9d090  
A2      : 0xffcf1314  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060d23  
A6      : 0x00060d20  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x00000010  A11     : 0x00000000  A12     : 0x00000005  A13     : 0x3fc97f64  
A14     : 0xc2cf1314  A15     : 0x00ffffff  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xffcf1314  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x4037d67d:0x3fc9d090 0x4037fddd:0x3fc9d0d0 0x4037ff39:0x3fc9d0f0 0x4037774b:0x3fc9d110 0x40377971:0x3fc9d140 0x42008710:0x3fc9d160 0x42018a63:0x3fc9d1b0 0x42018abd:0x3fc9d1d0 0x42018a4d:0x3fc9d240 0x420189bc:0x3fc9d260 0x42018640:0x3fc9d2a0 0x420186d1:0x3fc9d2d0 0x42009c50:0x3fc9d2f0 0x4200994f:0x3fc9d310 0x42009b37:0x3fc9d370 0x42009c09:0x3fc9d3c0 0x42008ff2:0x3fc9d3e0 0x42009194:0x3fc9d510 0x42002adc:0x3fc9d570 0x42002db5:0x3fc9d590 0x42001f4e:0x3fc9d5f0

ELF file SHA256: a95de7bf094e13ed

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202ba56
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

guess-name, 2024-03-19:
I used S3, and this shows S2

lbernstone, 2024-03-20:
Looks like you are running out of memory.  Can I suggest that you move the definition of the buffer you are using in the loop to inside the loop so it gets destroyed and recreated on every read?

guess-name, 2024-03-20:
1. Where do you see that you are out of memory? I don't understand this icon.
2. I opened the PRAM inside the chip, which is 8M in size and should meet the requirements
3. I have two types of circuit boards in my hand. This circuit board is a double-layer board, which has a problem, while the four-layer board does not have this problem
4. I have defined this problem, where it occurs is in the Update.end() function _verifyEnd(); An error message is displayed if(esp_ota_set_boot_partition(_partition)){

I did not use the ring buffer you mentioned here, I directly defined a fixed size, the size of 1500Hex, I think this memory is enough

SuGlider, 2024-03-20:
>Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Usually it is related to a bad pointer with `EXCVADDR: 0xffcf1314`
More information:

### [LoadProhibited, StoreProhibited](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited)
These CPU exceptions happen when an application attempts to read from or write to an invalid memory location. The address which has been written/read is found in the EXCVADDR register in the register dump. If this address is zero, it usually means that the application has attempted to dereference a NULL pointer. If this address is close to zero, it usually means that the application has attempted to access a member of a structure, but the pointer to the structure is NULL. **If this address is something else (garbage value, not in `0x3fxxxxxx` - `0x6xxxxxxx` range), it likely means that the pointer used to access the data is either not initialized or has been corrupted.**

>I tried to update from the BIN file in LittleFS, but I kept getting errors

Some verifications:
- what is the size of `ssx.bin` file?
- what is reported as the size of the LitteFS file `/ssx.bin`?
- is it correctly reported by `filess.size()`?
- does it match the value of `filesfs` variable when the loop ends?

lbernstone, 2024-03-20:
Those heap caps malloc errors indicate to me that it is unable to allocate working memory (not flash), and specifically dma capable memory.  You only have a limited amount of that type. I suggested to move `byte sdfsd[4096];` into the while loop instead of reusing the variable over and over.  Localizing it also means that you get that 4k back before you do the shasum on the image file.
If you have two identical boards, and one fails while the other does not, that is a hardware issue.

guess-name, 2024-03-20:
@SuGlider 
1. ssx.bin size is 336,000 bytes
2. Consistent with question 1
3. filess.size() Return the same as 1
4. The end of the loop is no problem

guess-name, 2024-03-20:
NOW BUG is

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace: 0x4037723e:0x3fc9ce20 0x4037a9b5:0x3fc9ce40 0x403802d1:0x3fc9ce60 0x4037d71d:0x3fc9cf90 0x4037fddd:0x3fc9cfd0 0x4037ff39:0x3fc9cff0 0x4037774b:0x3fc9d010 0x40377971:0x3fc9d040 0x42008710:0x3fc9d060 0x42018a63:0x3fc9d0b0 0x42018abd:0x3fc9d0d0 0x42018a4d:0x3fc9d140 0x420189bc:0x3fc9d160 0x42018640:0x3fc9d1a0 0x420186d1:0x3fc9d1d0 0x42009c50:0x3fc9d1f0 0x42009a4a:0x3fc9d210 0x42009b37:0x3fc9d270 0x42009c09:0x3fc9d2c0 0x42008ff2:0x3fc9d2e0 0x42009194:0x3fc9d410 0x42002adc:0x3fc9d470 0x42002db5:0x3fc9d490 0x42001f50:0x3fc9d4f0

ELF file SHA256: 9532ae8d2881cd10

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202ba56
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

[image]

guess-name, 2024-03-20:
@lbernstone I changed the BUFF to a pointer, and using MALLOC to open memory won't work either

lbernstone, 2024-03-20:
Can you try with psram disabled?  For some reason it is not able to access the memory it wants during a critical section (when the psram would be unavailable).

SuGlider, 2024-03-20:
> NOW BUG is
> 
> assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)
 
[image]

Why would it report that the related source of panic is about esp-pm?
Power Management is not enabled in ESP32 Arduino.

guess-name, 2024-04-02:
Leave the SPI-related pins empty

closed 2024-04-02 by guess-name