espressif/arduino-esp32 #9394: help me EFUSE

akac97, 2024-03-19:
### Board

ESP32-S3-WROOM-1 N16R8

### Device Description

vecna@vecna ~> espefuse.py check_error -c auto --port /dev/ttyACM0
espefuse.py v4.7.0
Connecting....
Detecting chip type... ESP32-S3

=== Run "check_error" command ===
No errors detected
vecna@vecna ~> espefuse.py summary -c auto --port /dev/ttyACM0
espefuse.py v4.7.0
Connecting....
Detecting chip type... ESP32-S3

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
K_RTC_LDO (BLOCK1)                                 BLOCK1 K_RTC_LDO                                   = 4 R/W (0b0000001)
K_DIG_LDO (BLOCK1)                                 BLOCK1 K_DIG_LDO                                   = -28 R/W (0b1000111)
V_RTC_DBIAS20 (BLOCK1)                             BLOCK1 voltage of rtc dbias20                      = 28 R/W (0x07)
V_DIG_DBIAS20 (BLOCK1)                             BLOCK1 voltage of digital dbias20                  = -36 R/W (0x89)
DIG_DBIAS_HVT (BLOCK1)                             BLOCK1 digital dbias when hvt                      = -36 R/W (0b11001)
ADC2_CAL_VOL_ATTEN3 (BLOCK1)                       ADC2 calibration voltage at atten3                 = 112 R/W (0b011100)
TEMP_CALIB (BLOCK2)                                Temperature calibration data                       = -0.2 R/W (0b100000010)
OCODE (BLOCK2)                                     ADC OCode                                          = 81 R/W (0x51)
ADC1_INIT_CODE_ATTEN0 (BLOCK2)                     ADC1 init code at atten0                           = 476 R/W (0x77)
ADC1_INIT_CODE_ATTEN1 (BLOCK2)                     ADC1 init code at atten1                           = -4 R/W (0b100001)
ADC1_INIT_CODE_ATTEN2 (BLOCK2)                     ADC1 init code at atten2                           = 116 R/W (0b011101)
ADC1_INIT_CODE_ATTEN3 (BLOCK2)                     ADC1 init code at atten3                           = 124 R/W (0b011111)
ADC2_INIT_CODE_ATTEN0 (BLOCK2)                     ADC2 init code at atten0                           = -196 R/W (0xb1)
ADC2_INIT_CODE_ATTEN1 (BLOCK2)                     ADC2 init code at atten1                           = -16 R/W (0b100100)
ADC2_INIT_CODE_ATTEN2 (BLOCK2)                     ADC2 init code at atten2                           = 72 R/W (0b010010)
ADC2_INIT_CODE_ATTEN3 (BLOCK2)                     ADC2 init code at atten3                           = 112 R/W (0b011100)
ADC1_CAL_VOL_ATTEN0 (BLOCK2)                       ADC1 calibration voltage at atten0                 = 444 R/W (0x6f)
ADC1_CAL_VOL_ATTEN1 (BLOCK2)                       ADC1 calibration voltage at atten1                 = 452 R/W (0x71)
ADC1_CAL_VOL_ATTEN2 (BLOCK2)                       ADC1 calibration voltage at atten2                 = 420 R/W (0x69)
ADC1_CAL_VOL_ATTEN3 (BLOCK2)                       ADC1 calibration voltage at atten3                 = 468 R/W (0x75)
ADC2_CAL_VOL_ATTEN0 (BLOCK2)                       ADC2 calibration voltage at atten0                 = 408 R/W (0x66)
ADC2_CAL_VOL_ATTEN1 (BLOCK2)                       ADC2 calibration voltage at atten1                 = 180 R/W (0b0101101)
ADC2_CAL_VOL_ATTEN2 (BLOCK2)                       ADC2 calibration voltage at atten2                 = 224 R/W (0b0111000)

Config fuses:
WR_DIS (BLOCK0)                                    Disable programming of individual eFuses           = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disable reading from BlOCK4-10                     = 0 R/W (0b0000000)
DIS_ICACHE (BLOCK0)                                Set this bit to disable Icache                     = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Set this bit to disable Dcache                     = False R/W (0b0)
DIS_TWAI (BLOCK0)                                  Set this bit to disable CAN function               = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disable app cpu                                    = False R/W (0b0)
DIS_DIRECT_BOOT (BLOCK0)                           Disable direct boot mode                           = False R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)                        Set the default UART boot message output mode      = Enable R/W (0b00)
PIN_POWER_SELECTION (BLOCK0)                       Set default power supply for GPIO33-GPIO37; set wh = VDD_SPI R/W (0b1)
                                                   en SPI flash is initialized                       
PSRAM_CAP (BLOCK1)                                 PSRAM capacity                                     = 8M R/W (0b01)
PSRAM_TEMP (BLOCK1)                                PSRAM temperature                                  = 85C R/W (0b10)
PSRAM_VENDOR (BLOCK1)                              PSRAM vendor                                       = AP_3v3 R/W (0b01)
BLOCK_USR_DATA (BLOCK3)                            User data                                         
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLOCK_SYS_DATA2 (BLOCK10)                          System data part 2 (reserved)                     
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Flash fuses:
FLASH_TPUW (BLOCK0)                                Configures flash waiting time after power-up; in u = 0 R/W (0x0)
                                                   nit of ms. If the value is less than 15; the waiti
                                                   ng time is the configurable value.  Otherwise; the
                                                    waiting time is twice the configurable value     
[... 201 lines cut ...]

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

### Other Steps to Reproduce

espefuse.py -p /dev/<serial_device> burn_efuse FLASH_TYPE 1

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-19:
Uh, do you have octal flash, or are you thinking that setting the efuse will magically make add a data lane to your flash?

SuGlider, 2024-03-19:
>after i burnd efuse to enable octal psram for my board

ESP32-S3-WROOM-1 N16R8 => 16MB Quad SPI Falsh + 8MB Octal PSRAM.

>espefuse.py -p /dev/<serial_device> burn_efuse FLASH_TYPE 1
>FLASH_TYPE (BLOCK0) SPI flash type = 8 data lines R/W (0b1)

This has modified the FLASH default setup... not the PSRAM.
eFuses can't be reverted... Suggestion: Users shall never change eFuses unless the user knows very well what is being done.

Try `--flash_mode qio` as  `esptool` option.
More information: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flash-modes.html#flash-modes

Anyway, this is not an Arduino issue, maybe you could get some help posting the issue at https://github.com/espressif/esptool/issues

-------------------------
```
File md5: d492935cc146b58ceb5595df28db714a
Flash md5: b5371df2d2846b0d698bcf28a8adcfab
MD5 of 0xFF is b5371df2d2846b0d698bcf28a8adcfab
```
This is about Flashing the original firmware and what has been uploaded/read back, not about PSRAM.

akac97, 2024-03-20:
What mod the psram To octal mode?

Jason2866, 2024-03-20:
What efuse command have you send?

lbernstone, 2024-03-20:
> What mod the psram To octal mode?

You can buy this and apply it directly over the existing devkit.
https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-S3-DevKitC-1-N32R8V?qs=Li%252BoUPsLEnvTvWIWLPCZ4g%3D%3D

lbernstone, 2024-03-20:
Yes, bricked. forever.  RTFM next time.

Jason2866, 2024-03-20:
If the esptool suggestion  `--flash_mode qio` have no effect. Than yes, it is dead.

closed 2024-03-20 by Jason2866

SuGlider, 2024-03-21:
> So i bricked my esp32s3 for forever? How recover it if possible?

Fundamentally, an eFuse is a single bit of non-volatile memory with the restriction that once an eFuse bit is programmed to 1, it can never be reverted to 0.

ROM code will try to enforce whatever is written in the eFuse for security. Otherwise, why would it be worth for?
But some bits are more critical than others... have you tried the ESPTOOL option `--flash_mode qio`?

Last resource: post a question about it to https://github.com/espressif/esptool/issues
Some good soul may want to help you there... but it is very probable that it is bricked for good.