espressif/arduino-esp32 #9396: fatal error: esp_eth_driver.h: No such file or directory

woshe13, 2024-03-21:
### Board

ESP32 s3

### Device Description

eya EA5013-v2  base on esp32s3

### Hardware Configuration

e

### Version

latest master (checkout manually)

### IDE Name

vscode

### Operating System

window11

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

e

### Sketch

```cpp
e
```

### Debug Message

```plain
E:/ESP-IDF/eya_arduino/components/arduino/libraries/WiFi/src/WiFiGeneric.h:33:10: fatal error: esp_eth_driver.h: No such file or directory
   33 | #include "esp_eth_driver.h"
      |          ^~~~~~~~~~~~~~~~~~
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-03-21:
Something you have missed while installing from github? File comes from the SDK. Did you miss to run `tools/get.exe`?

PureTek-Innovations, 2024-03-26:
I have the same problem.
This may be specific to ESP-IDF on PlatformIO.
My install can't find the file either.
It could be because PlatformIO has picked up both the Arduino framework as well as the ESP-IDF framework.
I often use the Arduino framework, but it is not referred to anywhere in this project.

PACKAGES:
 - framework-arduinoespressif32 @ 2.0.14
 - framework-espidf @ 3.40405.0 (4.4.5)
 - tool-cmake @ 3.21.3
 - tool-esptoolpy @ 1.40700.0+dev2 (4.7.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...

PureTek-Innovations, 2024-03-26:
I realised that this is in arduino-esp32 rather than ESP_IDF
But I have managed to fix it.
In my case, to use the W5500 SPI ethernet module, you need V3.0.0
This can either be added to the Arduino IDE with the correct link or add it to platformio like this:

```
[env:esp32-s3-devkitc-1]
platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.02.10/platform-espressif32.zip
framework = arduino
board = esp32-s3-devkitc-1

monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

The W5500 example compiles and works with both IDEs as long as I use V3.0.0

woshe13, 2024-05-05:
> Something you have missed while installing from github? File comes from the SDK. Did you miss to run `tools/get.exe`?

I  run 'tools/get.exe', but cannot fix it

VojtechBartoska, 2024-09-12:
Hello @woshe13, can I consider this as solved and close the ticket?

italocjs, 2024-12-01:
Hello, i had the exact same issue but was unable to fix, can you share your working platformio.ini?  i get "esp_eth_driver.h" error, included by ArduinoOTA.

```c++
;This platformio.ini file is good for usage in WSL
[env:native]
lib_ldf_mode = deep+
test_framework = googletest
test_ignore = test_esp32dev
platform = native
build_unflags = -std=gnu++11
build_flags = -D USE_NATIVE_ENVIRONMENT
    -std=gnu++17
    -Wall                ; Enable all common warnings
    -Wunused-function    ; Warn about unused functions

[env:esp32dev]
platform = espressif32
framework = arduino
board = simovatrack130

platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.7
    framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.1/esp32-arduino-libs-idf-release_v5.1-632e0c2a.zip
    toolchain-xtensa-esp32 @ 12.2.0+20230208

board_build.partitions = partitions.csv
board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

build_unflags = 
    -std=gnu++11
    -mtarget-align
    -Wswitch-unreachable
    -Wstringop-overflow
    -Wincompatible-pointer-types
    -Wnonnull-compare
    -fexceptions
    -Wpointer-arith

build_flags = 
    -std=gnu++17
    -Wall                
    -Wunused-function    
    -Wno-format         
    -mno-target-align
    -Wno-switch-unreachable
    -Wno-stringop-overflow
    -fno-exceptions
 
upload_speed = 921600
monitor_speed = 115200
; monitor_filters = esp32_exception_decoder 
monitor_filters = 
    esp32_exception_decoder
    send_on_enter

lib_ignore = 
    ArduinoOTA
    HTTPUpdateServer
    NetBIOS

lib_deps = 
    adafruit/Adafruit Unified Sensor @ 1.1.14
    adafruit/Adafruit BusIO @ 1.16.1
    adafruit/Adafruit ADXL343 @ 1.6.4
    https://github.com/italocjs/DFRobot_LIS.git @ 1.0.1+sha.5d34332
    https://github.com/italocjs/geofencing_esp32.git @ 1.0.0+sha.140dd27
    https://github.com/italocjs/NuS-NimBLE-Serial @ 2.0.4+sha.9ea1cdb
    mprograms/QMC5883LCompass @ 1.2.3
    adafruit/RTClib @ 2.1.4
    mikalhart/TinyGPSPlus @ 1.1.0
    bblanchon/ArduinoJson @ 7.1.0
    adafruit/Adafruit GFX Library @ 1.11.10
    adafruit/Adafruit SSD1306 @ 2.5.11
    NimBLE-Arduino @ 1.4.2

; extra_scripts = esp32_exception_decoder.py
```

rftafas, 2025-01-06:
PlatformIO does not support Arduino Core for ESP32 newer than 2.0.17. Also, initial issue seems to be solved. Please, re-open this if tested against newer versions and if the problem remains.

For information about tools, check esp32.com, the Discussion page on this repo or the documentation (including third party tools section).

closed 2025-01-06 by rftafas