espressif/arduino-esp32 #9397: upload the"IPv6Address.h" please

woshe13, 2024-03-21:
### Related area

wifi

### Hardware specification

esp32s3

### Is your feature request related to a problem?

master lost IPv6Address.h

### Describe the solution you'd like

master lost IPv6Address.h

### Describe alternatives you've considered

master lost IPv6Address.h

### Additional context

master lost IPv6Address.h

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-03-21:
IPAddress supports both V4 and V6 addresses now. This is from upstream Arduino. IPv6Address is no longer necessary

closed 2024-03-21 by me-no-dev

seankovacs, 2024-04-19:
This breaks older libraries though...

me-no-dev, 2024-04-19:
We do understand that, but the way to go is to be compatible across cores. We added the IPv6Address, because there was no solution from upstream Arduino. Now there is, so this should be the last break in this part of code. BTW the whole IPv6 support is revised and working properly for all situations.