espressif/arduino-esp32 #9402: FreeRTOS task how to use the stack?

JC-jack-h, 2024-03-22:
### Board

ESP32-S3-WROOM-1U

### Device Description

I create a task in my app,the task just have a single loop,but when I check it's space used,it used 576 Byte stack ? I want to konw more detils about the situation for optmize source used. The idf version: V4.4.1
[image]

### Hardware Configuration

when I set the same situation in helloworld example,also used may by 618 Byte.

### Version

other

### IDE Name

esp-idf V4.4.1

### Operating System

Window 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I want to use resources more economically, because it is possible to run a stack task with a size of only 128 bytes on STM32, why does ESP32 use hundreds of bytes？

### Sketch

```cpp
void chint_key_task(void* args)
{
   while(1)
   {
      vTaskDelay(2000);
   }
}

TaskHandle_t chint_key_task_Handler;
xTaskCreate(chint_key_task, "cnt_key_task", 1024 * 2, NULL, 5, &chint_key_task_Handler);
ESP_LOGW("tag","useed = %d\n", 1024 * 2 - uxTaskGetStackHighWaterMark(chint_key_task_Handler));
```

### Debug Message

```plain
"tag","useed = 576
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-03-22:
Such questions will best be explained by our experts in the ESP-IDF repo.

You have access to plenty or RAM, so that 2/4K stacks are not an issue. You could run a small stack, just make sure you are not printing inside or do other things that would use the stack.

lbernstone, 2024-03-22:
Note that this is architecture dependent.  This code only uses 232 bytes on a -C3, and over 728 on an -S3.

JC-jack-h, 2024-03-24:
Hello, I would like to know if there is any clear information or documentation that can explain the difference between the resource consumption of ESP32 dual-core architecture and single-core when running RTOS, or the details of dual-core architecture? Because from the experience of single-core development, it is estimated that the resources of an empty task are too large.In fact, I found that the Bluetooth and Wi-Fi resources consumed in the program are more than 100K, but I don't know enough about these 2 parts and how to optimize the resources, even if the menuconfig chooses to prioritize the allocation of WiFi Bluetooth resources to the PSRAM, but it doesn't work. So I focused on other seemingly simple tasks.

lbernstone, 2024-03-24:
This forum is for fixing issues with the code contained in the the repo.  For general discussion and questions, please use https://esp32.com