espressif/arduino-esp32 #9407: Getting abort() was called at PC 0x4201f923 on core 0

Murugesh-Hobbyist, 2024-03-23:
### Board

ESP32S3

### Device Description

runs for 5 min correctly and reboots.

### Hardware Configuration

5,6 - I2C,
10,11,12,13 - SPI,
17,18 - UART1,
7,8 - UART2

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows-11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

it works exactly as expected for around 5 minutes and restarts with a message.

### Sketch

```cpp
It is a split file voids were splitted. but the main program is here.

//------------------------------------------------------PenDrive-I2C-Communication----------------------------------------------------------------------

#include <Ch376msc.h>
#include <SoftwareSerial.h>
SoftwareSerial Pendrive(17,18); // RX, TX
Ch376msc flashDrive(Pendrive); 
char adatBuffer[255];
char adat[]="Vivamus nec nisl molestie, blandit diam vel, varius mi. Fusce luctus cursus sapien in vulputate.\n";
char adat2[] = "Lorem ipsum dolor sit amet, consectetur adipiscing elit. Duis efficitur ac est eu pharetra. \n";

unsigned long totSect = 0;
unsigned long freeSect = 0;
byte percentg = 0;
byte tmpCommand; //used to store data coming from serial port
boolean readMore;
static char helpString[]= {"h:Print this help\n\n1:Create\n2:Append\n3:Read\n4:Read date/time\n"
            "5:Modify date/time\n6:Delete\n7:List dir\n8:Print free space"
            "\n9:Open/Create folder(s)/subfolder(s)"};
void printInfo();

//------------------------------------------------------SD-SPI-Communication----------------------------------------------------------------------

[... 444 lines cut ...]
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

PC: 0x40377bc0: esp_restart_noos at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/soc/esp32s3\system_internal.c:151

Decoding stack results
0x40377fb2: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system\panic.c:408
0x4037b435: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system\esp_system.c:137
0x40380c91: abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib\abort.c:46
0x4201f91b: __cxxabiv1::__terminate(void at *)()) (/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++\eh_terminate.cc:47
0x4201f962: std::terminate() at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++\eh_terminate.cc:57
0x4201f56d: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++\eh_alloc.cc:300
0x4201f3a4:  is in operator new(unsigned int) (/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54).
0x42006225:  is in Adafruit_ADS1X15::begin(unsigned char, TwoWire*) (c:\Users\murug\OneDrive\Documents\Arduino\libraries\Adafruit_ADS1X15\Adafruit_ADS1X15.cpp:66).
0x420029a3: readADC() at H:\Personal\Industry_Automation\dualx\upgradingxxx_Single_Core\upgradingxxx_Single_Core.ino:296
0x42002f42: loop() at H:\Personal\Industry_Automation\dualx\upgradingxxx_Single_Core\upgradingxxx_Single_Core.ino:280
0x420096c5: loopTask(void*) at C:\Users\murug\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\main.cpp:50

Paste exception to decode...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-23:
Looks like the problem is in Adafruit_ADS1X15.  Post an issue in their repo.  They probably need to make a minor change to support the new ADC code in 3.0

atanisoft, 2024-03-23:
This also looks like a case of failed memory allocation, which may be a bug in the Adafruit library or in your code consuming more memory than expected.