espressif/arduino-esp32 #9408: I2C stops CLK after receiving read address ACK.

thinksilicon, 2024-03-23:
### Board

ESP32-WROOM-32D-N4

### Device Description

I'm using an ESP32-WROOM-32D-N4, it's connected to an RP2040 over I2C with external pull-ups.

### Hardware Configuration

Using I2C on IO25 (scl) and IO26 (sda).

See pictures from communication captured by logic analyzer.
[image]
[image]
[image]
[image]

### Version

v2.0.9

### IDE Name

PlatformIO

### Operating System

Opensuse tumbleweed

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm using the ESP32 as an I2C master to talk to an RP2040 slave. Writing from the ESP32 to the Pico works just fine. However when I try to read from the Pico the clock signal just stops after the read-ACK and eventually I'm getting a timeout error on the ESP32.

Please see the 3 screenshots I obtained with my logic analyzer. The first screenshot shows the first transaction:

[image]

1) master sends device address + write bit, slave ACKs
2) master sends 2 bytes (with ACKs from slave)
3) master sends stop
4) master sends device address + read bit, slave ACKs
5) ??? expecting the master to send CLK signals so slave can send data.

However it appears the ESP just hangs until it reaches some kind of timeout. You can see in the second screenshot that SDA is pulled down twice after about 13 ms. I don't know which side initiates this, but I could imagine it's the Pico trying to resync to the master.
[image]

You can see this pattern repeats over every subsequent request.
[image]

The last screenshot shows a following request. You can see the longer sync frames of the clock signal (where the Pico actually tries to send some data). You can then see a NACK + RepeatStart which doesn't make much sense here and then just the same communication as before.
[image]

### Sketch

```cpp
### MASTER ###
#include <Arduino.h>
#include <Wire.h>

#define DEVICE_ADDRESS 0x42

int readI2CByte(uint8_t deviceAddress, byte registerAddress, byte data ) {
  Wire.beginTransmission( deviceAddress );
  Wire.write( registerAddress );
  Wire.write( data );
  byte error = Wire.endTransmission();

  if( error == 0 ) {
    size_t b = Wire.requestFrom( deviceAddress, (size_t)1 );
    if( Wire.available() > 0 ) {
      uint data = Wire.read();
      Wire.endTransmission();
      Serial.print("Data received: ");
      Serial.println(data);
      return data;
    } else {
      Serial.println( "Error: No data received" );
      return -1;
    }
  } else {
    Serial.print( "Error starting communication: " );
    Serial.println(error);
    return -1;
  }
}

void setup() {
  Serial.begin(115200);

  Serial.print("ESP-IDF Version: ");
  Serial.println(esp_get_idf_version());

  Serial.println( "Starting: Enabling RP2040." );

  pinMode( 4, OUTPUT ); 
  pinMode( 32, OUTPUT );
  digitalWrite( 4, HIGH );  // enable the RP2040
  digitalWrite( 32, HIGH ); // turn on a LED

  Wire.begin( 26, 25, 100000U );
}

void loop() {
  // put your main code here, to run repeatedly:
  delay( 100 );
  if (Serial.available() > 0) {
    char receivedChar = Serial.read();
    if (receivedChar == 'R' || receivedChar == 'r') {
      Serial.println("RESET RP2040");
      digitalWrite( 32, LOW );
      digitalWrite(4, LOW);
      delay(100);
      digitalWrite(4, HIGH );
      digitalWrite( 32, HIGH );

    }
    if( receivedChar == 'A' || receivedChar == 'a' ) {
      Serial.println( "Reading I2C..." );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x00 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x01 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x02 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x03 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x04 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x05 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x06 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x07 );
      readI2CByte( DIMMER_ADDRESS, 0x07, 0x08 );
    }
  }
}

### SLAVE ###
#include <Arduino.h>
#include <Wire.h>

arduino::MbedI2C I2C{PICO_DEFAULT_I2C_SDA_PIN, PICO_DEFAULT_I2C_SCL_PIN};
volatile uint8_t send_buffer[2];
volatile uint8_t send_bytes;

void receiveEvent( uint numBytes ) {
    while( I2C.available() > 0 ) {
        byte command = I2C.read();
        Serial.print( "Received command " );
        char buffer[5];
        sprintf( buffer, "0x%02x", command );
        Serial.println( buffer );

        if( command == 0x07 ) {
            uint8_t channel = I2C.read(); // not used in this example
            send_buffer[0] = 0x12;
            send_bytes = 1;
        }
    }
}

void requestEvent() {
    I2C.write( (char*)send_buffer, (size_t)send_bytes );
}

void setup() {
  // Initialize serial communication for debugging
  Serial.begin(115200);

  while (!Serial);

  delay(100);
  Serial.print(F("\nStarting on ")); Serial.println(BOARD_NAME);

  I2C.begin( 0x42 );
  I2C.setClock( 100 * 1000 );
  I2C.onReceive(receiveEvent); // Set up the receiveEvent callback
  I2C.onRequest(requestEvent); // Set up the requestEvent callback

}

void loop() {
  delay(10);
}
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
Starting: Enabling RP2040.
Reading I2C...
[ 22342][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 22475][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 22608][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 22740][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 22873][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 23005][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 23137][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 23270][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received
[ 23402][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
Error: No data received

On the Pico I'm seeing 0x07 coming in. I also had debug output in my requestEvent() function. So I can see that it runs I2C.write(...)... but with missing clock it won't be able to send anything.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-03-24:
@me-no-dev - PTAL. Thanks!

thinksilicon, 2024-03-27:
I've also just upgraded to use the latest version (v2.0.14) with IDF 4.4.6, but same behavior.

lbernstone, 2024-03-27:
Did you try `Wire.endTransmission(true);` after your "command" write?

me-no-dev, 2024-03-27:
I do not think that the issue is on the ESP32 side. Looks to me like the Pico is pulling SCL low after the ACK of the read address and that is preventing the ESP32 to continue. It's the ESP32 that is pulling the SDA LOW to try and reset the bus, but it's failing.

me-no-dev, 2024-03-27:
On ESP32 specifically, the slave buffer needs to be pre-loaded in order for the master to read it. Maybe something like that is also true on the Pico? What will happen if you `I2C.write( (char*)send_buffer, (size_t)send_bytes );` in the Pico's `setup()`?

thinksilicon, 2024-03-31:
Ok,
I've tried a few variations:
1) using Wire.endTransmission(true); after Wire.write(...);
2) setting the tx buffer in the setup routine

I think I'll try opening a bug report with the ArduinoCore-mbed implementation for the Pico. It appears that indeed the clock line gets pulled low for some reason. I've seen there's a slaveWrite() on ESP32, but not for the Pico. Now I'm wondering if that piece is missing on the Pico. 
The [Wire.cpp](https://github.com/arduino/ArduinoCore-mbed/blob/main/libraries/Wire/Wire.cpp#L148) is relatively simple, it appears there's a thread running which should handle sending the bytes out. I didn't get much deeper what slave->write() exactly does. I've also tried uncommenting the slave->stop() in the source, but it didn't change anything.

I will also try to order some I2C device and check if the ESP32 is fine with that. At that point I can pretty certainly point to the ArduinoCore-mbed code as the main issue. If you don't mind leaving this issue open for now, I'll report back my findings.

thinksilicon, 2024-04-08:
Ok, I was able to verify that this is an issue with the ArduinoCore-mbed code. I wrote a very simple sketch with the Pico SDK and sending and receiving works using those functions. It's really just a problem the mbed code.

Closing this, sorry for the noise.

closed 2024-04-08 by thinksilicon