espressif/arduino-esp32 #9409: ESP32 SD CARD NOT DETECTED

bacaneriaslocas, 2024-03-23:
### Board

ESP32 WROOM 32D

### Device Description

Im using my own developed ESP32 board.
[PCB_PCB_Helios-3_2024-03-23.json](https://github.com/espressif/arduino-esp32/files/14733657/PCB_PCB_Helios-3_2024-03-23.json)
[Schematic_helios_V2_2023-09-24 (1).pdf](https://github.com/espressif/arduino-esp32/files/14733663/Schematic_helios_V2_2023-09-24.1.pdf)

### Hardware Configuration

The board counts with different sensors. First of all, for programing it I use a ft232rl usb to serial converter, the sensors that I have conected are: BMP388 (VSPI), MPU9250 (I2C), LoRa module (VSPI), microSD card holder (VSPI).

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When I run the SD test example code, Im not able to inicialice the sd card and this error is shown _Card Mount Failed_.
I have tried all sorts of things like using external modules, soldering direct cables to the spi pins and breaking conexions to avoid any noise in the signal, formating the sd cards, buying new ones, using a different spi interface; but the error is still there. I have been able to use the bmp388 thougth the VSPI interface without any problems and I have been able to use the sd card with an arduino nano.

### Sketch

```cpp
/*
 * Connect the SD card to the following pins:
 *
 * SD Card | ESP32
 *    D2       -
 *    D3       SS
 *    CMD      MOSI
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      SCK
 *    VSS      GND
 *    D0       MISO
 *    D1       -
 */
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
}

void removeDir(fs::FS &fs, const char * path){
    Serial.printf("Removing Dir: %s\n", path);
    if(fs.rmdir(path)){
        Serial.println("Dir removed");
    } else {
        Serial.println("rmdir failed");
    }
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

    Serial.print("Read from file: ");
    while(file.available()){
        Serial.write(file.read());
    }
    file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }
    file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
    file.close();
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
    Serial.printf("Renaming file %s to %s\n", path1, path2);
    if (fs.rename(path1, path2)) {
        Serial.println("File renamed");
    } else {
        Serial.println("Rename failed");
    }
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\n", path);
    if(fs.remove(path)){
        Serial.println("File deleted");
    } else {
        Serial.println("Delete failed");
    }
}

void testFileIO(fs::FS &fs, const char * path){
    File file = fs.open(path);
    static uint8_t buf[512];
    size_t len = 0;
    uint32_t start = millis();
    uint32_t end = start;
    if(file){
        len = file.size();
        size_t flen = len;
        start = millis();
        while(len){
            size_t toRead = len;
            if(toRead > 512){
                toRead = 512;
            }
            file.read(buf, toRead);
            len -= toRead;
        }
        end = millis() - start;
        Serial.printf("%u bytes read for %u ms\n", flen, end);
        file.close();
    } else {
        Serial.println("Failed to open file for reading");
    }

    file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }

    size_t i;
    start = millis();
    for(i=0; i<2048; i++){
        file.write(buf, 512);
    }
    end = millis() - start;
    Serial.printf("%u bytes written for %u ms\n", 2048 * 512, end);
    file.close();
}

void setup(){
    Serial.begin(115200);
    if(!SD.begin(25)){    // if you have seen the schematic you may think that this pin does correspond, but its because im using an external conexion
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

    listDir(SD, "/", 0);
    createDir(SD, "/mydir");
    listDir(SD, "/", 0);
    removeDir(SD, "/mydir");
    listDir(SD, "/", 2);
    writeFile(SD, "/hello.txt", "Hello ");
    appendFile(SD, "/hello.txt", "World!\n");
    readFile(SD, "/hello.txt");
    deleteFile(SD, "/foo.txt");
    renameFile(SD, "/hello.txt", "/foo.txt");
    readFile(SD, "/foo.txt");
    testFileIO(SD, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop(){

}
```

### Debug Message

```plain
22:43:11.386 -> rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
22:43:11.386 -> configsip: 0, SPIWP:0xee
22:43:11.386 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:43:11.386 -> mode:DIO, clock div:2
22:43:11.386 -> load:0x3fff0030,len:1240
22:43:11.386 -> load:0x40078000,len:12984
22:43:11.386 -> load:0x40080400,len:3648
22:43:11.386 -> entry 0x400805f8
22:43:12.104 -> Card Mount Failed
```

### Other Steps to Reproduce

The only logical conclusion that i have thougth is that the error might be on the arduino ide, but im not sure as i dont have an esp32 dev board to replicate the issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pipi61, 2024-03-23:
in schematic where the esp32?

bacaneriaslocas, 2024-03-23:
> in schematic where the esp32?

I uploaded the previous version of the board, now it´s corrected. Sorry for the mistake

lbernstone, 2024-03-23:
You need to initialize SPI with the pins you've assigned before you call SD.begin.  I assume you don't care whether it is actually VSPI, but just want to use SPI.
`SPI.begin(CLK, MISO, MOSI, SS);`

bacaneriaslocas, 2024-03-23:
> You need to initialize SPI with the pins you've assigned before you call SD.begin. I assume you don't care whether it is actually VSPI, but just want to use SPI. `SPI.begin(CLK, MISO, MOSI, SS);` Turning on verbose logging may give you more information if it isn't a physical connection problem.

#define MISO 19
#define MOSI 23
#define CLK 18
#define SS 14

... rest of code

void setup(){
    Serial.begin(115200);
    SPI.begin(CLK, MISO, MOSI, SS);
    if(!SD.begin(33)){
        Serial.println("Card Mount Failed");
        return;
    }

I have tried with this and i get the same result.
**Thanks for the patience and help**

lbernstone, 2024-03-24:
What is 33 & 14 there?  Your client select pin for the SD card is 13 according to your schematic.

bacaneriaslocas, 2024-03-24:
> What is 33 & 14 there? Your client select pin for the SD card is 13 according to your schematic.

Thanks, I did not realidad I was usigng diferent ss pins, btw do not pay atención to what ss pin im using because I'm trying different configurations like using an external module. I will correct the mistake and upload the result when I'm able to. Thanx.

bacaneriaslocas, 2024-03-24:
> What is 33 & 14 there? Your client select pin for the SD card is 13 according to your schematic.

I have corrected the code but the error persists.

bacaneriaslocas, 2024-03-24:
I have updated my esp32 board in the arduino IDE but im not getting different results
**New code that im triying**
`/*
 * pin 1 - not used          |  Micro SD card     |
 * pin 2 - CS (SS)           |                   /
 * pin 3 - DI (MOSI)         |                  |__
 * pin 4 - VDD (3.3V)        |                    |
 * pin 5 - SCK (SCLK)        | 8 7 6 5 4 3 2 1   /
 * pin 6 - VSS (GND)         | ▄ ▄ ▄ ▄ ▄ ▄ ▄ ▄  /
 * pin 7 - DO (MISO)         | ▀ ▀ █ ▀ █ ▀ ▀ ▀ |
 * pin 8 - not used          |_________________|
 *                             ║ ║ ║ ║ ║ ║ ║ ║
 *                     ╔═══════╝ ║ ║ ║ ║ ║ ║ ╚═════════╗
 *                     ║         ║ ║ ║ ║ ║ ╚══════╗    ║
 *                     ║   ╔═════╝ ║ ║ ║ ╚═════╗  ║    ║
 * Connections for     ║   ║   ╔═══╩═║═║═══╗   ║  ║    ║
 * full-sized          ║   ║   ║   ╔═╝ ║   ║   ║  ║    ║
 * SD card             ║   ║   ║   ║   ║   ║   ║  ║    ║
 * Pin name         |  -  DO  VSS SCK VDD VSS DI CS    -  |
 * SD pin number    |  8   7   6   5   4   3   2   1   9 /
 *                  |                                  █/
 *                  |__▍___▊___█___█___█___█___█___█___/
 *
 * Note:  The SPI pins can be manually configured by using `SPI.begin(sck, miso, mosi, cs).`
 *        Alternatively, you can change the CS pin and use the other default settings by using `SD.begin(cs)`.
 *
 * +--------------+---------+-------+----------+----------+----------+
 * | SPI Pin Name | ESP8266 | ESP32 | ESP32-S2 | ESP32-C3 | ESP32-S3 |
 * +==============+=========+=======+==========+==========+==========+
 * | CS (SS)      | GPIO15  | GPIO5 | GPIO5    | GPIO13   | GPIO13   |
 * +--------------+---------+-------+----------+----------+----------+
 * | DI (MOSI)    | GPIO13  | GPIO23| GPIO24   | GPIO14   | GPIO14   |
 * +--------------+---------+-------+----------+----------+----------+
 * | DO (MISO)    | GPIO12  | GPIO19| GPIO25   | GPIO15   | GPIO15   |
 * +--------------+---------+-------+----------+----------+----------+
 * | SCK (SCLK)   | GPIO14  | GPIO18| GPIO19   | GPIO16   | GPIO16   |
 * +--------------+---------+-------+----------+----------+----------+
 *
 * For more info see file README.md in this library or on URL:
 * https://github.com/espressif/arduino-esp32/tree/master/libraries/SD
 */

#include "FS.h"
#include "SD.h"
#include "SPI.h"

/*
Uncomment and set up if you want to use custom pins for the SPI communication
#define REASSIGN_PINS*/
int sck = 18;
int miso = 19;
int mosi = 23;
int cs = 33;

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
[... 167 lines cut ...]
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

    listDir(SD, "/", 0);
    createDir(SD, "/mydir");
    listDir(SD, "/", 0);
    removeDir(SD, "/mydir");
    listDir(SD, "/", 2);
    writeFile(SD, "/hello.txt", "Hello ");
    appendFile(SD, "/hello.txt", "World!\n");
    readFile(SD, "/hello.txt");
    deleteFile(SD, "/foo.txt");
    renameFile(SD, "/hello.txt", "/foo.txt");
    readFile(SD, "/foo.txt");
    testFileIO(SD, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop(){

}`

bacaneriaslocas, 2024-03-24:
just in case this helps

[image]

bacaneriaslocas, 2024-03-24:
-> [   876][W][sd_diskio.cpp:522] ff_sd_initialize(): GO_IDLE_STATE failed
882][E][sd_diskio.cpp:812] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
more errors i have been looking

SuGlider, 2024-03-24:
@P-R-O-C-H-Y - I think that you have a good expeirence with SDCard. Please help is possible. Thanks!

P-R-O-C-H-Y, 2024-04-02:
@bacaneriaslocas You need to uncomment the `#define REASSIGN_PINS` in your code to use the custom pins you declared.
```
/*
Uncomment and set up if you want to use custom pins for the SPI communication
#define REASSIGN_PINS*/
int sck = 18;
int miso = 19;
int mosi = 23;
int cs = 33;
```

Weygand, 2024-04-03:
I also have problems with the latest version and using the SD card. 
When I try to access the SD card via the MYSD library, it works without any problems. 
I have already tried the example, mulitple Codes and the code here.
With the arduino-esp32 I get the following messages: 

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: [  5012][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  5012][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[  5014][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  5020][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 10029][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 10029][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount Failed

Maybe its the same issue

P-R-O-C-H-Y, 2024-04-08:
> I also have problems with the latest version and using the SD card. When I try to access the SD card via the MYSD library, it works without any problems. I have already tried the example, mulitple [...]

@Weygand What version of the Arduino-esp32 are you using? Also what SoC is used? Thanks

garcia-s, 2024-05-14:
Hello, i'm having some problems with SD Cards. I've tried, like a ton of stuff, and I don't know if I'm doing something wrong or something is wrong with the API, my device, my brain or my reading skills. 

The device is: 

Freematic ONE+
Chip: ESP32-D0WDQ6
Compiling with: esp32 2.0.16

I'm in Fedora 40 but I disabled SElinux and I do have full permissions  to write to ttyUSB0

I'm using arduino-cli to compile and upload, here is my Makefile

`
compile:
	arduino-cli compile -b esp32:esp32:esp32:DebugLevel=verbose  --libraries ../libraries

upload:
	arduino-cli upload -b esp32:esp32:esp32 --port /dev/ttyUSB0 

attach: 
	arduino-cli monitor -b esp32:esp32:esp32 --port /dev/ttyUSB0 -c 115200 
`

I have like a very simple code to test;

`
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void setup() {

    Serial.begin(115200);

    for(;;) {
      if (!SD.begin(5,SPI,1000000)) {
        Serial.println("ERROR: SD Card Mount Failed");
        delay(2000);
        continue;
      }
      Serial.println("Finished initializing SD card. Getting card type...");
      return;
    }
}
`
And it gives me this error:

`[  2745][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  2749][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[  2758][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[  2765][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  3273][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  3277][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
`

I used those values since they were used on some Freematic sketches that were working before, however reverting to esp 1.x.x doesn't solve the issue. I might be doing something wrong, but I've read around and I'm really out of ideas here. If somebody can please point me in the right direction I will really appreciate that.

DavidStacer, 2024-11-20:
I was having a very similar card mount failed problem as many have expressed here.  It works if you use the default SPI GPIOs for the specific ESP32 variant as specified in the ascii art table in the post above by @bacaneriaslocas .   

If I used any custom GPIOs it would fail.  I added some debug code to begin function in SPI.cpp to see what was being set.  That call worked and was setting the correct pins. The next call SD.begin(cs) would also print my debug code in SPI.cpp.  I traced it down to line 30 in the [SD.cpp](https://github.com/espressif/arduino-esp32/blob/release/v3.1.x/libraries/SD/src/SD.cpp) file. It makes another call to spi.begin() but doesn't pass any parameters.  I think that just reverted SPI back to the default values.  Once I commented that spi.begin() call in SD.cpp it works on any combination of custom SPI pins.

lbernstone, 2024-11-21:
If SPI has already been initialized, then the call to spi.begin should [immediately exit out](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.cpp#L67) without any changes. Can you post some example code so we can try to work around it?

P-R-O-C-H-Y, 2025-01-09:
Hi, I am closing this issue as there was no activity for a long time and I have not occurred any new issues with SD on latest versions. If the issue persists feel free to reopen or open a new issue. Thanks

closed 2025-01-09 by P-R-O-C-H-Y