espressif/arduino-esp32 #9411: error: pll_cal exceeds 2ms! wifi issue

DanielTerletzkiy, 2024-03-24:
### Board

esp32c6

### Device Description

dfrobot firebeetle 2

### Hardware Configuration

this board has a waveshare display and is powered by a 2000mah battery

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

while operating on battery (as far as i know not happening while connected via usb) and waking up after deep-sleep mode (after a few hours of doing so), the device refuses to connect to any WiFI networks and prints this message into the Serial interface when booting: error: `pll_cal exceeds 2ms!`.

the board went to sleep and booted for about 5 days perfectly without fail and always connected to the internet afterwards, this issue started just yesterday, when i saw that the display had old data on it, when i plugged in my pc while on battery, i saw the message above and a constant reboot loop, because it cannot connect.

i have found out that the hardware reset button fixes this issue, but pressing this every time is not good enough, i have written code to reset the device via software, tried ESP.restart() that didn't work, abort() also without luck, next thing to try will be a cable connected with gpio to the rst button.

As of now i also don't know the exact time it takes for this problem to happen, but once it happens the device can only recover via hardware reset.

currently i also have not found any info on the internet, but i did find one that has the same board and problem as me here: https://github.com/esp8266/Arduino/issues/2566#issuecomment-1874736708

i also opened a forum post on the dfrobot page here: https://www.dfrobot.com/forum/topic/334203

### Sketch

```cpp
setup
///
Serial.begin(115200);
analogReadResolution(12);

#ifdef DEVELOPMENT
    delay(1000);
#endif

    General::InitDevice()

////

 

RTC_DATA_ATTR unsigned long bootCount = 0;
RTC_DATA_ATTR unsigned long requestFails = 0;

 

void General::InitDevice() {
    Serial.println("---------------------");

    Serial.print("bootCount: ");
    Serial.println(bootCount);

    Serial.print("requestFails: ");
    Serial.println(requestFails);

    if (requestFails > 2) {
        abort();
        //ESP.restart();
    }

    Serial.println("Init Device");
    preferences.begin(PROJECT, false);

    ++bootCount;

    String uid = preferences.getString("uid", "");
    if (uid.length() <= 0) {
        uid = General::GenerateUUID();
        preferences.putString("uid", uid);
    }

    Serial.print("uid: ");
    Serial.println(uid);

    String hostname = PROJECT + String("-") + uid.substring(0, 8); //project name + first 8 chars from uid
    WiFiClass::setHostname(hostname.c_str());

    Serial.print("SSID: ");
    Serial.println(WIFI_SSID);

    uint8_t wifiFails = 0;

    WiFiClass::mode(WIFI_STA);
    //WiFi.setAutoReconnect(true);

    WiFi.begin(WIFI_SSID, WIFI_PASS);
    while (WiFiClass::status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
        if (wifiFails > 24) {
            requestFails++;
            General::PutToSleep();
        }
        wifiFails++;
    }
    Serial.println("Connected!")

///

void General::PutToSleep(unsigned long microSeconds) {
    uptime += millis();
    General::UpdateStatus(DeviceState::SLEEP);

    //WiFi.setSleep(false);
    WiFi.disconnect();
    ESP.deepSleep(microSeconds);
}
```

### Debug Message

```plain
pll_cal exceeds 2ms!
pll_cal exceeds 2ms!
bootCount: 2
requestFails: 2
Init Device
uid: f5bf7243-f04c-49d6-ab3a-0d179fb352cc
SSID: -redacted-
......................
(gets put to sleep, wakes up after 2 sec and tries again, 
after two tries it does abort(); and the cycle continues
)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-24:
This is a power issue. I don't know how you are managing your power, but once the battery is unable to adequately supply the LDO, you are not going to get enough current to start WiFi.

SuGlider, 2024-03-24:
This is not an Arduino issue. I'll transfer it to the discussion session. Please treck it there. Thanks.