espressif/arduino-esp32 #9416: Missing file error

hanyswana, 2024-03-25:
### Board

ESP32 Dev Module

### Device Description

Huzzah ESP32

### Hardware Configuration

No other hardware connected

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Missing file or directory of flatbuffers/flatbuffers.h in file of /tensorflow/lite/schema/schema_generated.h

Attached is the model.

[model_norm_euc_parquet.zip](https://github.com/espressif/arduino-esp32/files/14738112/model_norm_euc_parquet.zip)

### Sketch

```cpp
#include "model_norm_euc_parquet.h"
// #include <TensorFlowLite.h>
#include <Arduino.h>

#include <tensorflow/lite/schema/schema_generated.h>
#include <tensorflow/lite/micro/tflite_bridge/micro_error_reporter.h>
#include <tensorflow/lite/micro/micro_interpreter.h>
#include <tensorflow/lite/micro/micro_mutable_op_resolver.h> // Modified include

// Create an area to store the model data
constexpr int tensorArenaSize = 16 * 1024; // Adjust size as needed
byte tensorArena[tensorArenaSize];

float input_data[19] = {15.7785, 294.6944, 150.8488, 58.0983, 67.3988, 72.9007, 30.7708, 31.4791, 16.6024, 10.7993, 4.1336, 5.9493, 6.0067, 7.3495, 5.0309, 4.456, 3.3453, 9.9208, 9.6588};

void setup() {
  Serial.begin(9600);

  static tflite::MicroErrorReporter micro_error_reporter; // Error reporter
  static tflite::MicroMutableOpResolver<10> resolver; // Modify this number based on the number of operations you need to add

  // Add operations to the resolver. Example for adding fully connected op:
  // resolver.AddFullyConnected();

  const tflite::Model* model = tflite::GetModel(tflite_model);
  if (model->version() != TFLITE_SCHEMA_VERSION) {
    Serial.println("Model schema version mismatch!");
    return;
  }

  static tflite::MicroInterpreter interpreter(model, resolver, tensorArena, tensorArenaSize, nullptr);

  TfLiteStatus allocate_status = interpreter.AllocateTensors();
  if (allocate_status != kTfLiteOk) {
    Serial.println("Tensor allocation failed");
    return;
  }

  Serial.println("line 46");

  // Get pointer to input tensor
  TfLiteTensor* input_tensor = interpreter.input(0);
  // Copy data to input tensor
  for (size_t i = 0; i < 19; i++) {
    input_tensor->data.f[i] = input_data[i];
  }

  Serial.println("line 55");

  // Run inference
  TfLiteStatus invoke_status = interpreter.Invoke();
  if (invoke_status != kTfLiteOk) {
    Serial.println("Inference failed");
    return;
  }

  Serial.println("line 64");

  // Get the output from the model
  TfLiteTensor* output_tensor = interpreter.output(0);

  // Assuming output is a single float value
  float output = output_tensor->data.f[0];
  Serial.print("Model output: ");
  Serial.println(output);

  Serial.println("line 74--done");
}

void loop() {
  // Empty loop
}
```

### Debug Message

```plain
fatal error: flatbuffers/flatbuffers.h: No such file or directory
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-03-25:
This actually seems more like an issue with tensorflow.  flatbuffers is there.  If you include it (`#include <flatbuffers/flatbuffers.h>`) before the tensorflow stuff, it should compile.

hanyswana, 2024-03-25:
[Espressif-arduino-esp32-lib.zip](https://github.com/espressif/arduino-esp32/files/14739857/Espressif-arduino-esp32-lib.zip)

This is the modified path of those missing files/directories in the library. Then, this error raised:

c:/users/hanys.harun/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\hanys.harun\AppData\Local\Temp\arduino\sketches\B9CF493AE81797DC309AB15466F5D4B4\sketch\TFLite_Espressif.ino.cpp.o:(.literal.__tcf_2+0x4): undefined reference to `tflite::MicroInterpreter::~MicroInterpreter()'
c:/users/hanys.harun/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\hanys.harun\AppData\Local\Temp\arduino\sketches\B9CF493AE81797DC309AB15466F5D4B4\sketch\TFLite_Espressif.ino.cpp.o:(.literal._Z5setupv+0x60): undefined reference to `tflite::MicroInterpreter::MicroInterpreter(tflite::Model const*, tflite::MicroOpResolver const&, unsigned char*, unsigned int, tflite::MicroResourceVariables*, tflite::MicroProfilerInterface*, bool)'
c:/users/hanys.harun/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\hanys.harun\AppData\Local\Temp\arduino\sketches\B9CF493AE81797DC309AB15466F5D4B4\sketch\TFLite_Espressif.ino.cpp.o: in function `__tcf_2':
C:\Users\hanys.harun\Documents\Arduino\TFLite_Espressif/TFLite_Espressif.ino:31: undefined reference to `tflite::MicroInterpreter::~MicroInterpreter()'
c:/users/hanys.harun/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\hanys.harun\AppData\Local\Temp\arduino\sketches\B9CF493AE81797DC309AB15466F5D4B4\sketch\TFLite_Espressif.ino.cpp.o: in function `setup()':
C:\Users\hanys.harun\Documents\Arduino\TFLite_Espressif/TFLite_Espressif.ino:31: undefined reference to `tflite::MicroInterpreter::MicroInterpreter(tflite::Model const*, tflite::MicroOpResolver const&, unsigned char*, unsigned int, tflite::MicroResourceVariables*, tflite::MicroProfilerInterface*, bool)'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

me-no-dev, 2024-03-25:
I can not reproduce the error with the attached sketch and model. It compiles fine.

```
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Using library TFLiteMicro at version 1.0.0 in folder: /Users/ficeto/Documents/Arduino/hardware/espressif/esp32/libraries/TFLiteMicro 
Sketch uses 316201 bytes (24%) of program storage space. Maximum is 1310720 bytes.
Global variables use 59480 bytes (18%) of dynamic memory, leaving 268200 bytes for local variables. Maximum is 327680 bytes.
```

lbernstone, 2024-03-25:
This looks like 2.0.14, not latest master
c:/users/hanys.harun/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe

hanyswana, 2024-03-26:
> I can not reproduce the error with the attached sketch and model. It compiles fine.
>
> ```
> Creating esp32 image...
> Merged 2 ELF sections
> Successfully created esp32 image.
> Using library TFLiteMicro at [...]

Which library did you used? The original one with error of no file/directory of flatbuffers/flatbuffers.h or the one that I attached with modified path of those missing file/directory?

hanyswana, 2024-03-26:
> This looks like 2.0.14, not latest master c:/users/hanys.harun/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe

Did you mean ESP32 version? If yes, I used the latest version 2.0.14 already.

lbernstone, 2024-03-26:
TFLite is not included in 2.0.14.  It is part of 3.0.0.  Please post correct information in your issue description.

hanyswana, 2024-03-26:
> TFLite is not included in 2.0.14. It is part of 3.0.0. Please post correct information in your issue description.

Sorry, I put the wrong version of ESP32 in the issue description. I already updated the correct version of ESP32 which is v2.0.14. 

So, that means TFLite cannot be used in ESP32 v2.0.14?

me-no-dev, 2024-03-26:
yes. TFLite is coming with version 3.0.0. There is an alpha3 version of it available through the board manager (development link). You can try it there

hanyswana, 2024-03-27:
> yes. TFLite is coming with version 3.0.0. There is an alpha3 version of it available through the board manager (development link). You can try it there

I already download the TFLite v3.0.0-alpha1 here: https://github.com/espressif/arduino-esp32/releases/download/3.0.0-alpha1/esp32-3.0.0-alpha1.zip

Now, I don't know where I should place this folder and how to use this new version of ESP32 as my board. Can you help me? Thank you.

lbernstone, 2024-03-27:
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide
Use the development link

hanyswana, 2024-03-27:
It was done. Thank you :))))

hanyswana, 2024-04-18:
I used ESP32 v3.0.0-alpha3 and it worked for the model, but the other function code was unstable. Then I tried version 2.0.15 (the latest stable version of ESP32), but I still got the same error as mentioned previously. Is there any method or solution to deploy Tensor Flow Lite Micro (TFLM) model with the stable version of ESP32?

me-no-dev, 2024-04-18:
When 3.0.0 final is released

hanyswana, 2024-04-18:
> When 3.0.0 final is released

When is the estimated release date?

closed 2025-01-15 by Parsaabasi